ESPWortuhr/Multilayout-ESP-Wordclock #31: [BUG] Funktionsanzeige nach verlassen der Webseite falsch

michi1g, 2020-11-30:
Wenn ich die Webseite verlasse nachdem ich z.B. Sekunden eingestellt habe, wird beim erneuten Besuch der Seite Wortuhr angezeigt, obwohl die Sekundenanzeige noch läuft.
Bei einem normalen reload der Seite tritt der Fehler nicht auf

dbambus, 2020-12-02:
Ja, das liegt daran, dass die Website die Config erst beim Besuch der "Einstellungen" Unterseite anfragt. Kannst du mal überprüfen, erst auf die "Einstellungen" Seite zu gehen und dann auf "Funktionen". Ist dann der Bug auch vorhanden ?

michi1g, 2020-12-02:
Bug ist auch vorhanden wenn man vorher auf Einstellungen gegangen ist

dbambus, 2020-12-02:
Ich habe da mal wieder nachgesehen. Ich kann den Fehler soweit auf den HTML Code zurückführen.

Bei Radio-Buttons kann man das checked Attribut als vorselektierbares Element verwenden, während man zum Beispiel bei einem DropDown Menü relativ einfach über das <select> Attr. eine Vorauswahl anhand eines eingelesenen Parameters machen kann. Dies ist bei Radio-Buttons nicht so einfach möglich. @michi1g kannst du da vielleicht suchen ? Wie man das in HTML und JS hinbekommt ?

Eine (un-)schöne Möglichkeit wäre, dass man da keine Checkboxen verwendet, sondern Dropdowns.

Eisbaeeer, 2020-12-02:
Du musst das value als  char checkbox[8] = "checked" oder  checkbox[0]=0 auswerten. Habe ich in anderen Projekten schon so gemacht. Gruß Eisbaeeer Von meinem/meiner Galaxy gesendet

Bei Radio-Buttons kann man das checked Attribut als vorselektierbares Element verwenden, während man zum Beispiel bei einem DropDown Menü relativ einfach über das  Attr. eine Vorauswahl anhand eines eingelesenen Parameters machen kann. Dies ist bei Radio-Buttons nicht so einfach möglich. @michi1g kannst du da vielleicht suchen ? Wie man das in HTML und JS hinbekommt ?

dbambus, 2020-12-03:
Hey @Eisbaeeer,

würde deine Lösung auch für Radio-Buttons funktionieren, da Checkboxen doch was anderes sind. Und wäre es möglich, dass du dich der Problematik annimmst ? 
Ich würde mich um andere Bugs kümmern wollen, um den Release noch vor Weihnachten hinzubekommen.

LG
dbambus

Eisbaeeer, 2020-12-03:
Radio Button geht sicher auch. Hier mal das Beispiel, wie das bei mir realisiert ist:
Im HTML:
`const char _PAGE_CONFIG_YOUTU[] PROGMEM  = "<label class=inline><input type=checkbox name=youtu value=checked %YOUTUSTAT% > YoutTube Abonnenten</label>";`
Im Code:
` TokenStringPair youtupair[1];
youtupair[0].setPair("%YOUTUSTAT%", youtuStat);
webpage.add_P(_PAGE_CONFIG_YOUTU, youtupair,1);`
Den Status von der Webseite abholen:
`void getConfig() {
// YouTubeCheckbox
  String youtubeString = server.arg("youtu");
  youtubeString.toCharArray(youtuStat,8);
}`

Das Problem ist eben, dass es kein wirkliches Value gibt, ob die Checkbox an oder aus ist.
Ich kann mir das schon mal anschauen, aber im Moment (also vor Weihnachten) wird das nichts, sorry.
Gruß Eisbaaaer

dbambus, 2021-01-28:
@ATho95 und @tali, 

ist das nicht im aktuellem Branch schon behoben ?

dbambus, 2021-01-28:
Ah richtig, grad getestet. Aber wenn man von einer alternativen Anzeige zurück auf die Wortuhr wechselt, dann wird die Anzeige erst nach einer vollen Minute aktualisiert gleiches gilt für die Änderung der Farbe, das gilt es noch zu beheben. Da würde ich mich aber morgen drum bemühen wollen.

closed 2021-02-01 by dbambus