ESPWortuhr/Multilayout-ESP-Wordclock #38: wortuhr 24

3damberger, 2020-12-16:
Hi ! 

Du sag mal, ist der Code auch für die Wortuhr 24 möglich ? also mit der 16x18Matrix ? 

Danke für die Info Marcus
    uvfzero@gmail.com

dbambus, 2020-12-16:
Grüß dich, 

Ich habe in der Tat vor für die V2.6.0 Mehr Layouts zu Supporten. :-) u.a. das 16x18 und das 18x18 von Microcontroller.de, aber da ich im Moment umziehe wird sich das noch zwei Wochen hinziehen. :-) Ich melde mich sobald ich eine Vorabversion habe.

3damberger, 2020-12-16:
Ja wie geil. Super. Mit nem esp ist s deutlich einfacher wie die version
von microcontroller.net. ich hab schon alles fertig gefräst . wäre schade
wenn ich da keinen quellcode bekomm. man hat halt viel viel mehr
möglichkeiten mit der 24h version. ansonsten ist dein code suuuupi . danke
schonmal .

Eisbaeeer, 2020-12-18:
Layouts lassen sich relativ einfach selbst erweitern bzw. anpassen. Das Layout ist in einem zweidimensionalen Array abgelegt. Du kannst dir ein vorhandenes Layout als Vorlage hernehmen. Das bekommst du sicher hin ;.)

dbambus, 2021-01-08:
Servus @3damberger, 

ich habe mal etwas gebastelt am Code, hier habe einen UhrType für die 24h Uhr erstellt. jedoch muss ich einiges am Unterbau verändern, damit ich auch die Uhrzeit korrekt anzeigen kann. Das werde ich erstmal Debuggen müssen, damit das nicht mit den anderen Uhren kracht. 

Aber fürs erste kann ich dir einen Stand geben, der "funktionieren" sollte. 

[24h_Uhr_Firmware.zip](https://github.com/dbambus/Worduhr/files/5787179/24h_Uhr_Firmware.zip)

Das Sternchen deshalb, da nur die Anzeige in 5 Minuten Schritten funktionieren sollte. Bitte probiere es mal und gib mit ein Feedback. Im schlimmsten Fall ist die Uhr schwarz, im besten Fall kann man sie Eingeschränkt nutzen.

3damberger, 2021-01-08:
Hi grüß dich !

Na das is ja super. Kann ich dann gleich testen mit der vorbereiteten
platine.
Hab heute auch mal versucht den code zu kombilieren. hat auch geklappt.
Aber:

Mqtt bekomm ich ned gespeichert. Er speichert meinen broker nicht und
schaltet mqtt immer wieder selbständig ab.
Hattest du das scho mal ? Die fertige bin ging gar nicht. ;-(

Gruß
   Marcus

dbambus, 2021-01-08:
Hey, 

Bitte MQTT noch nicht verwenden :-), es kommt eine größere Umstellung in der nächsten Zeit. Daher wird sich da noch da ein oder andere Ändern. Ich bin aber dran.

3damberger, 2021-01-08:
Okay. Ich teste und schreib dir. Hast du das layout von mikrocontroller.net
genommen?

dbambus, 2021-01-14:
Hey, 

ich habe in der Tat das Layout von mikrocontroller.net verwendet. In der Zwischenzeit habe ich weiter an dem Unterbau gearbeitet und hier eine neu Vorabversion mit dem neuen Uhrtype UHR_291 erstellt. Hier ist der Link zum aktuellen Entwicklungsbranch. 

https://github.com/dbambus/Worduhr/tree/master_V2.6.0

Hier direkt die Binary:

[firmware.zip](https://github.com/dbambus/Worduhr/files/5814568/firmware.zip)

3damberger, 2021-01-18:
Hi. Habs gerade mal getestet. Es ist halb elf. Die eine LED in der zweiten
reihe von unten leuchtet immer. Warum weiß ich nicht. LED laufen beim Start
alle durch. Und ip wird auch angezeigt.

Gruß Marcus

3damberger, 2021-01-18:
welche sprachoption hast du genommen für das layout bzw die frontplatte von
mikrocontroller.net. vielleicht liegt auch da der fehler

dbambus, 2021-01-18:
Hey, 

ich nehme an, dass es genau die 255ste LED ist. welche leuchtet. Hier hatte ich eine Logik eingebaut, welche bisher geprüft hat, ob der Wert 255 ist oder ein anderer. Quasi ein Abstellgleis für Werte, weil das bisherige LED Maximum bei 242 lag. 

Ich werde aber das System stark umkrempeln, sodass diese 255 unnötig wird. Hier bitte ich noch um Geduld, weil auch noch  das gesamte Wortuhr Repo auf ein neues umgestellt wird, kann ich noch nicht absehen, wann ich wieder an dieser Baustelle weiter arbeite. Aber es ist schön zu sehen, dass es doch funktioniert. 

Eine Bitte habe ich aber noch. Kannst du paar Bilder schicken von der Laufschrift, der Sekundenanzeige und von der Regenbogenfunktion ? Diese würde ich gerne noch debuggen. 

Ich habe das Layout verwendet, welches direkt [hier](https://www.mikrocontroller.net/articles/WordClock_mit_WS2812) abgebildet ist. 

LG
David

3damberger, 2021-01-18:
hi. kein problem . mach ich dir gerne.

Hier hast du mal meine nummer. über whatsapp geht das sicher besser. :
01752716375
Telegram und anydesk habe ich natürlich auch
Mein youtubeKanal : https://www.youtube.com/channel/UCrU2UPiKWtITF7tzeW5y-Dg

Ich ünterstütze dich gerne mit Bildern und videos

Gruß
   Marcus

closed 2023-02-22 by dbambus