ESPWortuhr/Multilayout-ESP-Wordclock #39: 114_Alternative.hpp

Opferwurst87, 2020-12-27:
Moin, da ich anders keine Lösung finde, versuche ich es hierrüber. Ich hoffe du kannst mir da weiterhelfen.
Ich habe die Uhr auf Basis der alten (mittlerweile offline) Version von Ulrich gebaut. Leider ist dort keine Uhrzeit in "ossi"-Zeit möglich, daher bin ich auf deine neuere Version gestoßen, welche das Unterstützt. Jedoch nutzt du eine andere Wort-Matrix, sodass vieles falsch beleuchtet wird.

Ich hab nun die 114_Alternative.hpp so angepasst, das alles korrekt mit meiner Wort-Matrix leuchten sollte, bekomme ich aber absolut nicht hin, die neue Version zum Laufen zu bringen. Alle Schritte in der readme habe ich ausgeführt. NodeMCU 1.0 ausgewählt. Es wird alles erfolgreich mit der Arduino IDE kompiliert (nur eine Warnung mit der Timezone) und hochgeladen. WLAN Verbindung funktioniert und IP-Adresse wird auch bezogen. Versuche ich den WebClient aufzurufen, geht das WLAN aus bzw. der ESP. Wo liegt mein Fehler? Alternativ würde ich mich über eine neue .BIN mit der angepassen Alternative.hpp freuen. Letzteres kann ich anbieten.
Danke!

dbambus, 2020-12-27:
Hallo Grüß dich, 

kannst du mal ein Bild von dem neunen Layout machen und hier reinstellen ? Dazu wäre es auch cool, wenn du deine Version der .hpp Datei hier einfügen kannst, dann kann ich mir das Problem mal anschauen. Welche Version hast du denn für deine Wordclock verwendet (V2.5.0) ? Und Welche IDE hast du verwendet (Arduino oder Platfomio)?

Prinzipiell sind die Schritte die du gemacht hast auch nachvollziehbar und richtig. Mir ist also nicht ganz klar, wo der Fehler liegt und würde es deshalb mal selbst probieren. 

Mit freundlichen Grüßen
David

Opferwurst87, 2020-12-27:
Hey, das "neue" Layout ist eigentlich das urspüngliche, welches auch bei qclocktwo bzw in der 114.hpp verwendet wird - nur bei der Alternative ist halt sowas wie "Happy Birthday" in der Matrix dabei. Die überarbeitete Alternative_144.hpp häng ich hier mal an, die sollte auch zum eigentlichen Layout passen.

[uhr_func_114_Alternative.zip](https://github.com/dbambus/Worduhr-v3/files/5744833/uhr_func_114_Alternative.zip)

Glaube aber nicht, dass es daran liegt. Wenn ich aus deinem Ordner die .ino in die Arduino IDE importiere (inkl. Bibliotheken) und ohne Änderungen neu kompiliere, habe ich das gleiche Fehlerbild. Also liegt der Fehler irgendwo bei mir - mache das auch zum ersten mal.

Ich weiß gar nicht genau, welche Version ich mir vor einem Jahr aus dem Github gezogen hab, kann auch die 2.0 gewesen sein. Funktioniert an sich ja, nur nicht mit dreiviertel/viertel. Spiel ich ne fertige .BIN auf, hab ich keine Probleme.

Wäre cool, wenn die alternative_144.hpp auch zum "eigentlichen Layout" passen würden.

Viele Grüße!

dbambus, 2020-12-27:
Okay, 

soweit verstehe ich, dass du ein layout hast, welches dem vom Original Wordclock Hersteller ähnelt und somit auch Dreiviertel enthält, aber nicht dem 114_Alternative entspricht. 

Ich habe deine .hpp Datei in meinem lokalem master branch (V2.5.0) eingefügt und es konnte ohne Probleme kompilieren. Ich habe die fertige Binary Datei hier:

[firmware.bin.zip](https://github.com/dbambus/Worduhr-v3/files/5745034/firmware.bin.zip)

Kannst du sie mal testen, ob sie bei deiner Uhr funktioniert ? ggf ein Bild des angezeigten Layouts einfügen. 

Mit freundlichen Grüßen
David

Opferwurst87, 2020-12-28:
Moin, also ich habe eben mal die neue Firmware aufgespielt. Egal ob ich 144.hpp, 114_Alternative.hpp oder 144 Clock2 (müsste das neue sein) nutze, ich habe immer die gleiche fehlerhafte anzeige nach dem "falschen" Layout. Wobei es mit der 114.hpp funktionieren sollte (nur halt keine "ossi"-Zeit), habe ich auch dort diese Anzeige:

[image]

Ich raffs einfach nicht...

Opferwurst87, 2020-12-28:
Hey, ich habs eben getestet und muss zugeben, dass ich ziemlich verpeilt bin! Meine Datei war ziemlich falsch, denn ich habe mich an einer Datei (inkl. Matrix) orientiert, die ich aktuell gar nicht nutze. Daher kannst du die obrige Firmware löschen. Anbei die korrekte Fassung der 144_Alternative.hpp im Anhang auf meine Matrix bezogen (entspricht der von qclocktwo).

[uhr_func_114_Alternative_2.zip](https://github.com/dbambus/Worduhr-v3/files/5745544/uhr_func_114_Alternative_2.zip)

Zusätzlich noch ein Bild von meiner Metallplatte mit "falscher" .hpp Datei. 
[image]

(Sorry für das Durcheinander. Nichtsdestotrotz, besteht weiterhin ein Unterschied zwischen deiner 114.hpp und Alternative_114.hpp, da unterschiedliche Matrizes)

Weiterhin weiß ich auch nicht, wo mein Fehler beim kompilieren liegt, wenn du also es nochmal mit der angehängten .hpp kompilieren kannst, wäre ich dir sehr dankbar. Ich hoffe, dass ich diesmal nichts falsch gemacht hab.

Danke!

dbambus, 2021-01-02:
Hallo, 

ich werde mich dem annehmen und ein Release auf Ende der kommenden Woche setzten in dem dieses Layout implementiert ist. 

Mit freundlichen Grüßen

dbambus, 2021-01-04:
Added with https://github.com/dbambus/Worduhr-v3/pull/41

Opferwurst87, 2021-01-28:
Moin, ich kam erst jetzt dazu den neuen Uhrtype zu testen. An sich funktioniert alles, leider wird mir "viertel vor" statt "dreiviertel" im Uhrtype 2Clock.hpp angezeigt.

In den Einstellungen des Uhrtypes, bekomme ich die Möglichkeit auf dreiviertel umzustellen gar nicht. In der Alternative.hpp ist diese Einstellung vorhanden. 

Siehe Screenshots

[image]
[image]

Ich weiß nicht genau wo man es anpassen muss, damit ich die Spracheinstellung auch in der 2Clock.hpp Variante bekomme, vermute aber, dass es in der uhr_func.hpp ab Zeile 889 angepasst werden muss? Vielleicht liege ich damit ja richtig. Wäre cool, wenn du das noch fixen könntest und ein Update (ggf. auch erst mit 2.6.0) zur Verfügung stellen kannst.

Danke!

dbambus, 2021-01-29:
Hallo @Opferwurst87,

ich habe vor paar Commits eine Funktion: 

```
bool definedAndHasDreiviertel() {
    return (G.Sprachvariation[ItIs45] == 1 &&
            G.UhrtypeDef == Uhr_114_Alternative) ||
           (G.Sprachvariation[ItIs45] == 1 && G.UhrtypeDef == Uhr_114_2Clock);
}
```
hinzugefügt, die genau das Prüft. Leider ist die Implementierung davon in der richtigen Stelle verloren gegangen. ich habe sie wieder hinzugefügt. Mit dem nächsten Pull Request sollte es funktionieren. :-)

closed 2021-02-01 by dbambus