ESPWortuhr/Multilayout-ESP-Wordclock #40: Ist die LDR-Empfindlichkeitseinstellung verschwunden?

clingner, 2020-12-31:
Vielen, vielen Dank für die wirklich tolle Weiterentwicklung der Firmware. Ich habe meine Uhr vor ein paar Jahren gebaut und hatte da noch die Version 2.11 von Ulrich drauf. Nun habe ich mir die Software von hier aufgespielt. Was schon mal mit VSCode eine super tolle Sache ist. Gegenüber den früheren Bastelarbeiten mit der Arduino IDE, ein mega Fortschritt!

Die Firmware funktioniert mit meiner alten Uhr recht gut. Doch habe ich in den Einstellungen nur die Option zum Ein- oder Ausschalten des LDRs gefunden. Es fehlt mir die Eingabemöglichkeit des LDR-Kalibrierungswertes, wie in der alten Version. Bei meiner Uhr werden mit dieser Firmware die LED's nur minimal abgedimmt, wenn ich den Sensor komplett verdecke. Das ist in der Nacht noch viel zu hell und blendet.
Gibt es diese Funktion nicht mehr einstellbar? Oder kann ich den Wert irgendwie anders einstellen? 

Zum Konfigurieren der Uhr verwende ich ein Smartphone mit Android Betriebssystem. Als Hardware habe ich einen D1 mini in Zusammenarbeit mit einem Levelshifter welcher mir das Signal auf die 5V statt der 3.3V hebt.

dbambus, 2021-01-02:
Hey clinger, 

die Möglichkeit ist in der Tat verschwunden. Die gesamte LDR Funktionalität hat nach dem Versionssprung auf V2.2.0 nicht mehr funktioniert. Erst in der Entwicklung der V2.5.0 ist das aufgefallen (Ich hatte bis dahin keinen LDR eingebaut). Hier habe ich aber nur die rudimentäre Funktionalität reaktiviert, weil ich mir nochmal Gedanken über die Kalibrierung machen wollte. 

Hier war bei zuletzt unklar, wie der LDR genau eingebaut werden sollte. Ob man eben eine große oder eine kleine Spannung erwarten sollte. Ich wollte hier Klarheit schaffen, indem ich ein Beispiel Aufbau in die Anleitung aufnehme und so Klarheit schaffe. Anschließend kann man die nötige Kalibrierung wieder einbauen. 

Ich würde mir hierzu ein Release auf das Ende der kommenden Woche setzen. :-)

dbambus, 2021-01-04:
Added with https://github.com/dbambus/Worduhr-v3/pull/41

closed 2021-01-04 by dbambus