ESPWortuhr/Multilayout-ESP-Wordclock #46: Erweiterung um animierte Anzeigen

ATho95, 2021-01-16:
Hallo, ich habe vor Jahren schonmal was für die Wortuhr gemacht. Bevor ich das publik machen konnte war das Projekt verschwunden. Bin jetzt auf dieses Repository gestossen. Falls interesse besteht, würde ich meine Änderungen bereitstellen. Da ich jetzt selber den master nutze, habe ich auch Anpassungen an meine Uhren-HW gemacht.
Hier mal ein Überblick:
- Übergangsanimationen
- Silvesterfeuerwerk
- LDR Anpassungen auch für Phototransistor (umgekehrte Kennlinie)
- ntp_flag setzen, wenn das Jahr 1970 ist
- Assert für EEPROM Inhalt zu groß
- Anpassungen für RGB und GRB Leds (ist eher ein Vorschlag, nur die RGB Variante zu nutzen)
- Favicon
- Webserver nur für "/" (verhindert doppelte Übertragung)
- Zeige WiFi conncting phase auf den LEDs
- Vereinfachung für LED Layout 114 (LED_LAYOUT_REVERSE)
- Bugfix für das Radio Button Problem auf der Funktionsseite
- Bugfix in payloadTextHandling()
- Bugfix in set hostname (host-button -> hostname-button)
- SERNR 101 (geänderte Config Struktur)
[image]

dbambus, 2021-01-16:
Hey @ATho95, 

das hört sich sehr sehr gut an, was du da vorschlägst und ich würde es sehr begrüßen, wenn du dem Projekt beiwohnst. Der "master" Branch ist bei mir ausschließlich für die finalen Releases da. Ich selbst entwickle aktiv im Branch "master_V2.6.0". Für genau dieses Release habe ich mir selbst vorgenommen mehr Layouts zu Supporten und Effekte zu integrieren. :-) Also genau das, was du ja schon gemacht hast. 

Kommen wir also nun zum Code selbst. @tali hat schon sehr viel am Unterbau der Wordclock, insbesondere der Website und der Compiler Settings, gearbeitet. Diese Änderungen betreffen auch einige deiner Implementierungsvorschläge (geänderte Configstruktur, bugfix hostname, Wifi connect LEDs, ..) andere Punkte bleiben offen, worüber ich mich sehr freuen würde. 

Ich selbst habe in der letzten Zeit viel am Uhrtype gearbeitet, um einerseits den Speicherverbrauch zu reduzieren und anderseits auch mehr Layouts zu unterstützen. Hier habe ich auch das UHR114_Reverse Layout angerührt, schau es dir einfach mal an :-). 

Schau dir einfach mal den Entwicklungsbranch und die bisherigen Änderungen seit 2.5.1 an und "Give It a Go" :-) bin schon gespannt.

ATho95, 2021-01-18:
Das Mergen hat etwas gedauert, ein Hook passte gar nicht mehr.
Leider ist der Start der FW instabil geworden. Ich vermute, dass das ein out of memory Problem ist. Free heap ist 17808 Byte.
Die FW kommt oft nur bis zum Lauftext und kommt dann auf das hier. Manchmal kommt sie auch darüber und läuft dann.

Exception Cause: 28  [LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads]

0x4023752e: pgm_read_byte_inlined at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/sys/xtensa/sys/pgmspace.h:72
0x4023a70b: _svfprintf_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:667

dbambus, 2021-01-19:
Hey @ATho95,

kannst du den Code auf Github pushen ? Dann könnte ich nachschauen, wo die Fehler herkommen. Eventuell könnte man die Features sequenziell mergen, es dauert zwar etwas, aber so könnte man Fehlerquellen isolieren. 

Hast du zuvor den ESP über das esptool.py komplett gelöscht ? Eventuell auch mal einen hard-reboot über das Abklemmen der Versorgungsspannung machen, nach dem flashen.

ATho95, 2021-01-19:
Ich bin ja ein Github Neuling und hatte so ein paar Probleme:
Line endings aus Github waren Windows like, das ZIP auf dem ich gearbeitet hatte ware Linux like. Ich hoffe beim committen wird das wieder gerade gebogen. Meine Änderungen sind unter @ATho95 branch master_V2_6_0 zu finden.
Die Abstürze von gestern sind nach dem kompletten Löschen des Flashs weg ???
Known Bugs:
- Minuten verschwinden bei Animationen
- Auto LDR funktioniert noch nicht

Wenn Du oder Ihr euch entscheidet, die Animation zu übernehmen, entfällt wahrscheinlich diese Transition in led_set() und ich könnte Änderungen für die known bugs machen. Mindestens der Minutenfehler hat damit zu tun, das show_minuten() in Word_array schreibt, aber Word_array_old an Neopixel übergeben wird.

Die Animation holt alles was es braucht aus Neopixel und schreibt auch wieder darein. Basierend auf den 4 Hooks:
	void begin();
	void loop(struct tm &tm);
	bool led_show_notify(bool flag, uint8_t minute);
	void demoMode(uint8_t &_minute, uint8_t _sekunde);

ATho95, 2021-01-20:
Ich kann die Abstürze jetzt reproduzieren. Bei den Einstellungen setze ich einen Haken bei Neustart - Ip Adresse anzeigen. Nach dem nächsten Reboot kommt er bis zur laufschrift("IP Adresse"). Vermutlich bekommt er da immer einen Interrupt vom IP-Stack und stürzt darin ab.
Das passiert auch in der originalen master_V2_6_0 FW, allerdings nur, wenn ich sie übersetzt habe. Das Release_2.6.0.bin File hat dieses Problem nicht.
Ich nutze die ESP8266 lib 2.7.4

dbambus, 2021-01-22:
Hey,

ich habe nun wieder etwas zeit gefunden und werde mir nun den Code genauer anschauen.

dbambus, 2021-01-23:
Hey @ATho95,

ich habe es im Wesentlichen zum Laufen gebracht, wichtig ist dabei allerdings, dass animation.begin(); nach der WLAN Init ausgeführt wird, da das Programm der WLAN.init über den WiFiManager sehr viel Ressourcen braucht. Deshalb habe Ich habe ich die .begin() nach der OTA Stage platziert, hier wird sie korrekt ausgeführt. Die Zeile mir dem anzeigen des WLAN Logos habe ich aaskommentiert, da sie nicht mehr nötig ist. 

Ich werde in den kommenden Tagen einen Commit mit meinen Änderungen machen, da es noch den ein oder anderen Bug gibt, den ich noch fixen will. 

Aber großes Lob, die Animation Class schaut wirklich gut gemacht aus. :-) Hier könnte man noch an der ein oder anderen Stelle optimieren, um doppelte Variablen zu vermeiden und den Speicherverbrauch reduzieren zu können.

Mit freundlichen Grüßen
David

ATho95, 2021-01-23:
Hallo David,

ich versuche jetzt einfach mal die Adresse aus der Notification.
Bezüglich Speicherverbrauch ist egal wo das animation.begin steht, da
mit dem #include die Klasse instanziiert wird. Ansonsten müsste man im
.begin() instanziieren.

Ich habe aber auch so meine Befürchtungen, dass die Animation auf den
"grossen" Uhren nicht so richtig läuft.

Speicher sparen kann man immer. Ich hatte z.B. ein uint8 wieder lang
gemacht weil der Compiler sagte, der Vergleich ist immer true.
Bei meinen Schwierigkeiten bin ich noch auf das uint8 s[5] gestossen. Da
werden in Zeile 696 in Wortuhr.ini 5 byte + '0' reingeschrieben.
sprintf(s, "rgb%d%d", i, ii);

Ich weiß inzwischen, dass ich nur noch Neopixel und ESP8266WiFi zusammen
binde muss und dann Illegal Instruction kriege. Der Bug muss in den Libs
liegen.
Eventuell liegt das am Stand 3.0.0-dev der ESP8266 SDK. Du hast wohl
eine Version 2.2.2-dev. Wie kommt man daran?
SDK 2.2.1 stürzte ja mit LoadProhibited ab.

Andreas

dbambus, 2021-01-24:
Hey, 

also ich code mit CLion mit dem Platform.io Plugin. Hier kommt es beim Kompilieren zu diesem LOG: 

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.6.2) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Converting Wortuhr.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
```

mit meinem aktuellem Branch `master_V2.6.0` ist es aber noch etwas Buggy, aber es funktioniert.

ATho95, 2021-01-27:
Nachdem ich mich durch den konfusen Stacktrace und die vermeintlich alte SDK Version vollkommen auf dem Holzweg verirrt hatte. Ich habe "meinen" Bug gefunden. War aber auch nicht einfach, da er aus dem Time Callback asynchron auftaucht UND der zeitliche Ablauf UND die DHCP Einstellungen im Router eine Rolle spielen.
Meine Fritzbox übermittelt ihre eigene IP-Adresse als NTP Server. Das führt dazu, dass **sntp_getservername() NULL** liefert, die im printf die Exception auslöst.
Letztendlich kann das jeden treffen.

Ich guck mir die neue Struktur später noch an. Wenn ihr das folende schonmal in Wortuhr.ino merged könnte das so manchen Frust verhindern.
```C++
void time_is_set() {
    time_t utc = time(nullptr);
    if (externalRTC) {
        RTC.adjust(DateTime(utc));
    }

    struct tm tm;
    localtime_r(&utc, &tm);
    _sekunde = tm.tm_sec;
    _minute = tm.tm_min;
    _stunde = tm.tm_hour;
    if (G.UhrtypeDef == Uhr_169) {
        _sekunde48 = _sekunde * 48 / 60;
    }

    if (sntp_getreachability(0)) {
    	char *servername = (char *)sntp_getservername(0);
    	char buf[20];
    	if (servername == NULL) {
    		strcpy(buf, IPAddress(sntp_getserver(0)).toString().c_str());
    		servername = buf;
    	}
        Serial.printf("Set new time: %02d:%02d:%02d (%s)\n", _stunde, _minute,
                      _sekunde, servername);
    } else {
        Serial.printf("Set new time: %02d:%02d:%02d (SNTP not reachable)\n",
                      _stunde, _minute, _sekunde);
    }
}
```

tali, 2021-01-27:
@ATho95 hilft #52 ?

ATho95, 2021-01-29:
> Was mir noch nicht so ganz gefällt, dass bei den Animationen sehr viele Variablen neu angelegt werden, die vorher schon global existierten.

Welche Variablen meinst du? Ich bin kein Freund von globalen Variablen, da sollte man schon genau hinsehen. 

> Aber verstehe ich das richtig, dass es eigentlich nur Transitions zwischen den Uhrzeiten sind ? Also man hat vorher die alte Zeit und nach dem Effekt die neue Zeit ?

Ja, genau. 
Die Info, dass sich etwas am Layout geändert hat wird mitgeteilt. 
Wenn sich lediglich die Farben von Vorder- oder Hintergrund geändert haben ermittle ich das aus dem Neopixel Stripe. Das Layout könnte man genauso aus dem Stripe ermitteln aber in der alten Wortuhr gab es die Info schon. Das entkoppelt die Animation vollkommen von dem anderen Code. Man könnte also auch die Wetterdaten damit animieren.

Ich habe da noch ein Problem mit dem PR #56 den kann ich nicht als branch auswählen. beim #52  ging das noch. Muss der von euch freigegeben werden oder kann ich da was machen?

dbambus, 2021-01-29:
> Welche Variablen meinst du? Ich bin kein Freund von globalen Variablen, da sollte man schon genau hinsehen.

Ich würde mir gerne die Zeit nehmen den Code genau zu verstehen :-) Das wird etwas dauern, wenn man nur hier und da mal Zeit hat zu lesen. Mir ist halt nur aufgefallen, dass die Werte der alten Matrix, der kommenden und der Transitions doppelt vorhanden waren. Aber ich würde mein Fading Animation Code einfach in die Animation.hpp implementieren, dann sollte das wieder passen. 

> Ja, genau.
> Die Info, dass sich etwas am Layout geändert hat wird mitgeteilt.
> Wenn sich lediglich die Farben von Vorder- oder Hintergrund geändert haben ermittle ich das aus dem Neopixel Stripe. Das [...]

Hier müsste ich mir erstmal eine entsprechende Wordclock bauen, ebenfalls auch die 291er. :-), dann kann man das angehen.

> Ich habe da noch ein Problem mit dem PR #56 den kann ich nicht als branch auswählen. beim #52 ging das noch. Muss der von euch freigegeben werden oder kann ich da was machen?

PR #56 habe ich selbst noch nicht freigegeben, da es noch nicht fertig getestet ist. Deswegen ist er noch nicht verfügbar. Du kannst allerdings in meinem Fork in der `main` branch schauen. Das ist der Code.

Es wäre cool, wenn du dir mal de Effekte anschauen kannst. Wenn vorher nämlich keine Hintergrundfarbe eingestellt ist, dann wird beim auswählen der Effekte ein roter Hintergrund gesetzt.

ATho95, 2021-01-29:
> Es wäre cool, wenn du dir mal de Effekte anschauen kannst. 

Ich habe jetzt deinen Code. Wenn du mit Effekten die Animation meinst, sehe ich nichts was rot wird.
Bei den Schiebeanimationen erscheint bei der nächsten Animation erst nochmal der vorherige Hintergrund. Da kann ich mal gucken.

ATho95, 2021-02-01:
Ich habe die Minuten LEDs mit in die Animation genommen und Änderungen an der Konfiguration von der Webpage aus werden sofort angezeigt.
Das Favicon habe ich auch gängig gemacht.
Für den Pull Request kämpfe ich im Augenblick mit dem github lint, da muss ich mein Eclipse noch anpassen.
Also bitte nichts an der Animation ändern.

dbambus, 2021-03-29:
Können wir dieses Issue schließen ?

ATho95, 2021-03-29:
erledigt

closed 2021-03-29 by ATho95