ESPWortuhr/Multilayout-ESP-Wordclock #61: LDR

ATho95, 2021-02-06:
Ich dachte, dass ich durch die LDR Erweiterung mit den 2 Eckpunkten einen Fehler eingeschleust habe. Dem ist wohl nicht so.
In der aktuellen Implementierung wirkt die LDR Steuerung 
- nur auf den Hintergrund
- nur beim Minutenwechsel

Ich fand es besser so wie es vorher war. Vor allem wenn man Kalibrierungswerte ändert, möchte man doch sofort ein Ergebnis sehen.

In den Berechnungen (z.B. rr = G.rgb[position][0] * ldrVal / 100;) können bei uint8 overflows entstehen. Das würde ich auf jeden Fall korrigieren.

dbambus, 2021-03-01:
Danke für den Hinweis @ATho95, kannst du zu dem Issue ein Pullrequest erstellen ?

ATho95, 2021-03-01:
die oben genannten Berechnungen produzieren offenbar keinen Overflow.
PR kommt.

closed 2021-03-02 by dbambus