ESPWortuhr/Multilayout-ESP-Wordclock #67: Was mache ich falsch?

udo1toni, 2021-02-18:
Ich habe hier ein Problem.

Meine Schritte: 
1. Wemos D1 Mini mittels ESP8266Flash.exe mit der aktuellen Version (2.5.1) bespielen. -> Reboot
2. AP Uhr anmelden und 192.168.4.1 aufrufen.
3. WLAN SSID und Passwort eintragen, speichern. -> Reboot (automatisch)
4. Mit der über den Router ermittelten IP verbinden, NTP Server eintragen, LEDStripe auf RGB änden. -> Poweroff
5. Mit Dem Ledstripe verbinden und Power up.
6. Die Punktmatrix zeigt korrekt die IP als Laufschrift an.
7. Die Uhrzeit wird für einen Sekundenbruchteil angezeigt, der Wemos bootet erneut. Nun wiederholen sich die Punkte 6 und 7 bis zum Powerdown.

Die Webschnittstelle ist ab Punkt 6 zu keiner Zeit erreichbar. Auch nach dem Abstecken vom LEDStripe bleibt der Zustand. 
Einzig das Flashen mit einem leeren Image und anschließendes Neu Flashen führend wieder zu einem nutzbaren Chip. Es scheint, als wäre eine gewählte Einstellung dafür verantwortlich. Da ich beim ersten Mal den LED Typ nicht geändert hatte, neige ich zum NTP-Server... Mein NTP Server befindet sich im LAN, der Wemos darf sich nicht mit dem Internet verbinden. Ich nutze allerdings einen DNS Resolver, der die internen Adressen ebenfalls auflöst, in der Form ntpserver.meinlan.lokal. 

Hat jemand Irgendwelche Ideen, was ich falsch mache?

dbambus, 2021-02-25:
Guten Tag, 

entschuldige die späte Antwort, war aber im KurzUrlaub. 

Ich nehme an, dass es sich um einen alten Bug handelt, der in der neueren Entwicklerversion behoben worden ist. Probieren Sie doch mal aus, den Flashspeicher zu löschen

`esptool.py --port COM5 erase_flash`

 und neu zu Flashen. Im Anschluss die Schritte erneut durchführen, jedoch nicht den Zeitserver einzustellen. Hier gab es früher mal einen Fehler, der zu dem beschriebenen Verhalten führen kann.

udo1toni, 2021-02-25:
Die Entwicklerversion muss ich vermutlich selbst bauen? Ist grundsätzlich kein Problem, ist nur etwas her, dass ich das letzte mal für Arduino & Co compiliert habe. 
Der Zeitserver ist schon eine wichtige Funktion :) weil die Uhr keinen Zugriff auf das Internet bekommen soll. Mithin ginge die Anfrage auf den per default hinterlegten Server ins Leere. Je nach Modul kann die interne Uhr schon ordentlich zu schnell/langsam laufen, unsynchronisiert macht das also auf keinen Fall Spaß. 
Und natürlich möchte ich auch mqtt nutzen :) das ist für mich der Hauptgrund umzudaten.

dbambus, 2021-02-26:
Hey, 

ich werde versuchen in den Kommenden Tage die Version 2.6.0 Final zu machen. Hier habe ich mal unkompliziert die Binary für dich gebaut. 

[firmware.bin.zip](https://github.com/ESPWortuhr/Wortuhr/files/6049216/firmware.bin.zip)

udo1toni, 2021-02-27:
Super, probiere ich morgen mal aus. Vielen Dank!

udo1toni, 2021-02-27:
So, ich habe die Datei ausprobiert, funktioniert super. Allerdings gibt es da doch noch en einen oder anderen Fehler...

- Die Anzeige "fünf nach halb" liefert derzeit nur "nach halb", das Wort "fünf" (für die Minuten) fehlt.

Unter Einstellungen: 

- Status -> Verbinden führt zur Anzeige `Seite /? nicht gefunden`

- Zeitserver scheint nur mit IP zu funktionieren (aber vielleicht war ich auch nur zu ungeduldig...)

- Sprachoptionen -> Es fehlt eine Auswahl für "dreiviertel" statt "viertel vor". Das Design der Uhr gibt es her :) und speziell wenn es schon die Auswahl "viertel" statt "viertel nach" gibt, ist der Alternativtext "dreiviertel" zwingend.

Smart Home: 

- Bei Auswahl "MQTT-Daten speichern" erscheint `Seite /?mqtt-state=1 nicht gefunden` - Nach Neuladen der Seite sind die Felder wieder leer und MQTT steht auf OFF.

- Im Zusammenhang mit Firefox **mobile** wird das Menü nicht aufgeklappt. 

- In Safari fehlen im Animationsmenü mehrere Punkte (z.B. "Keine Animation"), andere Punkte sind doppelt angegeben (ich habe allerdings nicht ausprobiert, ob die jeweils gleichen Punkte evtl. zu unterschiedlichen Animationen führen). In Chrome klappt die Bedienung sauber.  

- Die Auswahl des WLAN-Symbols beim Start führt zu einer dauerhaften Anzeige des Symbols. Erst das Umschalten (z.B. Farbwechsel und wieder zurück zu Uhrzeit) führt zur korrekten Anzeige der Uhrzeit. Sollen hier mehrere Optionen gleichzeitig aktiv sein können? zumindest die Auswahl des WLAN-Symbols hat alle anderen Haken wieder entfernt.

dbambus, 2021-03-01:
Hey @udo1toni, 

vielen Dank für die ausführliche Fehlerbeschreibung. 

> Die Anzeige "fünf nach halb" liefert derzeit nur "nach halb", das Wort "fünf" (für die Minuten) fehlt.

Dieses Issue habe ich soeben mit dem https://github.com/ESPWortuhr/Wortuhr/pull/65/commits/9cf7769c9fe8451864d7f4e7b9655141d73ff50e behoben. 

> Status -> Verbinden führt zur Anzeige Seite /? nicht gefunden

@ATho95 Kannst du dich dieses Issues annehmen ? 

> Zeitserver scheint nur mit IP zu funktionieren (aber vielleicht war ich auch nur zu ungeduldig...)

@tali Hast du vielleicht eine Idee, woran das liegen könnte ?

> 
Sprachoptionen -> Es fehlt eine Auswahl für "dreiviertel" statt "viertel vor". Das Design der Uhr gibt es her :) und speziell wenn es schon die Auswahl "viertel" statt "viertel nach" gibt, ist der Alternativtext "dreiviertel" zwingend.

Dieses Issue bin ich auch mit https://github.com/ESPWortuhr/Wortuhr/pull/65/commits/7c234a275bf08d2fade82dd5a4b043d78de3205d angegangen und habe es behoben.

> Bei Auswahl "MQTT-Daten speichern" erscheint Seite /?mqtt-state=1 nicht gefunden - Nach Neuladen der Seite sind die Felder wieder leer und MQTT steht auf OFF.

Das werde ich mir genauer anschauen. :-) 

> Die Auswahl des WLAN-Symbols beim Start führt zu einer dauerhaften Anzeige des Symbols. Erst das Umschalten (z.B. Farbwechsel und wieder zurück zu Uhrzeit) führt zur korrekten Anzeige der Uhrzeit. [...]

@ATho95 oder @tali, Kannst du dir das evtl auch anschauen ?

dbambus, 2021-03-01:
>Bei Auswahl "MQTT-Daten speichern" erscheint Seite /?mqtt-state=1 nicht gefunden - Nach Neuladen der Seite sind die Felder wieder leer und MQTT steht auf OFF.

Dieses Issue habe ich mit https://github.com/ESPWortuhr/Wortuhr/pull/65/commits/f24df8c382992d73d47d49ba64016a9cad9735df behoben. Hier sollte man allerdings erwähnen, dass die MQTT Funktionen in der Theorie funktionieren sollten, aber bisher noch nicht ausreichend getestet wurden. Hier erhoffe ich mir mal Feedback, werde aber selbst mal einen Server aufsetzten und mal testen (müssen).

Hier habe ich mal die aktuelle Version mit den neuen Änderungen erstellt. 
[firmware.zip](https://github.com/ESPWortuhr/Wortuhr/files/6061338/firmware.zip)

ATho95, 2021-03-01:
>    Status -> Verbinden führt zur Anzeige Seite /? nicht gefunden
@ATho95 Kannst du dich dieses Issues annehmen ?

habe ich beseitigt, PR folgt

> Die Auswahl des WLAN-Symbols beim Start führt zu einer dauerhaften Anzeige des Symbols. 

Das habe ich auch gelegentlich. Das WLAN Symbol bleibt offenbar bestehen, wenn keine Daten vom Zeitserver kamen.
Dass ich keine NTP Daten bekomme hatte ich mal mit Wireshark nachverfolgt. Bei mir gab es einen ICMP Error Response von meinem DSL Provider. Seitdem nehme ich meine Fritzbox als NTP Server.
Btw. den WLAN Traffic zum Internet kann man in der Fritzbox (http://fritz.box/html/capture.html) mitschneiden und mit Wireshark angucken.

>In Safari fehlen im Animationsmenü mehrere Punkte (z.B. "Keine Animation")

Habe ich auch im Firefox schon gesehen. Shift F5 (am cash vorbei neu laden) hilft hier.
Wenn das bei Safari nicht hilft, müsste man mal gucken ob Safari das Füllen der drop down Liste per Javascript nur fehlerhaft unterstützt. Aber das kann ich mir eigentlich nicht vorstellen.

ATho95, 2021-03-01:
@udo1toni, zum ursprünglichen Problem mit den endlos restarts.
Da gab es mal einen Fehler in time_is_set(). Wenn du da ab "String origin" alles bis zum Ende der Funktion auskommentierst, könnte man ein ähnliches Problem schonmal ausschließen.

Ansonsten würde ich den Stacktrace analysieren.
Ob das wie hier erklärt funktioniert weiß ich nicht, da ich Eclipse nutze.
https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

Aus der Kommandozeile geht es auf jeden Fall. Das Ergebnis ist allerdings manchmal sehr schwierig zu interpretieren.
java -jar EspStackTraceDecoder.jar C:\Users\Andreas\git\arduino-esp32\tools\xtensa-esp32-elf\bin\xtensa-esp32-elf-addr2line.exe ..\ESP8266_Wortuhr\Release\ESP8266_Wortuhr.elf stack.txt
In stack.txt steht dann natürlich der stack trace

udo1toni, 2021-03-02:
Puh, ich bin ja eher dummer Endanwender ;) so ein paar kleine Sachen zusammenzimmern hab ich schon mal hinbekommen, dies hier ist definitiv außerhalb meiner Skills ;)
Als NTP Server hatte ich meinen OpnSense Router eingetragen, da das IoT WLAN keinerlei Zugriff auf das Internet hat (aus Gründen...) Der OpnSense liefert die Uhrzeit auch recht zuverlässig aus, im IoT WLAN habe ich ca. 20 Teilnehmer, die größtenteils mit Tasmota 8.5.2 laufen, da gab es bisher noch nie Probleme.  Mit der aktuellen Firmware (sowohl die vom 27. als auch die von gestern) tritt der zyklische Neustart nicht mehr auf

Wie schon erwähnt, kann es gut sein, dass ich bei der Nutzung des fqdn im NTP Server Feld einfach etwas ungeduldig war, mit der IP funktioniert es jedenfalls zuverlässig.

Ansonsten bin ich mit dem Stand jetzt schon sehr happy. Die Auswahl für die Sprachvariante "viertel vor" bzw. "dreiviertel" steht allerdings nach wie vor nicht zur Verfügung. Den Sourcecode habe ich noch nicht verstanden ;) aber wegen der Änderungen im Code, als Uhrenvariante habe ich "Uhr 114" ausgewählt. Das funktioniert auch soweit korrekt. Kann es daran liegen, dass die entsprechende Auswahl nicht erscheint?

Was mir noch aufgefallen ist: Es gibt ja die Option, die Helligkeit abhängig von der Uhrzeit zu steuern. Das funktioniert bei mir irgendwie nicht. In der Uhr ist kein LDR verbaut, und da ich die Uhr nicht selbst gebaut habe, möchte ich hier nur sehr ungern eingreifen (es gäbe auch gar keine Stelle, wo ich den Sensor unauffällig platzieren könnte). 
Nun gibt es ja drei Felder, die sich mit der Displayhelligkeit beschäftigen (Helligkeit, Automatische Helligkeit und Einstellung des LDR). Liegt es also vielleicht einfach daran, dass ich nur einen Fehler bei der Einstellung habe? Ich hatte nur unter Helligkeit für die entsprechende Uhrzeit den Prozentwert eingetragen, das hatte aber auch nach mehreren Minuten keine Auswirkung, Oder wird die Helligkeit nur exakt zum angegebenen Zeitpunkt gesetzt, aber nicht, wenn man den Parameter ändert?

Die Aktivierung von MQTT klappt leider auch noch nicht, zwar kommt nun keine Fehlermeldung mehr, aber nach dem Neuladen der Seite sind die eingetragenen Daten (Serveradresse, Port und auch Zustand des Schalters) wieder leer bzw. OFF.

ATho95, 2021-03-02:
> Was mir noch aufgefallen ist: Es gibt ja die Option, die Helligkeit abhängig von der Uhrzeit zu steuern. ...

Die zeitabhängigen Werte werden mit jedem Minutenwechsel übernommen. Aber nur, wenn Auto-Helligkeit "an" oder LDR-Steuerung "aus" ist.
@dbambus, ich habe gerade mit den Auto-Werten gespielt. Wenn beide Parameter gleich sind, gibt's Division durch 0. Ich packe die Änderung noch in den letzten PR.
Und die LDR-Steuerung kann ausgebaut werden. Dazu solltest du die Autohelligkeit mal testen. ich habe einen Fototransistor verbaut. Der hat eine umgekehrte Kennlinie zum LDR. D.h. mit der LDR-Steuerung ist meine Uhr im Dunkeln hell.
Ich könnte das übernehmen.

udo1toni, 2021-03-03:
Was macht denn die Autohelligkeit genau, ist das ein Teil von LDR?

ATho95, 2021-03-03:
Im Prinzip machen beide das gleiche, wobei die Autohelligkeit flexibler ist. Deshalb könnte die LDR Steuerung entfernt werden.
Entscheidend für die Helligkeit der LEDs ist ja die Eingangsspannung am ESP. Die kann, je nach Beschaltung, im Dunklen höher oder niedriger als im Hellen sein.
Ich habe die Uhr damals gebaut und erst danach eine Software gesucht. Da hatte ich schon einen Fototransistor statt LDR verbaut und die LDR Steuerung funktionierte damit verkehrt herum. Im Dunklen waren die LEDs am hellsten.

closed 2021-03-29 by dbambus