ESPWortuhr/Multilayout-ESP-Wordclock #75: mqtt lässt sich nicht aktivieren

udo1toni, 2021-03-24:
Mit dem aktuellen Release V2.6.0 vom 24.03. 2021 ist es nach wie vor nicht möglich, einen  mqtt Server einzutragen. Stand jetzt steht der Schalter zum Aktivieren dauerhaft auf ON, 
Beim Speichern der Einstellungen werden sowohl die IP als auch der Port auf 1 gesetzt. 

Der Port sollte als default auf 1883 stehen (das ist der well known Port für mqtt).

dbambus, 2021-03-29:
Hallo @udo1toni, 

ich habe mich dem Problem mal mit https://github.com/dbambus/Wortuhr/commit/6f9c414583ddb15d1ec55ad8d79307c04c0334b6 angenommen. Ich werde das mal weiter testen, bis es wirklich funktioniert.

dbambus, 2021-03-30:
Ich habe das Problem mittlerweile näher eingegrenzt und nun sollte die Sicherung der Config Values wieder über die Website gehen. https://github.com/dbambus/Wortuhr/commit/3d03f8f6b0e301b798f77360791091afafae4d9a

Ich habe dazu mal einen Build als Binary gebaut. Könntest du Ihn bei Zeiten in deinem Setup testen?

[firmware.bin.zip](https://github.com/ESPWortuhr/Wortuhr/files/6231906/firmware.bin.zip)

udo1toni, 2021-04-07:
Ich kann bestätigen, dass die MQTT Einstellungen nun zum einen gespeichert werden und zum anderen der mqtt Client sich mit dem Broker verbindet - allerdings tut sich ansonsten nichts. :) 

Den Versuch, ein passendes JSON Objekt zu senden, quittiert wie Wortuhr zwar mit einer Debug Meldung, die Anzeigt springt aber nicht entsprechend um. Nun ja, ist noch experimentell :)

Ein, zwei Gedanken zu mqtt:

Wünschenswert wären Eingabefelder für User, Passwort, ClientID und root Topic. Das root Topic sollte nicht mit einem Slash beginnen, das ist zwar möglich, aber allgemein in mqtt nicht erwünscht. Es ist wichtig, das root Topic frei setzen zu können (mindestens sollte es dem Hostname folgen), da man sonst nicht mehrere Wortuhren mit dem selben mqtt Broker unabhängig voneinander betreiben könnte.
Es sollte unterhalb des Root topic mehrere Untertopics geben, z.B. set (das ergäbe z.B. Wortuhr/set als full Topic), 
Ein LastWillTopic wäre ebenso nützlich, damit können dann alle mqtt Teilnehmer erkennen, ob die Wortuhr Online oder Offline ist.

Da der Lauftext über den Effekt aktiviert werden kann, wäre es auch nett, den Lauftext über mqtt setzen zu können (sinnvollerweise ebenfalls über ein eigenes Topic). Der Client kann einfach alle Topics unterhalb des root Topic subscriben (mittels Wortuhr/#), unterschiedliche Topics tragen also bis auf den zusätzlichen Code für zusätzliche Funktionen nictht weiter auf.

dbambus, 2021-06-02:
Closed, weil MQTT lässt sich jetzt aktivieren. Neues Issue mit Enhancement wünschen wurde erstellt. Siehe https://github.com/ESPWortuhr/Wortuhr/issues/106

closed 2021-06-02 by dbambus