ESPWortuhr/Multilayout-ESP-Wordclock #76: Anzeieoption "dreiviertel" nicht vorhanden

udo1toni, 2021-03-24:
Mit dem aktuellen Release 2.6.0 vom 24.03.2021 steht die Option für dreiviertel nicht zur Auswahl (Uhrenlayout Uhr 114)
[image]

dbambus, 2021-03-29:
Hallo @udo1toni, bei dem UHR_114 Layout ist auch kein dreiviertel in der Matrix vorhanden. Welches Layout verwenden Sie genau? Ist es für Sie möglich ein Bild zu machen ?

udo1toni, 2021-04-14:
Ja, stimmt, das habe ich inzwischne auch festgestellt. Aber das Layout beinhaltet durchaus dreiviertel als Begriff, es ist lediglich nicht in der Matrix definiert. Ich habe inzwischen ein Update hochgeladen.

closed 2021-04-14 by udo1toni

dbambus, 2021-04-29:
In dem aktuellen `main` branch sollte der Support vollständig enthalten sein :-)