ESPWortuhr/Multilayout-ESP-Wordclock #77: Kein Zugriff auf Webinterface

Opferwurst87, 2021-03-24:
Grüße,

ich habe vorhin das Update auf 2.6.0 durchgeführt und komme seitdem nicht mehr über die IP-Adresse auf die Weboberfläche zum Konfigurieren. Mit der Version 2.5.1 war noch alles ok. Als erstes habe ich das Update über die IP-Adresse:81/update hochgeladen und im nächsten Versuch die .bin nochmal manuell über USB-Schnittstelle aufgespielt.

Nach dem flashen via USB kam dann immerhin das WLAN "Wortuhr_verbinden" oder so ähnlich. Gesagt, getan. Die Konfig öffnet sich und ich kann dort meine WLAN Daten eintragen (auch die Erase-Funktion habe ich probiert). Dannach scheint alles i.O. jedoch kann ich weiterhin nicht drauf zugreifen. 

Hier der Auszug aus dem seriellen Monitor der Arduino IDE nach erfolgreicher Konfig (WLAN-Kennwort entfernt).

`Begin Setup
--------------------------------------
--------------------------------------
ESP Uhr
Version         : 2.6.0
Chip ID         : 00BF92FF
Flash ID        : 00164020

CPU Speed       : 80 MHz 

Flash real Size : 4096 KByte
Flash ide  Size : 4096 KByte
Flash ide Speed : 40000000

Free Heap Size  : 26464 Byte
Sketch Size     : 594432 Byte 
Free Sketch Size: 2547712 Byte 

SDK Version     : 2.2.2-dev(38a443e)
RESET Info      : External System
COMPILED        : Mar 24 2021 08:59:29
LED Init
No external RealtimeClock found
*WM: [1] AutoConnect 
*WM: [3] setupHostname:  uhr
*WM: [2] Setting WiFi hostname 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: FRITZ!Box 7490
*WM: [3] Using Password: "Kennwort"
*WM: [3] WiFi_enableSTA enable
*WM: [3] enableSTA PERSISTENT ON 
*WM: [3] Mode after delay:  STA+AP
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
no RTC clock - disable SNTP startup delay
*WM: [2] Connection result: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.178.21
*WM: [3] hostname: STA:  uhr
Mode: STA+AP
PHY mode: N
Channel: 9
AP id: 0
Status: 5
Auto connect: 1
SSID (14): FRITZ!Box 7490
Passphrase (18): "Kennwort"
BSSID set: 0
Signal strength: 100

-- Begin WlanStart -- 
IP:192.168.178.21

-- Ende  WlanStart -- 

Websockest started
--------------------------------------
Ende Setup
--------------------------------------

Set new time: 23:45:22 (europe.pool.ntp.org)
1970-01-01 01:00:04 (+0100)
2021-03-24 23:45:23 (+0100)
...`

Die IP-Adresse 192.168.178.21 lässt sich auch nicht anpingen. In der Fritzbox wird mir das Gerät angezeigt (Reboot usw. durchgeführt). Jedoch ist es nach gewisser Zeit auch dort nicht mehr in der Geräteliste der verbundenen Geräte aufgeführt.
[image]

Im WirelessNetworkWatcher von Nirsoft wird mir das Gerät jedoch nicht angezeigt - alle anderen Geräte im Netzwerk schon
[image]

Jemand eine Idee, wo der Hund begraben ist?

ATho95, 2021-03-27:
Ich würde dir raten, einen Paketmitschnitt aus der Fritzbox heraus zu machen.
[http://fritz.box/html/capture.html](url)
Neben "wlan" den **Start** Button drücken. Es wird nach einem Filenamen auf dem PC gefragt. Wenn man einige Daten gesammelt hat, drückt man auf den **Stop** Button. Das File kann mit Wireshark analysiert werden.
Wenn du dich mit Netzwerken nicht so auskennst, kannst du mir den Trace auch schicken.

Das Layout kannst du unter den Einstellungen ändern.

Opferwurst87, 2021-03-28:
Moin, im Anhang ein kurzer Mitschnitt nach dem Herstellen der Stromversorung der Uhr. Wäre cool, wenn du es dir anschauen könntest. Die Auswertung in Wireshark sagt mir nämlich nichts.

[iad-if-wlan_28.03.21_1521.zip](https://github.com/ESPWortuhr/Wortuhr/files/6217373/iad-if-wlan_28.03.21_1521.zip)

ATho95, 2021-03-28:
Deine Fritzbox gibt dem ESP keine IP Addresse. (Stichwort DHCP)
Der ESP nimmt sich einfach die 192.168.178.21. Es kommt von der Fritzbox auch die Frage: Wer hat die 192.168.178.21 , die der ESP auch beantwortet. 
Als nächstes guckst du mal in der Fritzbox, ob der DHCP Bereich für die Vergabe von IP Addressen groß genug ist.
Das ist etwas versteckt unter: Heimnetz -> Netzwerk -> Tab Netzwerkeinstellungen -> weitere Einstellungen -> IPv4-Konfiguration: DHCP-Server vergibt IPv4-Adressen von ... bis ...
Und kontrollier mal, ob die .21 schon an andere Geräte vergeben wurde.

Opferwurst87, 2021-03-28:
Grüße,
also die .21 taucht nur in Verbindung mit dem ESP auf. Der DHCP in der FritzBox steht auf .20 - .200 - wobei die .20, .22 und .23 bereits vergeben sind.
Im Ergeignislog der FB erscheint nach der WLAN Konfig des ESP ein Eintrag für die Anmeldung, nach 30sek ein Eintrag für die Abmeldung. Teilweise erscheint auch der Hinweis, der WLAN-Key sei falsch. Den hab ich jedoch 100% richtig eingebenen. Auch mit deaktivierter Verschlüsselung kommen die Hinweise mit der an- und abmeldung.
Den dazwischengeschalteten FritzRepeater hab ich auch bereits vom Strom genommen und das alles ohne den Repeater versucht.
Alle inaktiven Geräte hab ich aus der Geräteliste entfernt (auch den ESP mit der .21) Im nächsten Versuch ist das Ding jedoch wieder mit der .21 "verbunden".
Mit der Version 2.5.1 hab er noch die .78 bezogen.

Wird denn der ESP mit dem Neuaufspielen einer .bin kompett resettet oder muss man da was anderes machen?

Ich glaube eher weniger, dass es mit der FB zu tun hat, werde es jedoch m Wochenende mit einem anderen Router ausprobieren können.

ATho95, 2021-03-29:
Wenn dein WLAN Passwort falsch wäre, sähe man im Trace den Versuch eine IP Addresse zu bekommen nicht.
Es gibt einen kleinen Unterschied zwischen deinem und meinem ESP im ersten DHCP Protokoll Paket. Eventuell hat der WiFiManager sich da was gemerkt, das du nur über ein komplettes Löschen des Flash weg bekommst.
esptool.py erase_flash
Oder aus Arduino heraus: Werkzeuge -> Erase Flash -> All Flash Contents

Opferwurst87, 2021-03-29:
Ich habe mal den DHCP Bereich der FB auf .30 - .200 geändert. Nach der Neueinrichtung des ESP, bezieht er die 192.168.178.30, aber auch darüber bekomme ich keinen Zugriff. 
Dann hab ich versuch am Smartphone einen Hotspot einzurichten. Ich hab den ESP dann damit verbunden - er bezieht dann die Adresse 192.168.122.213 - auch hier kein Zugriff über die IP Adresse auf die Konfig.
In der Arduino IDE hab ich es auf "Alle Flash Contents" umgestellt - damit das funktioniert muss ich dann was tun? Ein beliebiges Image aufspielen? So hab ich es zumindest jetzt versucht - alles ohne Erfolg.

Später werde ich es ggf. mal mit esptool.py probieren.

ATho95, 2021-03-29:
Ich habe mir den kleinen Unterschied (Bootp flags: 0x8000, Broadcast flag (Broadcast)) mal genauer angesehen. Dieses Flag sorgt u.a. dafür, dass die DHCP Antworten von der FB als Broadcast gesendet werden. Da ich einen Filter benutzt habe, der nur Pakete von/an deine MAC Adresse sichtbar macht, hatte ich die Broadcasts nicht gesehen. Die IP Addressvergabe läuft also doch, da muss ich meine obige Aussage zurücknehmen.
Ich gucke Morgen mal weiter.

ATho95, 2021-03-30:
Ich dachte, dass ich über den Vergleich mit meiner Uhr noch etwas herausfinde. Ich sehe bei mir keine MDNS Pakete. Bei dir werden diese Pakete relativ häufig gesendet.
Ohne MDNS funktioniert bei mir das Ansprechen über den Hostnamen auch nicht mehr. Nutze ich eh nicht, darum habe ich auch nicht gemerkt seid wann das nicht mehr geht.
Wenn du mal ein von mir übersetztes Binary ausprobieren möchtest, hänge ich hier gern mal eins an.

Ansonsten fällt mir dazu nichts mehr ein.

Ich habe einige ESPs (32 und 8266) im Dauereinsatz. Die waren nach Monaten auch nicht mehr erreichbar. In der FB war auch nur noch zu sehen, dass sie abgemeldet wurden. Ich habe damals ein ping auf meine FB eingebaut. Wenn der mit Fehlern zurückkam booten die ESPs neu. Das hilft bei dir nicht, aber es zeigt, dass es da wohl noch Probleme gibt.

Opferwurst87, 2021-03-30:
Hey danke für die Mühe, ich probiere gerne dein Binary aus, vielleicht funktioniert es ja! Zur Not spiel ich mir halt einfach wieder die Version 2.5.1 auf.

ATho95, 2021-03-30:
Na dann versuchs mal hiermit: 
[ESP8266_Wortuhr.zip](https://github.com/ESPWortuhr/Wortuhr/files/6231917/ESP8266_Wortuhr.zip)

Ich habe übrigens auch die FB 7490

Opferwurst87, 2021-03-30:
Hey, habs probiert, jedoch ist es das gleiche Fehlerbild...

Opferwurst87, 2021-04-03:
Grüße, ich habe nun eine komplett anderen ESP mit 2.6.0 bespielt und mit einem anderen Router (Speedport W724V) verbunden. Auch hier das gleiche Problem. Wird im Log angezeigt, dass das Gerät verbunden ist und kurze Zeit später, ist es wieder raus. Einfach kein Zugriff auf das Webinterface. Bei dem ESP war auch vorher die Version 2.5.1 drauf.

Kein Plan mehr, was ich sonst noch probieren kann.

ATho95, 2021-04-03:
Nur so eine Idee. Es könnte an der Versorgungsspannung (Netzteil und/oder Kabel) liegen wenn der WiFiManager vielleicht an der Sendeleistung dreht.

Opferwurst87, 2021-04-03:
Hmmm, sollte das dann nicht über den seriellen Monitor zu sehen sein?
Ich habe jetzt nochmal einen jungfräulichen ESP genommen und mit dem ESP flasher die Version 2.6.0 aufgespielt. Diesen dann auch nur über den USB Port von PC betrieben und nicht wie bei den anderen beiden über das 5V Netzteil. Wobei, soweit ich weiß wird der USB Port auch mit einer 5V Ausgangsspannung betrieben, oder? Wie dem auch sei... gleiches Problem.

dbambus, 2021-05-11:
Hat sich das Problem mittlerweile lösen können ?

Opferwurst87, 2021-05-12:
Ja, indem ich weiterhin 2.5.1 nutze. Alles andere funktioniert einfach nicht. Vielleicht liegt es auch an der Art und Weise des aufspielens der BIN. Gibt's noch ne andere Möglichkeit als der ESP flasher? Mit der arduino IDE konnte ich die 2.6.0 auch nicht kompilieren. Zuviele Fehlermeldungen mit denen ich nichts anfangen kann.

dbambus, 2021-05-12:
Du kannst es über ein OTA Update versuchen. <IP-Adresse der Wortuhr>:81 hier wäre ein Webupdater hinterlegt. In welchem du eine Binary hochladen kannst. 

Du kannst dabei diese Version verwenden. Hier habe ich einzig die Seriennummer geändert, sodass der EEPROM mit neuen Werten überschreiben wird. 

[2.6.0.zip](https://github.com/ESPWortuhr/Wortuhr/files/6466191/2.6.0.zip)

ATho95, 2021-05-12:
In der Arduino IDE kann man unter Werkzeuge -> Flash Erase: "Scetch + WiFi Settings" einstellen. Das würde ich mal probieren.
Wenn die 2.6.0 nicht kompiliert kannst das nicht benutzen sondern musst von Hand das esptool aufrufen.
Das müsste sinngemäß etwa so aussehen:
C:\Users\Andreas\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1/python3 C:\Users\Andreas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py --chip esp8266 --port COM6 --baud 3000000 **erase_region 0x3FC000 0x4000** --before default_reset --after hard_reset write_flash 0x0 C:\Users\Andreas\AppData\Local\Temp\arduino_build_337571/ESP_Telnet.ino.bin
Wie die Pfade bei dir gesetzt sind kannst du im IDE Status Fenster sehen wenn du die 2.5.1 flash'st

Opferwurst87, 2021-05-20:
Ich werde es demnächst nochmal über die OTA Funktion versuchen. Beim ersten Mal hatte ich damit auch keinen Erfolg. Vielleicht klappt es mit der geänderten Variante.

Das mit Flash Erase: "Scetch + WiFi Settings" habe ich bereits probiert, aber auch hier ohne Erfolg. 
Die letzte Option werde ich bei Bedarf auch nochmal testen. Ich melde mich dann, habe nur aktuell keine Zeit dafür. 
Danke für eure Hilfe!

closed 2021-06-02 by dbambus