ESPWortuhr/Multilayout-ESP-Wordclock #79: Structs ?

dbambus, 2021-03-31:
Hallo @ATho95 @tali , 

ich habe etwas in meinem Coding Styleguide für C++ reingeschaut und paar Ideen auf geschnappt. Was hält Ihr davon, wenn man die spezifischen Variablen vom Global Struct nochmals in einzelne Structs unterteilt. So könnte man den Speicherverbrauch im EEPROM reduzieren.

Ich habe hierzu mal einen Branch erstellt und ein paar Variablen mal gruppiert. https://github.com/dbambus/Wortuhr/commit/b60f69e3a2836ee78c5943097e99b72515bd47d3

Der Speicherverbrauch im RAM bleibt dabei gleich, im hinterlegten EEPROM wird er kleiner, aber der verbrauchte Programmspeicher steigt um 1 kB. Das kann ich mir bisher noch nicht erklären. 

Was gefällt euch besser ?

tali, 2021-03-31:
Das `__attribute__((packed))` packt die Bytes näher zusammen, braucht dann aber bei allen Zugriffen extra Code um das wieder ein-/auszupacken. Vermutlich kommt da der erhöhte Programmspeicher her. Langsamer wird es dadurch vermutlich auch.
Ist denn der EEPROM Speicherverbrauch so kritisch?

Eine gängige Praxis ist auch, nur wirklich im persistenten Medium zu packen, das dann beim Booten einmal in den RAM zu entpacken und dort nur noch mit einer normalen Struktur zu arbeiten.

Abspeichern muss ja auch gar nicht unbedingt in eine gepackte Struktur geschehen. Ich bin im Augenblick eher geneigt, in die andere Richtung zu gehen: Konfiguration als JSON serialisieren und das abspeichern. Gerne auch in ein Dateisystem.
Das hätte auch den Vorteil, dass wir später leichter neue Variablen einführen können, ohne explizit verschiedene Versionen vom Speicherlayout unterstützen zu müssen.

dbambus, 2021-03-31:
Hey, 

ja das mit JSON hatte ich mir auch schon überlegt. :-) Das mit dem Dateisystem ist aber eine schwierigere Angelegenheit, da man das nicht einfach in einer Binary auf ein bestehendes System bringen kann. Eine Möglichkeit wäre eine Art Textfeld auf eine Firstboot Page einzurichten und dort ein JSON einzufügen, das alle persistente Config enthält. Das war letztlich durch https://github.com/ESPWortuhr/Wortuhr/issues/27 gemeint.

Back to Topic. Speicherverbrauch im EEPROM ist nicht all zu wichtig, aber ein Faktor. Wenn ich mich recht erinnere gibt es ein Maximum für OTA Updates, was letztlich die möglichen neunen Features limitiert. Deshalb wollte ich mal schauen, was möglich wäre, um den bestehenden Code zu verbessern.
Mehr Bitweises Arbeiten habe ich da in den Vordergrund gestellt. :-) Da habe ich aber noch ein Paar Verbesserungsmöglichkeiten ^^