ESPWortuhr/Multilayout-ESP-Wordclock #91: Doppelter Code ?

dbambus, 2021-04-29:
Hallo @ATho95,

ich habe mal in dein Animation Code reingeschaut und da entdeckt, dass noch auf sehr viele #define verwiesen wird. Kann man diese Deklarationen so anpassen, dass sie auch für die anderen Layouts funktionieren ? Also auch das 24h Layout mit seinem 18x16 Array ? 

Man kann über usedUhrType->COLS_MATRIX() und ROWS_MATRIX() die aktuell verwendete Anzahl herausfinden und evtl über einen Klassenkonstruktor setzten, oder ? 

Ziel wäre eine weitestgehende Seperation von Uhr-/AnimationCode zum Layout der Uhr.
```
#define ROW_START 0 /* Startindex der Reihe */
#define MAX_ROWS 10 /* Anzahl der Reihen (Y-Richtung) */
#define COL_START 0 /* Startindex der Spalte */
#define MAX_COL 11 /* Anzahl der Spalten (X-Richtung) */

#define MAX_BAELLE MAX_COL
#define MAX_ANIM_PERIOD 5
#define MAX_RANDOM 10
#define STRIPE NULL
```

ATho95, 2021-04-29:
@dbambus, klar das ließe sich machen.
Das Feuerwerk könnte man in die Mitte schieben.
Eventuell wird der Speicher knapp. Das könnte man ja leicht testen, wenn man die #defines ändert.
Der Konstruktor wäre geeignet. Dann muß aber auch im Destruktor der Speicher wieder frei gegeben werden (race condition ?). Beim Ändern des Uhrentyps wäre ein Reset die beste Option (kein fragmentierter Speicher).
ROW_START und COL_START hatte ich ursprünglich angedacht für Uhren mit "LED-Rahmen". Ich weiß nicht wie akut das noch ist.

Btw, ich habe in der Zwischenzeit noch eine neue Animation: Schlange.

dbambus, 2021-04-29:
Das mit dem Speicher sollte schon klappen. Ich habe das mal mit anderen define Werten probiert. 
Ein Rest bietet sich dann an, aber man könnte ja immer mit den größt möglichen Werten initialisieren, darstellen muss man ja nur die relevanten Bereiche der Matrix. 

Das mit dem Rahmen bin ich noch am evaluieren. Aktuell stelle ich mich ein anderes Design mit der Rahmenoption vor in welcher die zusätzlichen LEDs an die Minuten LEDs konkateniert werden. das macht den Code dazu wesentlich einfacher. 

Generell würde ich die UHR_169 mit der derzeitigen Verschaltung als deprecated ansehen. Zumal dazu nicht mal eine Buchstaben Matrix als .svg vorliegt.

ATho95, 2021-04-29:
meld dich nochmal, wenn das mit dem Rahmen geklärt ist. Im Augenblick ist meine Zeit schon verplant.

dbambus, 2021-05-11:
Ich habe mal die PR gemerged. Jetzt kannst du deinen erstellen, der verbleibende ist ja nur eine geänderte SVG Datei. 

Die weiteren Fixes sollten auch kein Problem sein. Ich schaue mir das einfach im PR an.