ESPWortuhr/Multilayout-ESP-Wordclock #92: Fehler beim kompilieren - WebPageContent.gen.inc

hohoho-ia, 2021-04-29:
Hallo,
ich versuche gerade die Version 2.6 auf meinen ESP8266 aufzuspielen.
Leider erschein beim kompilieren immer der Fehler: **WebPageContent.gen.inc: No such file or directory**
in der Datei WebPageContent.cpp.
Leider bin ich in dem Thema nicht wirklich fit, kann mir hier jemand helfen, Danke!

dbambus, 2021-04-29:
Hallo @hohoho-ia, 

Kannst du deine Entwicklungsumgebung mal beschreiben ? (VS Code ?) 
Welche Platform.io Version verwendest du ? 

LG
David

hohoho-ia, 2021-04-29:
Hallo David,
erst mal Danke für die schnelle Reaktion. Ich möchte eben nur die Wortuhr bauen, mit Entwicklung hab ich so gar nichts am Hut.
Ich verwende Arduino 1.8.13. Mit PlatformIO habe ich mal getestet hier kommt aber immer ein anderer Fehler: Auto detecting Grunt for folder Wortuhr-2.6.0 failed with error: Error: Command failed: grunt --help --no-color
Der Befehl "grunt" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.

In Arduino erstelle ich ein Verzeichnis "Wortuhr" und kopiere dorthin alle Datein aus "src", "include" und das Verzeichnis "webpage".
Danach starte ich die Wortuhr.ino > alle Libraries sind installiert > beim kompilieren kommt dann eben besagter Fehler. Und ich weiß nicht was das für eine Datei oder Bibliothek sein soll.

Gruß
Jürgen

dbambus, 2021-04-29:
Hallo @hohoho-ia,

echt klasse, eine eigne Wordclock von Grund auf zu bauen ist schon ein kleines Erlebnis :-). Grundsätzlich ist zur Softwareseite zu sagen, dass wir weg von Arduino gegangen sind und nur noch platfom.io unterstützen. Hier müsste man über Python grunt nachinstallieren, eigentlich sollte das platfom.io selbst machen können. Ich verwende hierzu Visual Studio Code mit dem platfom.io plugin.

Du kannst aber auf das alte [Release 2.5.1](https://github.com/ESPWortuhr/Wortuhr/releases/tag/2.5.1) gehen und dir die Source Files holen. Diese sind noch zu Arduino kompatibel und sollten sich vollständig bauen lassen. 

Mit dieser Version kannst du leicht mit dem internen Updater auf die neuste Version kommen.

ATho95, 2021-04-29:
@dbambus , warum check't ihr das File nicht einfach mit ein?
Ich nutze Eclipse mit Arduino Plugin und habe Grunt selbst installieren müssen.

dbambus, 2021-04-29:
@tali Eigentlich sollte es möglich sein die Datei mit einzuchecken.

hohoho-ia, 2021-04-30:
Hallo zusammen,
Danke für die Unterstützung. Ich hab das mit Platform.io nochmal probiert aber leider nicht hinbekommen. Ich bin hier erst mal raus. Die Version 2.5 läuft (über Arduino) bei der werde ich es dann wohl lassen.
@dbambus  Du hast recht, der Aufbau der Wortuhr ist ein tolles Projekt und das Ergebnis begeistert alle. Danke nochmal für die tolle Arbeit hier!

tali, 2021-04-30:
Ist denn NPM als Abhängigkeit wirklich so schwierig, dass wir die Webseiten nicht überall frisch generieren können?
Die `WebPageContent.gen.inc` ist nun mal generiert und keine Quelldatei.

dbambus, 2021-04-30:
Ich hab auf einem fremden Rechner mal Platform.io direkt für die Worduhr aufgesetzt und musste im Fall von MAC OS X erst mal Homebrew installieren um damit npm zu installieren. Also der Aufwand für nicht versierte Leute ist damit schon größer. Ist halt die Frage, was man dem Otto-Gitnutzer zumutet.

ATho95, 2021-04-30:
@tali, ich stimme dbambus voll und ganz zu

Eisbaeeer, 2021-04-30:
Die Seite kann nicht mit ins Paket, da sie dynamisch erstellt wird. Hat ja @tali schon erwähnt. Ist übrigens in anderen Projekten auch so und genau das ist ja unter anderem der Vorteil von Platform.io gegenüber z.B. der IDE (mal abgesehen von dem grottigen Editors der IDE). Das sollte auch ein Github Nutzer hinbekommen (finde ich).

killee, 2021-05-04:
Hi Zusammen,
also ich tue mir auch schwer das alles zum Laufen zu bekommen.
Nach dem ich vermutlich alle Tools gesammelt habe, scheitere ich jetzt doch irgendwie an grunt.
Ich habe das Projekt in VSC mit PlatformIO geladen, npm, git und was weis ich alles installiert.
Jetzt bricht er ab mit:
node_modules/grunt/bin/grunt build
Der Befehl "node_modules" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
*** [include\WebPageContent.gen.inc] Error 1

Leider bringt mich jetzt hier meine Google Fähigkeiten nicht mehr weiter... An was könnte das liegen?

Wäre super wenn ihr ganz grob die nötigen Tools und schritte zum bauen dokumentieren könntet. 
Viele Grüße
Kilian

dbambus, 2021-05-05:
@killee wäre das für Windows ? 

Ich kann mal schauen, ob ich da eine Schritt für Schritt Anleitung auf einem Vanilla System erstellen kann. 

LG
David

killee, 2021-05-05:
Ja habe es auf Windows laufen.
Ja das wäre schön!

dbambus, 2021-05-11:
Hallo, 

ich habe es immer noch nicht wirklich auf dem Windows zum laufen bekommen. Bei mir kommen auch nur nichtssagende Fehlermeldungen. Mir scheint es fast so, dass die Verzeichnisstruktur von Windows da einen Fehler verursacht. 

@tali und @ATho95  könntet Ihr euch auch mal dran versuchen ? Evtl in einer VM ?

ATho95, 2021-05-12:
Nach den Anleitungen für Windows hatte ich gleich so ein Bauchgefühl, dass das schwierig wird. Ich habe dann eine Anweisung für Linux in der WSL (Windows Subsystem for Linux, zu finden unter Apps und Features -> Optionale Features -> Mehr Windows Funktionen) ausgeführt. Läuft gut, wenn man die line endings in der index.html auf unix ändert.

Installation:
sudo apt-get update && sudo apt-get upgrade
curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.33.0/install.sh | bash
restart bash
nvm install node
npm install -g grunt-cli

Run once for each project:
cd <project_root>
npm install

Run Grunt:
cd <project_root>
grunt

tali, 2021-05-15:
Könnt ihr bitte einmal ausprobieren, ob das mit #99 bei Euch besser klappt?

ATho95, 2021-05-16:
In meiner WSL Umgebung läuft es nach wie vor

dbambus, 2021-05-16:
In meiner MacOS Umgebung funktioniert alles wie zuvor. Auf der Windows Installation funktioniert nun grunt, aber es kommt zu Problemen mit `htmllint`, weil Windows ein anderes Line-Ending verwendet. Dazu habe ich einen neuen PR gestellt https://github.com/ESPWortuhr/Wortuhr/pull/100 der das Line Ending ignoriert. Damit funktioniert das Bauen dann auch auf  der Windows Kiste.

killee, 2021-05-17:
Ja super, jetzt bekomme ich es auch gebaut!
Vielen Dank!

closed 2021-05-17 by dbambus