ESPWortuhr/Multilayout-ESP-Wordclock #94: #defines LED_LAYOUT_REVERSE config over webfrontend

killee, 2021-05-04:
Hi Zusammen,

wäre super wenn auch das über das Web Frontend zu konfigurieren wäre.

Viele Grüße
Kilian

dbambus, 2021-05-05:
Ich mach mich mal dran und schau mal was möglich ist. 

Ohne jetzt viel überlegen zu müssen wäre es nicht einfach für reguläre Layouts (nicht Uhr169) `Word_array[usedUhrType->NUM_PIXELS - i ] `

@ATho95 würde das klappen ? auch mit deinen Animationen ?

dbambus, 2021-05-05:
Um welches Layout würde es den gehen ? @killee

killee, 2021-05-05:
Hi,

ich habe es jetzt mit der Version 2,5.1 angepasst... Hab die andere ja nicht zum Kompilieren bekommen.
Habe da die uhr_func generell umgebaut, die war da noch ziemlich verquer...

Ich habe das:
E S K I S T L F Ü N F    ==> ES IST FÜNF
Z E H N Z W A N Z I G    ==> ZEHN ZWANZIG
D R E I V I E R T E L    ==> DREI|VIERTEL
T G N A C H V O R J M    ==> NACH VOR
H A L B Q Z W Ö L F P    ==> HALB ZWÖLF
Z W E I N S I E B E N    ==> ZW|EI|N|S|IEBEN
K D R E I R H F Ü N F    ==> DREI FÜNF
E L F N E U N V I E R    ==> ELF NEUN VIER
W A C H T Z E H N R S    ==> ACHT ZEHN
B S E C H S F M U H R    ==> SECHS UHR

mit der Anordnung:
        {109, 108, 107, 106, 105, 104, 103, 102, 101, 100, 99},
        {88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98},
        {87, 86, 85, 84, 83, 82, 81, 80, 79, 78, 77},
        {66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76},
        {65, 64, 63, 62, 61, 60, 59, 58, 57, 56, 55},
        {44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54},
        {43, 42, 41, 40, 39, 38, 37, 36, 35, 34, 33},
        {22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32},
        {21, 20, 19, 18, 17, 16, 15, 14, 13, 12, 11},
        {0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10},
        {110, 111, 112, 113}, 

Wenn ich den Trunk bauen könnte, würde ich es ja einbauen ;-)...

Grüße
Kilian

killee, 2021-05-05:
Insgesamt wäre es aber super wie schon wo anders erwähnt, wenn man das Layout und LED Matrix über das Webfrontend frei konfigurieren könnte.
Würde da auch den unterbau vorbereiten...

ATho95, 2021-05-05:
> 
> 
> Ich mach mich mal dran und schau mal was möglich ist.
> 
> Ohne jetzt viel überlegen zu müssen wäre es nicht einfach für reguläre Layouts (nicht Uhr169) `Word_array[usedUhrType->NUM_PIXELS - i ] `
> 
> @ATho95 würde das klappen ? auch mit deinen Animationen ?

@dbambus, dieses Word_array scheint ein Artefakt aus der Zeit zu sein, wo du (?) damit ein Überblenden versucht hast, oder?
Das ist eigentlich redundant: `Word_array[index] = index;`. Es sei denn man möchte das Layout darüber abbilden.
Der Vergleich mit `Word_array_old` stellt eigentlich nur fest, dass die Minuten eine 5er Grenze überschritten haben. D.h dieses "Flag" könnte in einem bool liegen.

Ich überarbeite gerade die Animation im Hinblick auf #91. Ist fertig, aber ich möchte noch Speicher sparen.
Die braucht dann COLS_MATRIX(), ROWS_MATRIX(), getFrontMatrix(r, c) und getMinArr(). Wie das im Unterbau gelöst wird ist egal.

dbambus, 2021-05-06:
Hey @ATho95,

Das Word_array ist eigentlich da, um direkt abbilden zu können, was auf der Uhrmatrix aktiv ist. Es ist in der tat nicht wirklich gut gecodet, vergleiche `Word_array[index] = index;` 

Aber ich habe mir schon länger meine Gedanken gemacht. Der Informationsgehalt des `Word_array` ist eigentlich binär, ich will ja nur zu jeder Zeit wissen, welche LED an ist, Farbe außen vor.

Deshalb habe ich einen Branch mit so etwas: 

```
    virtual void Letter_set(const uint16_t index) {
        uint8_t row = index / COLS_MATRIX();
        uint8_t col = index % COLS_MATRIX();
        if (row % 2 == 0) {
            col = COLS_MATRIX() - col - 1;
        }
        Word_array[row] = 1 << col;
    }
```

Hier speicher ich die Informationen binär in 32 Bit Wörtern in einem Array[NUM_ROWS] So kann ich im Logger einen Output liefern der so ausschauen könnte: 

```
Uhr zeigt:
row[0]: 11011100000
row[1]: 00000000000
row[2]: 00000011110
row[3]: 00000000000
row[4]: 00000000000
row[5]: 00000000000
row[6]: 00000000000
row[7]: 00000000000
row[8]: 00000000000
row[9]: 00000000111
```

Damit könnte man verschiedene (shift-)Operationen machen und in letzter Instanz led_set() ausführen.

killee, 2021-05-06:
Ja das wäre ein guter Ansatz. Ich habe dann zum debuggen auch die komplette Textmatrix angelegt.
const char front_layout_mc[11][11] = {
        {'E', 'S', 'K', 'I', 'S', 'T', 'L', 'F', 'U', 'N', 'F'},
        {'Z', 'E', 'H', 'N', 'Z', 'W', 'A', 'N', 'Z', 'I', 'G'},
        {'D', 'R', 'E', 'I', 'V', 'I', 'E', 'R', 'T', 'E', 'L'},
        {'T', 'G', 'N', 'A', 'C', 'H', 'V', 'O', 'R', 'J', 'M'},
        {'H', 'A', 'L', 'B', 'Q', 'Z', 'W', 'O', 'L', 'F', 'P'},
        {'Z', 'W', 'E', 'I', 'N', 'S', 'I', 'E', 'B', 'E', 'N'},
        {'K', 'D', 'R', 'E', 'I', 'R', 'H', 'F', 'U', 'N', 'F'},
        {'E', 'L', 'F', 'N', 'E', 'U', 'N', 'V', 'I', 'E', 'R'},
        {'W', 'A', 'C', 'H', 'T', 'Z', 'E', 'H', 'N', 'R', 'S'},
        {'B', 'S', 'E', 'C', 'H', 'S', 'F', 'M', 'U', 'H', 'R'},
        {'#', '#', '#', '#'},
    };
Anzeigen kann ich dann über das Terminal was direkt die Uhr anzeigen würde:

0: ES IST
1:
2:
3:
4: HALB
5:
6:
7: ELF
8:
9:
10: #

Anbei mal wie ich das noch für die 2.5.1 gemacht habe:
#include "Uhrtype.hpp"

//Eine LED zum level schiften vor der Matrix genutzt
#define LEDS_BEVOR_MATRIX 1

class UHR_114_t : public iUhrType
{
public:
    const uint8_t matrix_top_left[11][11] = {
        //aus 10  11

        {0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10},
        {21, 20, 19, 18, 17, 16, 15, 14, 13, 12, 11},
        {22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32},
        {43, 42, 41, 40, 39, 38, 37, 36, 35, 34, 33},
        {44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54},
        {65, 64, 63, 62, 61, 60, 59, 58, 57, 56, 55},
        {66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76},
        {87, 86, 85, 84, 83, 82, 81, 80, 79, 78, 77},
        {88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98},
        {109, 108, 107, 106, 105, 104, 103, 102, 101, 100, 99},
        {110, 111, 112, 113}, //ergänzt

    };

    const uint8_t matrix_button_right[11][11] = {
        {103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113},
        {102, 101, 100, 99, 98, 97, 96, 95, 94, 93, 92},
        {81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91},
        {80, 79, 78, 77, 76, 75, 74, 73, 72, 71, 70},
        {59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69},
        {58, 57, 56, 55, 54, 53, 52, 51, 50, 49, 48},
        {37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47},
        {36, 35, 34, 33, 32, 31, 30, 29, 28, 27, 26},
        {15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25},
        {14, 13, 12, 11, 10, 9, 8, 7, 6, 5, 4},
        {0, 1, 2, 3}};
    const uint8_t matrix_button_left[11][11] = {
        //aus 10  11

        {109, 108, 107, 106, 105, 104, 103, 102, 101, 100, 99},
[... 295 lines cut ...]

            //------------------------------------------------------------------------------

        case h_elf:
            Letter_set(getMatrix(7, 0));
            Letter_set(getMatrix(7, 1));
            Letter_set(getMatrix(7, 2));
            break;

            //------------------------------------------------------------------------------

        case h_zwoelf:
            Letter_set(getMatrix(4, 5));
            Letter_set(getMatrix(4, 6));
            Letter_set(getMatrix(4, 7));
            Letter_set(getMatrix(4, 8));
            Letter_set(getMatrix(4, 9));
            break;

        default:
            break;
            //------------------------------------------------------------------------------
        };
    };
};

dbambus, 2021-05-16:
@killee 

Möchtest du nicht einen eigenen Fork eröffnen in welchem du deine Änderungen einfügst ? Dann würde ich mir das gerne anschauen wollen. Mit dem aktuellen Stand existiert ja auch eine Anleitung wie man die Umgebung Platform.io mit samt der nötigen Plugins unter Windows und Unix Systemen verwendet. 

LG
David

killee, 2021-05-16:
Ja klar mach ich gerne wenn ich es bauen kann. Schaue die Tage das nochmal an.
Grüße
Kilian

ATho95, 2021-05-31:
Ich habe mir mal Gedanken gemacht, wie ein Upload eines Uhr Layouts aussehen könnte und bin dabei auf ein Json Format File gekommen, das man im Browser hochladen kann.
Für die Worduhr 114 Reverse würde das so aussehen:

{
"numpixels":114,
"rows":10,
"cols":11,
"hasdreiviertel":1,
"has24HourLayout":0,
"ledpositions": [
  [103,  1],
  [102, -1],
  [ 81,  1],
  [ 80, -1],
  [ 59,  1],
  [ 58, -1],
  [ 37,  1],
  [ 36, -1],
  [ 15,  1],
  [ 14, -1]
],
"minarray":[
  [0, 1, 2, 3],
  [0, 1, 2, 3]],
"words": {
  "es_ist":[0, 0, 2, 0, 3, 3],
  "nach":[3, 2 , 4],
  "v_nach":[3, 2 , 4],
  "vor":[3, 6, 3],
  "v_vor":[3, 6, 3],
  "viertel":[2, 4, 7],
  "dreiviertel":[2, 0, 11],
  "fuenf":[0, 7, 4],
  "zehn":[1, 0, 4],
  "zwanzig":[1, 4, 7],
  "halb":[4, 0, 4],
  "eins":[5, 2, 4],
  "uhr":[9, 7, 3],
  "h_ein":[5, 2, 3],
  "h_zwei":[5, 0, 4],
  "h_drei":[6, 1, 4],
  "h_vier":[7, 7, 4],
  "h_fuenf":[6, 7, 4],
  "h_sechs":[9, 1, 5],
  "h_sieben":[5, 5, 6],
  "h_acht":[8, 1, 4],
  "h_neun":[7, 3, 4],
  "h_zehn":[8, 5, 4],
  "h_elf":[7, 0, 3],
  "h_zwoelf":[4, 5, 5]
  }
}
Ich denke der Aufbau ist selbsterklärend:
   ledpositions: 
      links oben anfangend: LED position in der Kette, die Nachbar LEDs folgen mit +1 oder -1  Abstand
   words:
      "enum ledText" : row , col, length (bei mehreren Wörtern werden diese mit row, col, length angehängt)

Ein proof of concept läuft auch schon, wenn das Json File als String im Flash liegt.
Überschlägig würde eine Uhr mit 20 rows und allen bis jetzt definierten Wörtern weniger als 500 Byte in Anspruch nehmen. Damit passt eine generische Uhr noch mit in das EEPROM.

Ein nächster Schritt könnte dann sein , dass alle bekannten Layouts als Json Daten im Flash abgelegt sind.

Was haltet ihr davon?

killee, 2021-05-31:
Hi @ATho95 ,

ja das schaut für mich perfekt aus!
Mein Layout könnte ich so direkt alles Umsetzen.

Die Definition der Wörter wollte ich auch noch so umbauen wie du dies umgesetzt hast.

Einzige Ausnahme ist wenn die "S-Matrix" senkrecht verläuft...
Noch ein flag rotate?
Brauch ich dann überhaupt noch die Definition der "S-Matrix"?

Grüße
Kilian

dbambus, 2021-06-01:
Hey, 

die Arbeit sieht echt gut aus und das ist definitiv eine sehr gute Grundlage :-) 

Zum JSON Format:

Ist es notwendig hasDreivirtel zu definieren ? Wenn es nicht in den gelisteten Wörtern vorkommt ist es automatisch 'false' bspw. 

Aus Anwendersicht:

Wird es nach wie vor eine Auswahloption auf der Website geben, in derer die predefined JSONs selektiert werden können + 1 Custom  ? 

Könnte man nicht eine default Wortuhr fest im EEPROM hinerlegen, ähnlich zu den iUhrtype und dann nur die Werte von den überschriebenen, erbten Klassen in den JSON-String packen? Das würde auch das Problem mit den verschiednen Matrizen lösen.

Meines Wissens nach lässt dich das EEPROM und der Flash nacheinander mit platformio beschreiben, wie sieht es dann och mit der Möglichkeit eines OTA updates aus ? Wäre es hier nicht optimaler die Software holt sich die Layouts direkt aus dem hier hinterlegten Repo? (Das soll nur als Gedankenanstoß dienen)

LG
David

ATho95, 2021-06-01:
> Ist es notwendig hasDreivirtel zu definieren ?

Ist nicht notwendig, ebenso hasZwanzig, hasWeatherLayout, evtl. auch has24HourLayout

Den Sekundenrand hatte ich gar nicht auf dem Schirm. Aus
"ledpositionsseconds":[0, 1, 2, usw.]
kann man auch hasSecondsFrame und NUM_RMATRIX ableiten

> Wird es nach wie vor eine Auswahloption auf der Website geben, in derer die predefined JSONs selektiert werden können + 1 Custom ?

Genau
> Könnte man nicht eine default Wortuhr fest im EEPROM hinerlegen

Das hatte ich im "nächsten Schritt" in etwa so angedacht. Es gibt dann nur noch die eine Uhren-Klasse, die aufgrund von Json Daten arbeitet. Entweder hat man default-Werte oder füttert die Klasse erst mit der Ur-Uhr.  Anschließend werden die Grundwerte nur mit Änderungen überschrieben. Wenn sich die Anzahl der LEDs nicht ändert bräuchte man noch nicht einmal einen Reset.

> Meines Wissens nach lässt dich das EEPROM und der Flash nacheinander mit platformio beschreiben, wie sieht es dann och mit der Möglichkeit eines OTA updates aus ? Wäre es hier nicht optimaler die [...]

Ich kenne platformio nicht, vermute aber, dass die auch die Espressif Python tools nehmen. Damit kann man auch einzelne Bereiche flashen.
Ein EEPROM im eigentlichen Sinn gibt es ja nicht im ESP. Die EEPROM Klasse verwendet einen 4kB großen Bereich im Flash. Da ist jetzt auch die G Struktur abgelegt und es sind noch ca. 3,5kB frei. Die bereits vorhandenen Layouts müssen als JSON in den IROM Bereich. Das Custom Layout könnte im EEPROM unterkommen. (oder im Flash File System, aber das ist bisher nicht verwendet und so sollte es auch bleiben)
Solange das Flash einen OTA Bereich hat, funktioniert OTA (bzw. der Webupdate).

@killee die SMATRIX habe ich immer links liegen gelassen, ich glaube auch nicht, dass die für senkrechte LED-Streifen gedacht ist. Obwohl, wenn man richtig generisch sein möchte muss man auch das unterstützen.

dbambus, 2023-03-10:
Hey, nach einer langen Zeit kann ich nun dieses Ticket schließen

closed 2023-03-10 by dbambus