ESPWortuhr/Multilayout-ESP-Wordclock #98: Passwort für "Uhr" Wlan

Psychedili, 2021-05-14:
Hallo,
Ich würde das WLAN PW für die Version 2.5.0 benötigen.

dbambus, 2021-05-15:
Hallo, 

das Passwort war im früheren Release "12345678"

Psychedili, 2021-05-15:
Danke für die schnelle Antwort.
Jetzt habe ich noch eine Frage. Ich habe in der Version 2.5.0 das Uhrenlayout anpassen müssen.
Wenn ich jetzt die .bin Datei der Version für ein OTA Update benutze ist dann mein Uhrenlayout auch überschrieben oder bleibt das erhalten?

dbambus, 2021-05-16:
Hallo @Psychedili,

Wenn du direkt die vorkompilierte Binary verwendest, wird das Uhren Layout, dass du in deinem Code damals angepasst hast, überschrieben werden. Diese Release-Binary ist für Neulinge gedacht, die ein fertiges Image wollen und ein standardmäßiges Layout verwenden. 

Du musst für dein spezielles Layout den frischen SourceCode von diesem Repo hernehmen und anschließend dein Layout in diesen Einpflegen, anschließend kompilieren und flashen. 

LG
David

closed 2021-05-18 by dbambus