ESPWortuhr/Multilayout-ESP-Wordclock #112: .bin Datei

Psychedili, 2021-07-13:
Hallo, 
ich hoffe mir kann jemand weiterhelfen. 
Wie kann ich eine .bin Datei mit PIO erzeugen bzw wo finde ich diese beim Mac?

Vielen Dank

tali, 2021-07-26:
Wenn du `pio run` aufrufst, dann wird dadurch `.pio/build/nodemcuv2/firmware.bin` erzeugt.

n3roGit, 2021-11-04:
Bei mir klappt es auch nicht mit dem kompilieren. Eine bin Datei wäre hier wirklich klasse! Nicht jeder ist ein Programmierer ;)

`Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-------------------------------------------------------------------------------------------------Tool Manager: Installing platformio/tool-mkspiffs @ ~1.200.0
Tool Manager: tool-mkspiffs @ 1.200.0 has been installed!
Tool Manager: Installing platformio/tool-mklittlefs @ ~1.203.0
Tool Manager: tool-mklittlefs @ 1.203.210628 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (3.2.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.30002.0 (3.0.2)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - tool-mklittlefs 1.203.210628 (2.3)
 - tool-mkspiffs 1.200.0 (2.0)
 - toolchain-xtensa 2.100300.210717 (10.3.0)
Converting Wortuhr.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing git+https://github.com/tzapu/WiFiManager.git
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\Users\Userrname\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 180:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Userrname\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Userrname\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Userrname\.platformio\platforms\espressif8266\builder\main.py", line 238:       
    target_elf = env.BuildProgram()
  File "C:\Users\Userrname\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Userrname\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Userrname\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Userrname\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Userrname\.platformio\packages\tool-scons\scons-local-4.2.0\SCons\Util.py", line 748:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Userrname\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1104:
    project.install_dependencies()
  File "C:\Users\Userrname\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 940:
    lm.install(spec)
  File "c:\users\Userrname\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49:
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "c:\users\Userrname\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 96:
    force=force,
  File "c:\users\Userrname\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97:
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "c:\users\Userrname\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141:
    vcs = VCSClientFactory.new(tmp_dir, url)
  File "c:\users\Userrname\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55:
    src_dir, remote_url, tag, silent
  File "c:\users\Userrname\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 138:
    super(GitClient, self).__init__(*args, **kwargs)
  File "c:\users\Userrname\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 72:
    self.check_client()
  File "c:\users\Userrname\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 162:
    "Please install Git client from https://git-scm.com/downloads"
================================== [FAILED] Took 6.27 seconds ==================================
The terminal process "C:\Users\Userrname\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload', '--environment', 'nodemcuv2'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

tali, 2021-11-04:
Der interessante Teil der Fehlermeldung ist dieser hier:
> UserSideException: Please install Git client from https://git-scm.com/downloads:

tali, 2021-11-04:
Aber stimmt, da sollten wir auch mal schauen, dass wir auch bei WifiManager richtige Releases verwenden und nicht die Version aus Git.

n3roGit, 2021-11-04:
Nach einen reboot war der Fehler weg. jetzt habe ich einen neuen.

Nebenbei: Tolles Projekt und vielen Dank für die flotte Antwort!!!

Ich bau diese Uhr gerade als Weihnachtsgeschenk. Vielleicht kannst du mir ja helfen, die Datei zu kompilieren :)

`> Executing task: C:\Users\username\.platformio\penv\Scripts\platformio.exe run --environment nodemcuv2 <

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (3.2.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.30002.0 (3.0.2)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.100300.210717 (10.3.0)
Converting Wortuhr.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <NeoPixelBus> 2.6.9
|   |-- <SPI> 1.0
|   |-- <I2S> 1.0
|-- <ArduinoJson> 6.18.5
|-- <WebSockets> 2.3.6
|   |-- <ESP8266WiFi> 1.0
|   |-- <Ethernet> 2.0.0
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <RTClib> 1.14.2
|   |-- <Wire> 1.0
|-- <PubSubClient> 2.8.0
|-- <WiFiManager> 2.0.4-beta+sha.72b5331
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS> 1.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266HTTPUpdateServer> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <Wire> 1.0
|-- <EEPROM> 1.0
Building in release mode
Compiling .pio\build\nodemcuv2\src\Network.cpp.o
npm ci --silent
build_version_h(["include\version.gen.h"], ["package.json"])
Compiling .pio\build\nodemcuv2\src\font.cpp.o
Compiling .pio\build\nodemcuv2\src\Wortuhr.ino.cpp.o
Generating LD script .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\nodemcuv2\lib4d1\SPI\SPI.cpp.o
Compiling .pio\build\nodemcuv2\lib39a\I2S\I2S.cpp.o
[... 314 lines cut ...]
include/uhr_func.hpp:37:23: error: 'class NeoPixelBus<NeoMultiFeature, NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x> >' has no member named 'GetPixelColor'; did you mean 'SwapPixelColor'?
   37 |     return strip_RGB->GetPixelColor(i);
      |                       ^~~~~~~~~~~~~
      |                       SwapPixelColor
include/uhr_func.hpp: In function 'void led_clear_pixel(uint16_t)':
include/uhr_func.hpp:116:20: error: 'class NeoPixelBus<NeoMultiFeature, NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x> >' has no member named 'SetPixelColor'; did you mean 'SwapPixelColor'?
  116 |         strip_RGB->SetPixelColor(i, 0);
      |                    ^~~~~~~~~~~~~
      |                    SwapPixelColor
Compiling .pio\build\nodemcuv2\lib62c\NeoPixelBus\internal\Esp32_i2s.c.o
Compiling .pio\build\nodemcuv2\lib62c\NeoPixelBus\internal\HsbColor.cpp.o
In file included from C:/Users/username/wortuhrcode/Wortuhr/src/Wortuhr.ino:52:
.pio\libdeps\nodemcuv2\NeoPixelBus\src/NeoPixelBus.h: In instantiation of 'NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::NeoPixelBus(uint16_t) [with T_COLOR_FEATURE = NeoMultiFeature; T_METHOD = NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x>; uint16_t = short unsigned int]':
C:/Users/username/wortuhrcode/Wortuhr/src/Wortuhr.ino:165:42:   required from here
.pio\libdeps\nodemcuv2\NeoPixelBus\src/NeoPixelBus.h:159:47: error: 'PixelSize' is not a member of 'NeoMultiFeature'
  159 |         _method(countPixels, T_COLOR_FEATURE::PixelSize, T_COLOR_FEATURE::SettingsSize)
      |                                               ^~~~~~~~~
.pio\libdeps\nodemcuv2\NeoPixelBus\src/NeoPixelBus.h:159:75: error: 'SettingsSize' is not a member of 'NeoMultiFeature'
  159 |         _method(countPixels, T_COLOR_FEATURE::PixelSize, T_COLOR_FEATURE::SettingsSize)
      |                                                                           ^~~~~~~~~~~~
Compiling .pio\build\nodemcuv2\lib62c\NeoPixelBus\internal\HslColor.cpp.o
*** [.pio\build\nodemcuv2\src\Wortuhr.ino.cpp.o] Error 1
======================================================================================================================================= [FAILED] Took 30.52 seconds =======================================================================================================================================The terminal process "C:\Users\username\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'nodemcuv2'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.`

n3roGit, 2021-11-04:
Ich bau mir jetzt mal einen Docker Container und versuche es darin zu kompilieren. Ich berichte gleich, ob es da anders aussieht

tali, 2021-11-04:
Ich schau mir das später mal an.
Übrigens gibt es bei den Releases tatsächlich fertige .bin Dateien zum flashen: https://github.com/ESPWortuhr/Wortuhr/releases/tag/V2.6.1

n3roGit, 2021-11-04:
Sehr Cool! Werde ich gleich mal testen!

Zur Info:
in einem frischen Docker mit den ganzen genannten Abhängigkeiten bekomme ich folgende Meldung:

`root@BuildWortuhr:~/Wortuhr# pio run -t upload
*************************************************************************************************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
*************************************************************************************************************************************************************************************************************************************************************************************************

Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing espressif8266
Error: Detected unknown package 'espressif8266'
root@BuildWortuhr:~/Wortuhr#
`

n3roGit, 2021-11-04:
Okay bin etwas weiter innerhalb des Docker:

Folgende Pakete habe ich nun installiert:
apt install git curl python3 python3-venv npm
curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py -o get-platformio.py

Dann passiert folgendes beim bauen
`In file included from /root/Wortuhr/src/Wortuhr.ino:64:
include/WebPage_Adapter.h: In function 'void webSocketEvent(uint8_t, WStype_t, uint8_t*, size_t)':
include/WebPage_Adapter.h:331:27: warning: format '%ld' expects argument of type 'long int', but argument 3 has type 'time_t' {aka 'long long int'} [-Wformat=]
  331 |             Serial.printf("Conf: Time: %ld\n", tv.tv_sec);
      |                           ^~~~~~~~~~~~~~~~~~~  ~~~~~~~~~
      |                                                   |
      |                                                   time_t {aka long long int}
include/WebPage_Adapter.h:436:27: warning: format '%ld' expects argument of type 'long int', but argument 3 has type 'time_t' {aka 'long long int'} [-Wformat=]
  436 |             Serial.printf("Conf: Time: %ld\n", tv.tv_sec);
      |                           ^~~~~~~~~~~~~~~~~~~  ~~~~~~~~~
      |                                                   |
      |                                                   time_t {aka long long int}
In file included from /root/Wortuhr/src/Wortuhr.ino:86:
include/NeoMultiFeature.hpp: At global scope:
include/NeoMultiFeature.hpp:14:55: error: expected class-name before '{' token
   14 | class NeoMultiFeature : public Neo3ElementsNoSettings {
      |                                                       ^
include/NeoMultiFeature.hpp:22:33: error: 'ColorObject' has not been declared
   22 |                                 ColorObject color) {
      |                                 ^~~~~~~~~~~
include/NeoMultiFeature.hpp:50:12: error: 'ColorObject' does not name a type
   50 |     static ColorObject retrievePixelColor(const uint8_t *pPixels,
      |            ^~~~~~~~~~~
include/NeoMultiFeature.hpp:82:12: error: 'ColorObject' does not name a type
   82 |     static ColorObject retrievePixelColor_P(PGM_VOID_P pPixels,
      |            ^~~~~~~~~~~
include/NeoMultiFeature.hpp: In static member function 'static void NeoMultiFeature::applyPixelColor(uint8_t*, uint16_t, int)':
include/NeoMultiFeature.hpp:23:22: error: 'getPixelAddress' was not declared in this scope
   23 |         uint8_t *p = getPixelAddress(pPixels, indexPixel);
      |                      ^~~~~~~~~~~~~~~
include/NeoMultiFeature.hpp:27:26: error: request for member 'B' in 'color', which is of non-class type 'int'
   27 |             *p++ = color.B;
      |                          ^
include/NeoMultiFeature.hpp:28:26: error: request for member 'R' in 'color', which is of non-class type 'int'
   28 |             *p++ = color.R;
      |                          ^
include/NeoMultiFeature.hpp:29:24: error: request for member 'G' in 'color', which is of non-class type 'int'
   29 |             *p = color.G;
      |                        ^
include/NeoMultiFeature.hpp:32:26: error: request for member 'G' in 'color', which is of non-class type 'int'
   32 |             *p++ = color.G;
      |                          ^
include/NeoMultiFeature.hpp:33:26: error: request for member 'R' in 'color', which is of non-class type 'int'
   33 |             *p++ = color.R;
      |                          ^
include/NeoMultiFeature.hpp:34:24: error: request for member 'B' in 'color', which is of non-class type 'int'
   34 |             *p = color.B;
      |                        ^
include/NeoMultiFeature.hpp:37:26: error: request for member 'R' in 'color', which is of non-class type 'int'
   37 |             *p++ = color.R;
      |                          ^
include/NeoMultiFeature.hpp:38:26: error: request for member 'B' in 'color', which is of non-class type 'int'
   38 |             *p++ = color.B;
      |                          ^
include/NeoMultiFeature.hpp:39:24: error: request for member 'G' in 'color', which is of non-class type 'int'
   39 |             *p = color.G;
      |                        ^
include/NeoMultiFeature.hpp:43:26: error: request for member 'R' in 'color', which is of non-class type 'int'
   43 |             *p++ = color.R;
      |                          ^
include/NeoMultiFeature.hpp:44:26: error: request for member 'G' in 'color', which is of non-class type 'int'
   44 |             *p++ = color.G;
      |                          ^
[... 51 lines cut ...]
include/uhr_func.hpp: In function 'void led_clear_pixel(uint16_t)':
include/uhr_func.hpp:116:20: error: 'class NeoPixelBus<NeoMultiFeature, NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x> >' has no member named 'SetPixelColor'; did you mean 'SwapPixelColor'?
  116 |         strip_RGB->SetPixelColor(i, 0);
      |                    ^~~~~~~~~~~~~
      |                    SwapPixelColor
Compiling .pio/build/nodemcuv2/lib57f/NeoPixelBus/internal/HslColor.cpp.o
Archiving .pio/build/nodemcuv2/libbb1/libSPI.a
Indexing .pio/build/nodemcuv2/libbb1/libSPI.a
Compiling .pio/build/nodemcuv2/lib57f/NeoPixelBus/internal/HtmlColor.cpp.o
In file included from /root/Wortuhr/src/Wortuhr.ino:52:
.pio/libdeps/nodemcuv2/NeoPixelBus/src/NeoPixelBus.h: In instantiation of 'NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::NeoPixelBus(uint16_t) [with T_COLOR_FEATURE = NeoMultiFeature; T_METHOD = NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x>; uint16_t = short unsigned int]':
[... cut ...]
.pio/libdeps/nodemcuv2/NeoPixelBus/src/NeoPixelBus.h:159:75: error: 'SettingsSize' is not a member of 'NeoMultiFeature'
  159 |         _method(countPixels, T_COLOR_FEATURE::PixelSize, T_COLOR_FEATURE::SettingsSize)
      |                                                                           ^~~~~~~~~~~~
*** [.pio/build/nodemcuv2/src/Wortuhr.ino.cpp.o] Error 1
npm notice
npm notice New major version of npm available! 7.5.2 -> 8.1.2
npm notice Changelog: <https://github.com/npm/cli/releases/tag/v8.1.2>
npm notice Run `npm install -g npm@8.1.2` to update!
npm notice
`

n3roGit, 2021-11-05:
@tali konntest du etwas raus finden?

n3roGit, 2021-11-08:
@tali sorry, falls ich nerve, aber ich komme leider absolut nicht weiter :(

dbambus, 2021-11-15:
Hey @n3roGit, 

ich habe mit https://github.com/ESPWortuhr/Wortuhr/pull/125 und https://github.com/ESPWortuhr/Wortuhr/pull/126 die Compiler Errors behoben, jetzt sollte die binare wieder bauen. Könntest du das mal testen ?

n3roGit, 2021-11-15:
Teste ich morgen direkt! Vielen Dank!

BTW. Ich hab gesehen das im Code sprach ansagen in den Einstellungen sind. Wo wird beschrieben, wie man dafür die Elektronik verdrahtet muss?

dbambus, 2021-11-15:
Hey, 

kannst du mir sagen wo im Code du die Stellen gefunden hast. Mir ist nichts bekannt mit Sprachansagen. 

Mit freundlichen Grüßen
David

n3roGit, 2021-11-16:
[image]
Super, das bauen geht jetzt auch! Vielen Dank!

[image]
In den Einstellungen habe ich "Sprachoptionen" gefunden. Oder ist das ein Relikt?

Ich habe mich auch gefragt, wofür unter LDR der Punkt "Kalibrierung" 0-10 ist.

dbambus, 2021-11-16:
Hey @n3roGit,

Die Sprachoptionen beziehen sich auf die Ausdrucksweise der Uhrzeiten, die ja im deutschen Sprachraum nicht einheitlich sind. 

Mit der kalibrierung kannst du letztlich die Empfindlichkeit des Sensors und die damit resultierende Helligkeitsänderung einstellen.

closed 2021-11-16 by dbambus