ESPWortuhr/Multilayout-ESP-Wordclock #123: Frontplatte Vorlage

n3roGit, 2021-11-08:
Moin moin,

besteht die Möglichkeit eine CAD oder Fusion 350 Vorlage von euch zu erhalten? Das würde mir die Erstellung eines eigenen templates deutlich vereinfachen :D

dbambus, 2021-11-15:
Moin moin, 

sorry, hat etwas gedauert. Ich habe Zugang zu einem LaserCutter hier am örtlichen Fablab, deswegen fertige ich die Fronten indem ich transparente Acrylplatten mit schwarzen Acryllack homogen ansprühe und dann mit dem LaserCutter die Buchstaben wieder freilege. Der Rest ist als gelagertes Stecksystem aus Pappelholz konzipiert, mit einer Schaumstoff Einlage (Plastazone) zur Selektion der Buchstaben.

Eine Fertigungsweise auf Basis von gefrästen Platten wäre cool, aber da habe ich keine Erfahrung. Möchtest du da vielleicht behilflich sein ?

Mit freundlichen Grüßen
David

n3roGit, 2021-11-15:
Danke für die Antwort! Ich hab mittlerweile eine Vorlage und diese angepasst. Ich kann die gerne bereitstellen. Das würde sicher dem einen oder anderem helfen. Aktuell bin ich aber noch beim Bau. Wenn ich durch bin hab ich sicher einige schöne Fotos vom Bau

dbambus, 2021-11-15:
Hey, 

Sehr cool, freu mich, wenn es neue Bilder gibt.

Eisbaeeer, 2021-12-01:
Falls Interesse besteht, kann ich meine DXF ebenfalls bereitstellen. Passend für WS2812B mit 60 LED´s / Meter. Gefräst wurde MDF Platte von vorne und hinten. Die Frontplatte wird mit Neodym Magneten befestigt (aufgeklebt auf Plexiglas Front).
[image]
[image]

n3roGit, 2022-01-11:
Hier sind meine Bilder :)
[image]
[image]
[image]
[image]
[image]

Wichtig: Bei meiner Vorlage war das Wort "UHR" um eine Stelle weiter nach rechts gerückt. Ich musste die Vorlage in der Software dafür anpassen.

dbambus, 2022-01-11:
Es sieht wirklich super aus :-)

Das ich die Bilder in das Repo mit aufnehmen ?

n3roGit, 2022-01-11:
> Es sieht wirklich super aus :-)
> 
> Das ich die Bilder in das Repo mit aufnehmen ?

Klar =)

Falls benötigt schreibe ich gerne noch etwas dazu. Sag einfach bescheid!

closed 2022-02-22 by dbambus