ESPWortuhr/Multilayout-ESP-Wordclock #131: Animation - bug

Eisbaeeer, 2021-12-01:
Hallo Leute
Ich habe heute mal die aktuelle Version vom Github compiliert. Sieht schon sehr gut aus. Bezüglich Animationen wollte ich Euer Wissen in Anspruch nehmen, da der Wetterbereich von mir mit den Animationen nicht so richtig klappt. Zum einen wird der Wetterbereich öfters innerhalb einer Minute aktualisiert bzw. umgeschaltet, was wohl mit den Animationen nicht fluchtet.
Jetzt die Frage, können wir das ändern?
Was muss ich triggern beim Wechsel der Anzeige, damit die Animation sauber durchläuft.
Ich habe mal zur Veranschaulichung zwei Videos gemacht, in denen man das Problem erkennt.
https://cloud.weimars.net/index.php/s/EgKneCfM7HCMrFx
Grüße Lars

dbambus, 2023-02-22:
Hallo @Eisbaeeer,

das Issue ist jetzt relativ alt und ich habe hier keine Uhr zum testen. Besteht es denn weiterhin ? 

Mit freundlichen Grüßen
David

Eisbaeeer, 2023-03-21:
Jep, es besteht weiterhin. Ich mache bei Gelegenheit ein Video davon.

dbambus, 2023-12-31:
Müsste mit https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/346 gefixed sein.

closed 2023-12-31 by dbambus