ESPWortuhr/Multilayout-ESP-Wordclock #133: weißes Blitzen bei farbigen Buchstaben

Nemo135, 2021-12-18:
Uhrtyp: Uhr114_2Clock
LED: SK6812RGBW
Farbanimation: BunteBuchstaben
Version: vorkompiliertes Binary 2.6.1

Hallo,
bei diesen Einstellungen werden die Buchstaben farbig angezeigt,
in uregelmäßigen Abständen blitzen die Buchstaben, die an an sind, immer mal weider weiß.
Eine Regelmäßigkeit konnte ich dabei nicht feststellen.
G. Marco

dbambus, 2022-01-11:
@ATho95 Kannst du dieses Verhalten reproduzieren ? Ich denke die verantwortliche Codepassage müsste zu Animation.h gehören.

@Nemo135 sind das wirklich unregelmäßige Abstände ? Andernfalls könnte es ein Problem mit einem zyklischen Abruf sein.

Nemo135, 2022-03-01:
Hallo dbambus,
wollte zu dem Fehler mal Rückmeldung geben.
Nachdem ich eine zweite Wortuhr gebaut hatte, die das Problem nicht gezeigt hat, hab ich angefangen zu suchen...
Ich vermute, das mein verwendetes USB-Netzteil hier das Problem war... jedenfalls ist das Problem jetzt weg !
Sorry für den Aufwand.
G.Marco

closed 2022-03-01 by Nemo135