ESPWortuhr/Multilayout-ESP-Wordclock #134: Firmware-paket

myrhre, 2021-12-22:
Hallo, 
kann mir jemand helfen, wo ich die neueste Firmware finde zum updaten über das web-interface. Wäre super, wenn mir jemand helfen könnte.

udo1toni, 2021-12-25:
Gewöhnlich hier: https://github.com/ESPWortuhr/Wortuhr/releases

Möchtest Du den aktuellen Stand, müsstest Du selbst als den Sourcen kompilieren lassen.

dbambus, 2022-01-11:
Hallo @myrhre,

@udo1toni hat recht, unter releases findet man die aktuellen stabilen Versionen. 

Zum Updaten über das Webinterface musst du darin auf den Reiter "Ueber" navigieren und darin findest du einen Link, der dich auf die Seite http://<IP-Adress>:81/update verweist. Dort kannst du den Update Prozess starten.

Mit freundlichen Grüßen
David

closed 2022-01-11 by dbambus