ESPWortuhr/Multilayout-ESP-Wordclock #136: Wortuhr hängt sich auf

n3roGit, 2022-01-11:
Hallo zusammen,

meine Uhr ist fertig und hängt mittlerweile an der Wand. Allerdings habe ich aktuell noch folgendes Problem.

Alle 1-2 Tage hängt sich die Uhr einfach auf und verbindet sich dann nicht mehr mit dem WLAN. Man kann dann nur das "Herz" sehen.
Ab und dann macht der ESP dann noch einen reboot und bekommt sich wieder ein. In der Regel bleibt die Uhr aber in diesem Zustand. 

Nachdem ich Sie Stromlos genommen habe ich neu verbinde geht alles wieder.

Aktuell kann ich das Problem noch nicht eingrenzen.

Habt ihr eine Idee?

dbambus, 2022-01-11:
Hey @n3roGit,

ich werde das mal evaluieren, welche Version verwendest du ? Macht dein Router regelmäßig eine Zwangstrennung zum Provider, vielleicht bricht die Verbindung zum NTP server ab? 

Mit freundlichen Grüßen 
David

n3roGit, 2022-01-11:
> Hey @n3roGit,
> 
> ich werde das mal evaluieren, welche Version verwendest du ? Macht dein Router regelmäßig eine Zwangstrennung zum Provider, vielleicht bricht die Verbindung zum NTP server ab?
> 
> Mit freundlichen Grüßen David

Dank dir!
Nein ich habe hier keine Zwangstrennung. Der NTP ist auch in diesem Fall mein PiHole. Dieser ist die ganze Zeit an.

Kann ich über den USB sonst die Serielle Konsole abgreifen? Würde das helfen?

dbambus, 2022-01-11:
Hallo @n3roGit, 

Alles klar, ja ein Log würde helfen. Du kannst einfach über die Serielle Schnittstelle mit 115200 Baud/s loggen. Vielleicht bekommt man ja so Aufschluss über das Event. 

Mit freundlichen Grüßen
David

n3roGit, 2022-01-12:
@dbambus 
Kann es sein, dass nach der Einrichtung die serielle Schnittstelle nicht mehr erreichbar ist?

Ich habe mir gerade ein testboard zusammen gebastelt. Nach der Verbindung mit dem Wifi komme ich da nicht mehr via Serial ran.

[image]
[image]

n3roGit, 2022-01-12:
[image]
[image]

Sodele. Debug läuft. Mal abwarten, was passiert.

n3roGit, 2022-01-17:
Hmm, mit meinem testaufbau hat bis jetzt alles funktioniert. Ich Bau die Tage die fertige Uhr nochmal auseinander und werde es dort loggen

closed 2022-02-22 by dbambus