ESPWortuhr/Multilayout-ESP-Wordclock #138: ESP8266 verbindet sich nicht mit AP

SuperGroova, 2022-02-06:
Hallo zusammen,

ich habe bereits vor 3 Monaten einen ESP8266 NodeMCU V2 mit dem Programm aus dem Repositorie bestückt und es hat alles wunderbar geklappt. Nun wollte ich das Programm auf einen neuen ESP8266 aus der gleichen Bestellung laden. Das Übertragen hat funktioniert und der ESP ist über WLAN zu sehen und auch auszuwählen. Wenn ich nun auf den ESP zugreife und den AP einrichten möchte, gibt der ESP die Meldung, dass keine Verbindung zu dem AP aufgebaut werden konnte. Ich habe es nun schon mit vier verschiedenen Netzwerken versucht. Leider immer ohne Erfolg. 

Beim PlatformIO wird unter Project Task ein Ordner namens Default und einer namens nodemcuv2 angezeigt. Ist es dabei egal welchen dieser Ordner mal als Upload auswählt? Ich habe bereits beide Varianten ausprobiert.

Ich hoffe, dass mir geholfen werden kann. Vielen Dank im Voraus.
Gruß

SuperGroova, 2022-02-17:
Unter der Quellcodeverwaltung wurden 3 automatisch generierte json files angezeigt, welche ich commited habe. Mit dem nächsten Upload hat alles soweit funktioniert. Ich werde die Tage die einzelnen Funktionen noch mal durch testen und würde hier davon berichten.

dbambus, 2022-02-17:
Super, danke :-)

dbambus, 2022-03-02:
Hey @SuperGroova, 

im Wesentlichen kann das Problem am WiFiManager liegen, dessen beta Version wir nach wie vor in den Build einbauen. je nach stand dessen kann sich die AP Geschichte falsch verhalten. Da warte ich aber seit Monaten vergeblich auf einen stabile Version. 

Ich denke das Release 2.6.1 sollte soweit stabil sein. 

Mit freundlichen Grüßen
David

closed 2022-03-02 by dbambus