ESPWortuhr/Multilayout-ESP-Wordclock #141: Dutch

pienmatla, 2022-03-01:
First: i am a noob at programming but i really want to learn. 

Question: If i send you a complete clockface in dutch (with translation etc) can someone change the code for me so it corresponds with the right words? 

Also is there a step by step description of how to make this clock? Thanks

dbambus, 2022-03-01:
Hey @pienmatla, 

how good is your German ? 

It is pretty easy to add a new language. You just have to add a new Uhrtype. For example copy this [.hpp file](https://github.com/ESPWortuhr/Wortuhr/blob/main/include/Uhrtypes/uhr_func_114_Alternative.hpp
) and apply changes for your language. If you've done right, that should be it (90% sure), you can then upload the file to a private fork and test it. 

If you still got problems with the header file, I can set up a private zoom conference to tell you more about the code, if you want. 

Cheers David

pienmatla, 2022-03-01:
Thanks for your respond. 

My German is not really good but good enough to understand the time. The only thing is: in German you say es ist zwanzig nach eins. In Dutch you say "het is tien voor half twee" (es ist zehn vor halb zwei).

Do you have any clue how to fix that? 

Kind Regards, 
Pien

dbambus, 2022-03-01:
Hey @pienmatla,

Maybe this is a helping guide for you :-) 

https://gerben.algemeenbekend.nl/wordclock/images/

Just try to apply your changes as far as you can, I can help afterwards. 

Cheers 
David

pienmatla, 2022-03-01:
Thank you @dbambus! 

Is there a way I can contact you? By mail or something. I will send you some questions. I have made a clockface and am wondering if the ideas i have are realistic 😀

dbambus, 2022-03-01:
Hey @pienmatla,

may be I ve just send you an email. Can you check ? 

Cheers 
David

pienmatla, 2022-03-01:
Hey David,

Pic 1: clockface
Het = es
Is = ist
Vijf = funf
Tien = zehn
Drie = drei
kwart = viertel
over = nach
voor = vor
half = halb
twaalf = zwolf
twee = zwei
een = eins
zeven = sieben
drie = drei
vijf = funf
elf = elf
acht = acht
vier = vier 
negen = neun
tien = zehn
zes = sechs
uur = uhr

Pic 2: the led strips. I will solder them so they are connected

Pic 3: the frame i put on top of the leds

This is what i have so far. I also have an Nodemcu V3. I am struggling how to connect the leds to the nodemcu and with the programming. Like i said, zwanzig nach .. needs to be zehn vor halb ..  ( tien voor half ..). 

I can put the right led addresses in the code but changing the twenty to what it needs to be seems difficult to me. I also dont want to have the 4 leds that will show minutes. I use 110 leds as u see in the pictures. 

Can you show me how to connect the wires of the led to the nodemcu and how to fix the code? :) 

Kind regards, 
Pien

pienmatla, 2022-03-01:
Hi @dbambus,

I got your message and sent u a reply. Thank you very much!

dbambus, 2022-03-02:
Hey @pienmatla,

I added the dutch translation for the layout only in a ned branch https://github.com/dbambus/Wortuhr/tree/main_dutch_added feel free to test it :-) 

David

closed 2022-04-19 by pienmatla