ESPWortuhr/Multilayout-ESP-Wordclock #156: REQUEST: Words for minute indication

pienmatla, 2022-05-10:
Hi,

I was wondering if it's possible to change the minute indication to words instead of single LED's. Example:

12:00 = es ist **nur** zwolf uhr
12:01 = es ist zwolf uhr **gewesen**
12:02 = es ist zwolf uhr **gewesen**
12:03 = es ist **beina** funf nach zwolf
12:04 = es ist **beina** funf nach zwolf
12:05 = es ist **nur** funf nach zwolf

Excuse my German, haha. 

Is there someone who could help me with this, it's a huge request. I know it's maybe a lot of work but I only need some instructions on how to realise this, or anything close :).

pienmatla, 2022-05-19:
Yes, just for fun and someone asked me if i could make it. I cant do it myself but maybe someone of you guys could help me :). I have made a new clockface with the words inside it (in dutch). The only thing i need is how to change the code.

gpmontt, 2022-05-19:
try to download the repo and try it by yourself.
share the clockface in dutch. 
try to replace the minute with your new layout.

dbambus, 2022-05-19:
I will try to implement it into my Code first. Expect a new Branch later the day :-)

dbambus, 2022-05-19:
Hey @pienmatla, 

I Added the options in my branch on https://github.com/dbambus/Wortuhr/tree/Added_Dutch_With_Gewesen ... It should work. Not yet tested.

closed 2022-05-29 by pienmatla