ESPWortuhr/Multilayout-ESP-Wordclock #158: WPS function To connection with wlan

gpmontt, 2022-05-18:
Hallo,
Das ist kein Issue, aber ich wurde gerne implementieren die möglichkeit  zu ein knopf zu drücken und mit das wps wlan zu verbinden 😄. 

Oder ist diese funktionalität schön bereit ? 
Sehr gutes Arbeit 👍🏽

dbambus, 2022-05-19:
Hey @gpmontt,

Ich hatte mal die WPA Funktion implementiert, aber mit dem Umstieg auf https://github.com/tzapu/WiFiManager ist diese leider wieder weggefallen. dennoch denke ich das dieser Schritt richtig war, da man sich so mehr auf die Arbeit am Wordclock Core konzentrieren kann und sich nicht mit dem WiFi FronEnd beschäftigen muss. 

Wenn du willst, kannst du dich aber gerne einlesen wie man beim WiFIManager die WPA Funktion nachrüsten kann. 

Mit freundlichen Grüßen
David

udo1toni, 2022-05-19:
WPS ist gewöhnlich deaktiviert, da es ein Sicherheitsrisiko darstellen kann, Ich frage mich auch, wie oft die Uhr in ein neues WLAN integriert werden soll. Gewöhnlich muss man lediglich ein WLAN Gerät mit dem temporären AP verbinden, die Default Adresse aufrufen, die Zugfangsdaten für das WLAN eintragen und die Uhr neu starten. Ob der Aufwand für WPS hier wirklich lohnt?

closed 2022-12-15 by dbambus