ESPWortuhr/Multilayout-ESP-Wordclock #182: "Uhr" wird zur vollen Stunde nicht angezeigt

bjoern52499, 2022-12-29:
Ich habe den Release 2.6.2 endlich bei mir zum laufen bekommen. Alles funktioniert soweit - bis auf die Darstellung von "Uhr" zur vollen Stunde. Ich habe versucht die Releases untereinander zu vergleichen, finde aber mangels Programmierkenntnisse hierzu keinerlei Lösung. Habt ihr einen Tipp für mich?
In der Weboberfläche kann ich nur "Es ist" ausblenden. "Uhr" jedoch nicht ein- oder ausblenden.

Wenn ich die Zeit manuell auf eine volle Uhrzeit setze, zeigt die Uhr beispielsweise "Es ist Zwölf Uhr" an, wobei "Uhr" aber nach etwa einer Sekunde wieder verschwindet.

Danke euch!

dbambus, 2022-12-29:
Hallo Björn, 

ich kann den Fehler mit meiner Version der Uhr und der manuellen Einstellung der Uhrzeit nicht reproduzieren. ich forsche aber nochmal nach. 

Welchen Uhrtypen hast du ? Welche Spracheinstellungen hast du ausgewählt ? 

Gruß David

bjoern52499, 2022-12-29:
Hallo David,erstmal finde ich es erstklassig das du dir das anschaust. Vielen Dank!Ich habe keine Spracheinstellungen gewählt - die Uhr ist standardmäßig bei mir auf deutsch. Ich habe Uhr114 als Layout und die Zuordnung der LEDs entsprechend meines Layouts angepasst.Mir fiel eben auf, dass „UHR“ zur vollen Stunde zwar angezeigt wird, aber nach etwa 1-2 Sekunden wieder Vers verschwindet.Kann ich noch irgendwas an Input liefern?Liebe GrüßeBjörn

dbambus, 2022-12-29:
Hallo Björn, 

die Darstellung von UHR wird in der Uhr_func.h ab Zeile [554](https://github.com/ESPWortuhr/Wortuhr/blob/main/include/uhr_func.hpp#L554) realisiert.

```
    case 0: // volle Stunde
        usedUhrType->show(uhr);
        voll = 1;
        break;
```

Hier gab es aber keine Änderungen seit der Version 2.6.1. 

Eine Idee von mir wäre, wenn du den ESP8266 einmal komplett löscht (Stichwort erase_flash) und anschließend das Release 2.6.2 nochmal neu aufspielst. 

Eine andere Möglichkeit wäre diese beta Firmware, die ich hier beifüge, drauf zu flashen. Mit dieser Firmware funktionierte es jedenfalls bei mir. 
[release2.6.3Beta.bin.zip](https://github.com/ESPWortuhr/Wortuhr/files/10320546/release2.6.3Beta.bin.zip)

bjoern52499, 2022-12-29:
Hi David,

danke für deine Mühe.
Ich habe in der Uhr_func.hpp (warum hpp und nicht h ist mir schleierhaft?)
nachgeschaut - hier steht das was in Zeile 554 sein soll allerdings in
Zeile 682. Ich habe schon versucht den case 0 ab Zeile 554 zu ersetzen,
dann wird mir beim kompilieren allerdings angezeigt, dass es sich um eine
"read only" variable bei "voll" handelt.
[image: image.png][image: image.png]

Ich habe versucht die von dir angehängte Firmware auf den ESP zu flashen
(wemos Mini D1 sofern das von Belang ist), jedoch bleibt die Uhr dann
hängen und der ESP blinkt nur noch.
Ich hatte vor 10 Jahren in der Technikerschule auch C, C++ etc, verstehe
davon aber nur noch Bruchteile. Für mich sieht es so aus als würde "Uhr"
zwar auf 1 gesetzt, aber nach etwa 1-2 Sekunden wieder auf null
zurückgesetzt. Selbst dann, wenn ich die Zeit manuell im Browser setze. Der
Rest funktioniert komischerweise soweit.
Ich bekomme außer dem 2.62er release kein anderes ans laufen weil dann
bspw. die hash.h fehlt und es ständig kompilierungsfehler gibt.

Kann es denn sein, dass "Uhr" irgendwo fälschlicherweise zurückgesetzt
wird? Ich könnte mir in den Hintern beißen,dass ich im Code nicht
durchblicke...

Danke Dir!
LG
Björn