ESPWortuhr/Multilayout-ESP-Wordclock #189: Neues Layout für Minuten

merlin2533, 2023-01-15:
Wir haben es entsprechend nachgebaut im gleichen Style wie die 114.

Für die Minutenanzeige haben wir ein durchgehendes LED Band genommen mit 7 LEDs statt nur mit 4 einzelne die man zusammen lotet wie die anderen.

Wollt ihr diesen Typ auch auf aufnehmen?

dbambus, 2023-01-15:
hallo @merlin2533,

das hört sich interessant an :-) ich habe heute tatsächlich an dieser Funktion herum gebastelt, aber in eine etwas andere Richtung, siehe letzte commit https://github.com/ESPWortuhr/Wortuhr/commit/6782f125c923e07fe4b5af185346f33d64d5b48a. Hier gehe ich immer davon aus, dass die Minuten als die letzten vier LEDs interpretiert werden. 

Man kann das aber leicht ändern, indem man in /include/Uhrtypes/ die jeweilige Uhr_func.hpp anpasst. Bspw mit: 

```
    uint16_t min_arr[4] = {110, 112, 114, 116};

    //------------------------------------------------------------------------------

    virtual const uint16_t getMinArr(uint8_t col, uint8_t row) override {
        return min_arr[col];
    };
```

mit dem `Overwrite` überschreibt man die hinterlegte Funktion. 

Ob und wie ich das in der Zukunft leider machen kann muss ich mir noch durch den Kopf gehen lassen. Einzige aktuelle Möglichkeit wäre hier ein Pull request mit einem eigenem Layout der diesen overwrite drin hat. :-) dann kann ich das mergen.

dbambus, 2023-01-18:
Wenn du dein Layout als .hpp fertig hast, ggf auch mit schönen Bildern deiner Montage. Dann kann ich das gerne in das Repo aufnehmen. :-) 

Einfach ein pull request erstellen.

merlin2533, 2023-01-29:
Hallo Daniel

Habe mir aus Holz das ganze ausgefräst, sodass ich die LED Strips sehr einfach montieren kann und dem MC
Eifnach melden

melvinisken, 2023-02-04:
Ich habe genau das Gleiche gemacht und bin jetzt erst hierauf gestoßen. Ich habe das versucht, folgendermaßen umzusetzen, vermutlich etwas zu komplex. Ich musste auch noch die ROWS_MATRIX überschreiben, sonst habe ich einen out of bounds error bekommen. Verstehe nicht ganz, wieso das bei der Vorlage (DE10x11.2clock) nicht gemacht werden musste. Egal. 
Jetzt habe ich aber noch das Problem, dass die Minutenanzeige wunderbar funktioniert, allerdings wenn ich die Farbe per Colorwheel ändere, nur die Buchstaben die Farbe ändern, die Minuten nicht. Nach dem nächsten Minuten-Update übernehmen die Minuten die Farbe dann aber. Wo muss ich da noch was ändern?

[DE10x11.2clockIsken.txt](https://github.com/ESPWortuhr/Wortuhr/files/10608413/DE10x11.2clockIsken.txt)

Hier noch ein paar Bilder zur Veranschaulichung:
[image]
Vor Drehen des ColorWheels:
[image]
Nach dem Ändern der Farbe:
[image]
Wenn die nächste Minute kommt, sind dann alle Minuten (in dem Fall) türkis.

melvinisken, 2023-02-04:
Noch ein Nachtrag: bei der Anzeige "Regenbogen" sind auch die falschen LEDs aktiv.
[image]

dbambus, 2023-02-04:
Hallo @melvinisken, Hallo @merlin2533, 

Denn Fehler mit der falschen aktualisierten Farbe konnte ich jetzt nicht reproduzieren. Bei welcher Einstellung tritt der Fehler denn auf? 

Ich habe deine bereitgestellte .hpp angeschaut und euch hier mal eine leicht modifizierte Version bereitgestellt. Auch wieder als .txt und als overwrite für die bestehende 10x11Clock Version.
 
Hier habe ich zwei Dinge geändert: 

1. Ich habe die ROWS_Matrix() overwrite entfernt. Hier kam es in Kombination zu keinen Fehlern. 
2. Zusätzlich habe ich NUM_SMATRIX() overwirte hinzugefügt. Hier muss ich zugeben, dass ist bescheidener Code aus alten Zeiten. Diese Funktion gibt die Anzahl der LEDs in der Frontmatrix plus 4 Min_LEDs wieder. Das wird aktuell nur für die Rainbow cycle Funktion verwendet. Dazu später mehr.

[DE10x11.2clockIsken.txt](https://github.com/ESPWortuhr/Wortuhr/files/10609325/DE10x11.2clockIsken.txt)

Ich bitte euch mal diese Version zu testen und mir Feedback zu geben. Wie ich schon gesehen habe verwendet ihr ja den aktuellsten Fork aus meinem Repo. Wichtig ist, da ist noch ein fieser Bug, wenn man die Animation "Matrix" auswählt. Hier also Vorsicht, sonst landet man in einem Bootloop.

Jetzt nochmal zu den Effekten, wie Regenbogen, etc.. Ich würde das gerne so Coden, dass es die Minuten LEDs nicht mit leuchten, sofern nur Farbspiele ausgeführt werden. das hätte den Vorteil, dass man keine SMATRIX mehr definieren muss und letztlich das MinArray gesondert von der FrontMatrix handhaben kann. Das sorgt dafür das solche Variationen wie jede zweite LED leuchten zu lassen wesentlich einfacher zu implementieren ist und man es vielleicht auch zur Laufzeit im Webinterface ändern kann. 

Was hält ihr davon, damit ich mal eine zweite Meinung bekomme?

Mit freundlichen Grüßen
David

merlin2533, 2023-02-04:
David bei mir läuft alles sauber kann nicht reproduzieren
Aufbau sieht cool aus, ich habe es aus Vollmaterial gefräst, falls jemand eins Brauch melden oder in Etsy verkaufe sie auch

melvinisken, 2023-02-05:
Ich habe gerade nicht ganz so viel Zeit, ich melde mich später noch mal. Ich habe deine Version mal schnell getestet. Das Problem ist da immer noch da. Wenn ich im Modus Wortuhr auf das Farbrad gehe und dort die Farbe verstelle, ändern sich alle Pixel, nur die Minuten nicht. Erst, wenn die nächste Minute anbricht. Ich kann auch noch mal ein Video machen.
Die Änderungen aus deinem File konnte ich nicht übernehmen, nur das override der SMATRIX statt der Cols und Rows. 
Die Sache mit LanguageAbbreviation and dann setLetter statt Letter_set kompiliert nicht. Ich bin auf dem Main Branch vom 1.2. fehlt das da noch?
Und vielleicht noch eine kleine Sache, ich benutze einen D1 mini Clone, vielleicht liegt es daran... Obwohl mich das ein bisschen wundern würde.

dbambus, 2023-02-05:
Hallo @melvinisken, 

ich habe grade das Release 2.7.0 abgeschlossen. Jetzt ist der Stand auf meinem Dev Fork und der Stand hier gleich. Jetzt sollte es mit dem neuen .hpp File ohne Fehler funktionieren. :-)

melvinisken, 2023-02-06:
Mit der neuen Version bekomme ich folgenden Bootloop:
`
--------------------------------------
Begin Setup
--------------------------------------
Version   : 2.7.0
Sernr     : 70
Programm  : 3
Conf      : 96
rgbw.0.0   : 61
rgbw.0.1   : 40
rgbw.0.2   : 5
rgbw.0.3   : 0
rgbw.1.0   : 0
rgbw.1.1   : 0
rgbw.1.2   : 0
rgbw.1.3   : 0
rgbw.2.0   : 61
rgbw.2.1   : 40
rgbw.2.2   : 5
rgbw.2.3   : 0
rgbw.3.0   : 61
rgbw.3.1   : 40
rgbw.3.2   : 5
rgbw.3.3   : 0
Zeitserver: ope.pool.ntp.org
Lauftext  : nie
H6        : 100
H8        : 100
H12       : 100
H16       : 100
H18       : 11
H22       : 255
H24       : 0
LDR       : 25700
LDRCal    : 25700
Uhrtype    : 49
MQTT_State    : 46
MQTT_Server    : 168.4.1
MQTT_User    :
MQTT_Pass    : word
MQTT_ClientId    : ntId
MQTT_Topic    : ordclock
MQTT_Port    : 65535
autoLdrEnabled    : 2
autoLdrBright    : 30
autoLdrDark    : 1
animType    : 0
Uhrtype    : 0
animDuration    : 0
animColorize    : 1
animDemo    : 1
bootLedBlink    : 255
bootLedSweep    : 100
bootShowWifi    : 101
bootShowIP    : 102
Colortype    : 57
OWM_apikey: �������������������������������
OWM_city  : ����

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4020d452 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x4020d452 in setup at ??:?

>>>stack>>>

ctx: cont
sp: 3ffffdb0 end: 3fffffc0 offset: 0190
3fffff40:  3fff2588 00000000 3fff25c4 4020d316
3fffff50:  feefeffe feefeffe feefeffe feefeffe
3fffff60:  feefeffe feefeffe feefeffe feefeffe
3fffff70:  feefeffe feefeffe feefeffe feefeffe
3fffff80:  feefeffe feefeffe feefeffe feefeffe
3fffff90:  feefeffe feefeffe feefeffe 3fff275c
3fffffa0:  3fffdad0 00000000 3fff2748 40227000  
3fffffb0:  feefeffe feefeffe 3ffe8750 40100405  
<<<stack<<<

0x4020d316 in setup at ??:?
0x40227000 in loop_wrapper() at core_esp8266_main.cpp:?
0x40100405 in cont_wrapper at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v00095700
~ld`
Danach geht es wieder von vorne los. Ich werde versuchen, das kurzfristig genauer zu untersuchen.

melvinisken, 2023-02-06:
Ich habe nur kurz gegoogelt, könnte ein Speicherproblem sein. Ob es daran liegt, dass ich nur einen D1 mini benutze oder das zusätzliche Layout eingebunden habe, muss ich noch testen.

dbambus, 2023-02-06:
Hallo Malvin,Wurde vor dem Flashen der Speicher gelöscht ? Esptool erase flash  Mit freundlichen GrüßenDavid

melvinisken, 2023-02-06:
Ja, das war es. Hatte ich vorher noch nie gebraucht :-) Allerdings funktionieren die Minuten jetzt nicht mehr. Also die Minuten 1 und 2 gehen noch, die letzten drei LEDs leuchten dauerhaft weiß und die Minuten 3 und 4 werden nicht angezeigt. Der Farbwähler funktioniert aber immerhin für die beiden vorhandenen Minuten :-)

dbambus, 2023-02-06:
Ich hab jetzt spontan geschaut. Eine Stelle habe ich gefunden, die evtl. ein Problem darstellen kann. Aber das kann kein dauerhafter Grund sein. 

Ich werde mal an meine Dev Uhr paar LEDs zusätzlich anlöten und es mal testen :-)

dbambus, 2023-02-06:
Ich habe es mal fix getestet mit meiner Uhr und paar extra LEDs. 

Die Uhranzeige mit den Minuten funktioniert mit der .hpp die ich oben drüber gepostet habe 
[image]

Was (noch) nicht funktioniert ist das auslassen der LEDs während Farbspielereien wie Regenbogen, etc.

melvinisken, 2023-02-06:
Die hpp hatte ich auch eins zu eins kopiert. Seltsam, ich werde morgen noch mal genau nachgucken. Mehr Änderungen als oben in den Diffs habe ich nicht gemacht..

melvinisken, 2023-02-07:
So, wollte mich jetzt dran setzen und genauer gucken, und jetzt läuft es korrekt. Auf der einen Seite schön, auf der andere Seite natürlich seltsam, solch ein Effekt. Ich werde es mal genauer beobachten und mich ansonsten noch mal zurückmelden.

dbambus, 2023-02-22:
Schließe dieses Issue mal mit verweis auf https://github.com/ESPWortuhr/Wortuhr/issues/201 und dem aktuellem Commit.

closed 2023-02-22 by dbambus