ESPWortuhr/Multilayout-ESP-Wordclock #190: Spiegeln der Anzeige

merlin2533, 2023-01-15:
Eine weitere Funktion für die Layouts waren optimal
Spiegel

Ihr habt die Reverse Funktion entsprechend gecoded diese ist schon top. Optimal wäre jetzt noch zu spiegeln. Denn wenn die LED Streifen verkehrt Rum eingebaut sind dann müsste man sie alle spiegeln.

Dies über das Frontend zu aktivieren wäre genial

dbambus, 2023-01-15:
hallo @merlin2533, 

ich bin im Hintergrund grad daran am werkeln, wird sich aber noch ein Wochenende ziehen :-) letztlich will ich Drehungen um +90,+180 und -90 Grad und spiegeln über die Weboberfläche einbauen. Alles was dazu gemacht werden muss ist Funktionen zu schreiben, welche die Forntmatrix kurz vor der led.set() manipulieren. 

Daher etwas Geduld, kommt aber noch.

merlin2533, 2023-01-16:
Um eine Matrix in C zu spiegeln, kann man eine Schleife verwenden, die über die Spalten und Zeilen der Matrix iteriert und die entsprechenden Elemente vertauscht. Hier ist ein Beispiel für eine Funktion, die eine 2D-Matrix spiegelt:

Copy code
void mirrorMatrix(int matrix[][N], int rows, int cols) {
    for (int i = 0; i < rows; i++) {
        for (int j = 0; j < cols / 2; j++) {
            int temp = matrix[i][j];
            matrix[i][j] = matrix[i][cols - 1 - j];
            matrix[i][cols - 1 - j] = temp;
        }
    }
}

In diesem Beispiel wird die Funktion "mirrorMatrix" mit einer 2D-Integer-Matrix, der Anzahl der Zeilen und Spalten aufgerufen. Innerhalb der Funktion werden zwei Schleifen verwendet, um über die Zeilen und Spalten der Matrix zu iterieren. In jeder Iteration werden die entsprechenden Elemente vertauscht.

Hinweis: "N" ist eine Konstante die vorher definiert wurde und die Größe der Matrix angibt.

merlin2533, 2023-01-16:
Here is an example of C code that rotates a square matrix by 90 degrees clockwise:

Copy code
#include <stdio.h>

void rotate90(int n, int matrix[n][n]) {
    for (int i = 0; i < n / 2; i++) {
        for (int j = i; j < n - i - 1; j++) {
            int temp = matrix[i][j];
            matrix[i][j] = matrix[j][n - 1 - i];
            matrix[j][n - 1 - i] = matrix[n - 1 - i][n - 1 - j];
            matrix[n - 1 - i][n - 1 - j] = matrix[n - 1 - j][i];
            matrix[n - 1 - j][i] = temp;
        }
    }
}

merlin2533, 2023-01-17:
virtual const uint16_t getMinArr(uint8_t col, uint8_t row) override {
        #define LED_LAYOUT_ROTATE 1
        //if (LED_LAYOUT_ROTATE = 1)
            //{
                for (int i = 0; i < sizeof(min_arr[i]); i++) {
                    for (int j = 0; j < sizeof(min_arr[i][j]) / 2; j++) {
                        int temp = min_arr[i][j];
                        min_arr[i][j] = min_arr[i][sizeof(min_arr[i][j]) - 1 - j];
                         min_arr[i][sizeof(min_arr[i][j]) - 1 - j] = temp;
                    }
                }

           // }

        return min_arr[col][row];
    };

dbambus, 2023-01-18:
Hallo @merlin2533,

vielen Dank für den Vorschlag. Ich hatte gedacht, du meinst die gesamte Front spiegeln? Nicht nur das Minuten Array?

Ich hab das nun ein Reverse vom MinArray implementiert, aber etwas anders als deinen Vorschlag, der zweifelsfrei auch möglich ist :-) 

Ich hab das über die std functions von C++ umgesetzt. 

`std::reverse(std::begin(minArray), std::end(minArray));`

Hier meine Implementierung: 
https://github.com/ESPWortuhr/Wortuhr/pull/192/commits/370b72cfa1032dd12f8dbab3428d57022daad73a

Kurze Erklärung: 

Ich versuche, `IF`-Abfragen innerhalb von Uhrentypen zu umgehen und einen allgemeineren Ansatz zu finden, der für alle Fronten gelten kann. Daher habe ich die Funktion `showMinute()` überarbeitet und dort die Option reverse hinzugefügt. Dazu war es notwendig, dass ich an dieser Stelle auf das MinArray zugreifen kann, also habe ich die `getMinArr()`-Funktion entsprechend umgeschrieben. Es wären auch andere Optionen möglich gewesen, aber so habe ich auch die Möglichkeit, die Start-LED des MinArrays zu shiften. Falls ich das noch implementieren möchte :-), aber für Layouts mit Minuten LEDs in den Ecken ist das ein interessanter Gedankengang.

Das Spiegeln der FrontMatrix ist etwas aufwendiger zu implementieren, da ich hier noch etwas am Unterbau der Uhr umcoden muss. Aktuell ist die Variable "frontMatrix" ein lineares Array, daher muss das noch auf ein 2D Array umgestellt werden. :-)

dbambus, 2023-01-19:
Hallo @merlin2533, 

hey ich habe mal an dem Feature weitergearbeitet. Könntest du das mal testen ? Jetzt sollte es möglich sein das Layout über die Website zu spiegeln und zu Reverse / Transponieren ?  

https://github.com/dbambus/Wortuhr/tree/Reversed

Noch ist ein Ein extra Branch in meinem Fork. Ich kann Fehler nicht ausschließen.

dbambus, 2023-03-10:
Hey, ich habe das in https://github.com/ESPWortuhr/Wortuhr/pull/216 gelöst

closed 2023-03-10 by dbambus