ESPWortuhr/Multilayout-ESP-Wordclock #198: Matrix hinzufügen

n3roGit, 2023-02-07:
Hallo zusammen,

es wäre super wenn ihr die angehängte Matrixdatei im Repo hinzufügen könntet. Leider hat meine Wortuhr das Layout von "uhr" um ein Zeichen verschoben. bei mir steht dort dann immer "luh". Also muss ich bei jedem Update erst das Projekt kompilieren und dabei meistens vorher alle Komponenten aktualisieren.

Es muss lediglich in der Datei DE10x11.hpp der folgende Abschnitt angepasst werden:
`        case FrontWord::uhr:
            setLetter(LED_MAP(99));
            setLetter(LED_MAP(100));
            setLetter(LED_MAP(101));
            break;
`

dbambus, 2023-02-07:
Hallo @n3roGit, 

ich kann das machen. Wie soll dieser Typ genannt werden ? bspw. DE10x11.Nero

Ist an dem Buchstaben Layout noch irgendwas anders bzw. brauchst du die defs für REVERSE nicht? 
Wenn ja, kannst du dann die komplette .hpp hier hinzufügen ? Der Upload geht nur als .txt Datei, dass kann ich dann aber abändern. 

Voraussichtlich kommt das in ein Bugfix Release 2.7.1

n3roGit, 2023-02-07:
Ui das ging ja Flott!!
Da ich kein Entwickler bin weiß ich nicht was du mit dem Reverse meinst. Ich hatte nach dem Bau der Uhr mich dran gemacht und diese Stelle angepasst, da nur das Wort "Uhr" bei meinem Layout anders ist. Wenn es hier Abhängigkeiten gibt wäre ich für die entsprechenden Anpassungen natürlich sehr dankbar!

Als Namen würde ich DE10x11.LUH vorschlagen. Wenn jemand mal das selbe Problem hat, wird er dann leicht auf das angepasste Profil stoßen :)

dbambus, 2023-02-07:
Hallo @n3roGit,

Okay, dann mach ich das fix. Wie ist denn das Buchstaben layout ? Bzw. kannst du kontrollieren, ob alle Buchstaben so passen ? 

```
/*
 * Layout Front
 *
 * E S K I S T R F Ü N F
 * Z E H N Z W A N Z I G
 * D R E I V I E R T E L
 * T G N A C H V O R U M
 * H A L B G Z W Ö L F J
 * Z W E I N S I E B E N
 * K D R E I R H F Ü N F
 * E L F N E U N V I E R
 * N A C H T Z E H N B X
 * U S E C H S U H R Y F
 */
```

n3roGit, 2023-02-07:
nicht ganz. 

/*
 * Layout Front
 *
 * E S J I S T L F Ü N F
 * Z E H N Z W A N Z I G
 * D R E I V I E R T E L
 * T F N A C H V O R J M
 * H A L B X Z W Ö L F T
 * Z W E I N S I E B E N
 * L D R E I C I F Ü N F
 * E L F N E U N V I E R
 * U A C H T Z E H N S I
 * C S E C H S T L U H R
 */
 
 
 
[image]

dbambus, 2023-02-22:
Vielen Dank @n3roGit 

Ich habe dein Layout mit https://github.com/dbambus/Wortuhr/commit/e4776b343f800389eabb2f8051825078a7284a44 hinzugefügt. Damit wurde es im nächsten Release enthalten sein. 

Mit freundlichen Grüßen
David

closed 2023-02-22 by dbambus

n3roGit, 2023-02-23:
@dbambus ich habe zu danken! =)