ESPWortuhr/Multilayout-ESP-Wordclock #201: Weitere Matrizen hinzufügen

melvinisken, 2023-02-17:
Wir hatten in dem Issue #189 das ja schon umgesetzt, es wäre super, wenn das Layout mit den sieben extra LEDs sowohl für Englisch als auch für Deutsch übernommen werden könnte. Ich habe mit deinem Projekt ein Projekt für mein Diodenlaser-Wiki erstellt und es wäre super, wenn die Leute ohne zu kompilieren den Code übernehmen könnten. 
Ich habe mal die beiden Dateien genommen und "plusSeven" angehängt, kann man aber auch gerne umbenennen. Das englische Layout habe ich noch nicht ausgetestet, dazu muss ich mir noch eine Uhr bauen, die Erste ist schon weg :-) Da ich meine Seite zweisprachig anbiete und der größere Teil englischsprachig ist, wäre Englisch auf jeden Fall wichtig.

[En10x11plusSeven.txt](https://github.com/ESPWortuhr/Wortuhr/files/10771733/En10x11plusSeven.txt)
[De10x11plusSeven.txt](https://github.com/ESPWortuhr/Wortuhr/files/10771736/De10x11plusSeven.txt)

Zur Info, hier ist die Doku, die ist aber noch nicht offiziell, bislang erst nur runtergeschrieben, wird noch erweitert: https://wiki.the-iskens.com/documentation/projects/project-word-clock-wortuhr/

dbambus, 2023-02-22:
Hallo @melvinisken,

vielen lieben dank für deine Idee :-) und echt cool die Anleitung. Wäre es für dich Vorstellbar, dass ich die Anleitung hier in der ReadMe verlinken oder sogar in einer deutschen Übersetzung hier bereit stellen kann ? 

Okay, nun zu deinem Wunsch:

Im Prinzip kann man die Uhrentypen durch die Möglichkeiten der Vererbung in C++ viel schlanker machen ^^. Hier im Codebeispiel von EN10x11plusSeven kann man statt des Interfaces iUhrtype von En10x11_t erben. So kann man sich alle Funktionsdefinitionen sparen, die im Vergleich dazu gleich geblieben sind. 

```
#include "EN10x11.hpp"

/*
 * Layout Front
 *
 * I T L I S A S A M P M
 * A C Q U A R T E R D C
 * T W E N T Y F I V E X
 * H A L F S T E N F T O
 * P A S T E R U N I N E
 * O N E S I X T H R E E
 * F O U R F I V E T W O
 * E I G H T E L E V E N
 * S E V E N T W E L V E
 * T E N S E O'C L O C K
 *
 */

class En10x11plusSeven_t : public En10x11 {
public:

    //------------------------------------------------------------------------------

    virtual const uint16_t numPixels() override { return 117; };

    //------------------------------------------------------------------------------

    virtual inline const uint16_t numPixelsWordMatrix() { return 117; }

    //------------------------------------------------------------------------------

    virtual const void getMinArr(uint16_t *returnArr, uint8_t col) override {
        for (uint8_t i = 0; i < 4; i++) {
            returnArr[i] = numPixels() - (7 - (i * 2));
        }
    };

};

En10x11plusSeven_t _en10x11plusSeven;
```

Aber ich habe mir etwas anderes überlegt und die letzten Tage den Code etwas  erweitert. 
Und zwar finde ich deine Idee mit den 7 LEDs sehr gut und es ist optisch ansprechend. Also habe ich mir eine Lösung überlegt, wie man diese Option in den Code integrieren kann, ohne spezielle Uhrtypen hinzuzufügen. Stattdessen habe ich mehr LEDs standardmäßig initialisiert und Optionen zur Auswahl der Minuten Darstellung auf der Website und in der config.h hinzugefügt. 
Für weitere Details kannst du dir den Commit unten ansehen. Es wird in der nächsten Version enthalten sein. 

https://github.com/dbambus/Wortuhr/commit/710a1f846d035e370ecb160f551131c747b0500b

melvinisken, 2023-02-23:
`Wäre es für dich Vorstellbar, dass ich die Anleitung hier in der ReadMe verlinken oder sogar in einer deutschen Übersetzung hier bereit stellen kann ?`
Ja, klar, kein Problem. Ich mache die Seite auch eigentlich zweisprachig, deutsch und englisch. Allerdings ist das zum Teil ganz schöner Aufwand, wenn man alles immer doppelt editieren muss. Daher gibt die Anleitung gerade nur auf Englisch und wenn ich denke, dass sich nicht mehr so viel dran ändert, kann ich die einmal übersetzen. Kannst du dann hier reinkopieren oder verlinken.

Cool, das mit der Vererbung. Meine aktiven Programmierzeiten liegen etwas zurück und die Arduinos habe ich immer relativ straigtforward programmiert, ohne große Objektorientierung :-) Das macht's natürlich einfacher. Aber mit dem neuen Code ist es ja noch besser. 

Ich hatte in nem anderen Thread auch schon die Frage zur Übersetzung insgesamt gesehen, das wäre nämlich noch die nächste Geschichte, dass man das Userinterface auf Englisch umstellen kann. Ich hätte jetzt erst mal ganz platt einfach das Template übersetzt und eine kompilierte Version auf Englisch angeboten, aber wenn man das auch auswählen kann, wäre das natürlich besser. Ich könnte da auch die Fleißarbeit im Template übernehmen, wenn ich einmal weiß, wie das gemacht werden sollte (also, wenn du eine Struktur überlegt hast, wie das gehen könnte). 

Und noch eine kleine Idee, aber hat jetzt nicht so viel damit zu tun: wenn man die 7-LED-Variante nutzt hat man ja quasi drei ungenutzte LEDs übrig, die könnte man jetzt ja noch nutzen, um z.B. sowas wie den WLAN-Status oder Alarm AN/AUS, Mini-Wetteranzeige (blau=Regen, gelb=Sonne) anzuzeigen. Man kann ja beliebige Zeichen in die Frontplatte lasern. Aber das wäre etwas "für später" :-)

closed 2023-04-29 by dbambus