ESPWortuhr/Multilayout-ESP-Wordclock #211: Große Wortuhr

Yrrah1960, 2023-03-07:
Hallo zusammen,
erstmal vielen Dank für das tolle Wortuhr-Projekt.
Ich baue auch gerade eine Wortuhr im Format 1 x 1 m. Für die Beleuchtung der einzelnen Buchstuben (außer den Minuten) verwende ich 2 LED´s (Matrix [DE10x11.nero.hpp]). Ich habe mich mal daran versucht die Matrix abzuändern. Das Anzeigen der Uhrzeit klappt, aber alle anderen Funktionen nicht. Da reichen meine Programmierkenntnisse noch nicht aus.
Bisher habe ich die Software von "panbachi" verwendet, die aber leider nicht diese Funktionsvielfalt hat.
Bei panbachi ist die Eingabe der Leistung des Netzteiles möglich und dadurch wird die maximal zulässige Helligkeit der LED´s gesteuert. Vielleicht eine Anregung für eine neue Programmerweiterung.

Könnt ihr mir weiterhelfen?

Gruß

dbambus, 2023-03-08:
Hallo @Yrrah1960,

ja sehr cool, eine große Uhr ist auch noch auf meiner Bucketliste :-). Wenn du fertig bist kannst du ja mal paar Bilder zeigen. 

ich hätte es wahrscheinlich so implementiert, dass ich zwei Stripes parallel verlege und die beiden DI Pins des Stripes parallel am Node anschließe. Ist aber nicht getestet, vielleicht klappte es ja. Nun zu deinem Problem. 

Ich bin grad am umbauen des Backends innerhalb der Uhr, um tatsächlich genau solche Problemstellungen einfach zu triggern, evtl auch über das Webinterface. Aber deswegen kann ich nur bedingt helfen grade. 

Vielleicht wäre einfacher Workaround in der Led.hpp die folgende Funktion: 

```
void Led::setPixel(uint8_t rr, uint8_t gg, uint8_t bb, uint8_t ww, uint16_t i) {

    if (G.Colortype == Grbw) {
        strip_RGBW->SetPixelColor(i, RgbwColor(rr, gg, bb, ww));
    } else {
        strip_RGB->SetPixelColor(i, RgbColor(rr, gg, bb));
    }
}
```

so abzuändern: 

```
void Led::setPixel(uint8_t rr, uint8_t gg, uint8_t bb, uint8_t ww, uint16_t i) {
    
    if (i < 110) {
    i *= 2;
         if (G.Colortype == Grbw) {
            strip_RGBW->SetPixelColor(i, RgbwColor(rr, gg, bb, ww));
            strip_RGBW->SetPixelColor(i+1, RgbwColor(rr, gg, bb, ww));
         } else {
             strip_RGB->SetPixelColor(i, RgbColor(rr, gg, bb));
             strip_RGB->SetPixelColor(i+1, RgbColor(rr, gg, bb));
         }
    }
    else
    {
    i *= 2;
        if (G.Colortype == Grbw) {
             strip_RGBW->SetPixelColor(i, RgbwColor(rr, gg, bb, ww));
        } else {
            strip_RGB->SetPixelColor(i, RgbColor(rr, gg, bb));
        }
    }
}

```

Man müsste dann allerdings auch noch die `numPixel` in der UhrTypeDef, also der `DE10x11Nero.hpp`, auf die neue Gesamtzahl and LEDs (inclusive Minuten) anpassen.

Yrrah1960, 2023-03-08:
[image]

Die Uhr 1x1 ist schon fertig. Da ich auch schon eine in 50 x 50 cm gebaut und deine Software aufgespielt habe möchte ich auch gerne den Leistungsumfang für die große Uhr.
Ich versuche mich mal an deinem Vorschlag.

echo1483, 2023-03-10:
Hallo Yrrah1960, woraus besteht dein Buchstaben-Layout. Für einen Freund habe ich auch eine große Uhr gefertigt 850x850 mm. Das Layout wird in Stahlblech gelasert. Das Problem mit den LED´s habe ich gelöst indem ich LED-Streifen 30 pro Meter verwende und jede 2. ausgelötet habe.  War eine doofe Arbeit, weil ja noch die Brücken einlötet werden musste. LG echo

dbambus, 2023-03-10:
@Yrrah1960 

Das sieht ja richtig stark aus. :-) Wie hat denn mein Vorschlag geklappt?

@echo1483

Ich arbeite für genau diesen Fall an der Software. Ich habe hier vor eine 400x400mm Wordclock zu bauen und will dafür 74 LED/m Stripes verwenden, davon aber nur jede zweite in Benutzung. Dafür will ich eine Softwarelösung entwickeln, die auch mit allen Funktionen kooperiert. 

Mit freundlichen Grüßen
David

echo1483, 2023-03-11:
Ist schon echt groß, wenn die Uhr fertig ist stelle ich mal ein Foto ein. LG echo
[image]
[image]
[image]

Yrrah1960, 2023-03-11:
Hallo zusammen,

meine Blende besteht auch aus einer Stahlplatte, die ich mit Salz/Essigwasser eingesprüht habe, damit sich eine Rostschicht bildet.
Beim Innenleben der Uhr habe ich eine einfache Lösung gewählt. Platte / Leisten aus dem Baumarkt und einen Rahmen gebaut. Danach noch das Raster aus dem 3D-Drucker....
[image]
Dadurch bin ich mit meiner Anzahl an LEDs (30/m) bei 2 Stück pro Buchstaben gelandet.

Das mit dem anpassen der Software hat noch nicht geklappt. Habe die vorgeschlagene Änderung vorgenommen und noch in der DE10x11nero die LEDs die zusätzlich leuchten sollen eingefügt. 

Heute widme ich meinem anderen Hobby
[image]

Gruß und ein schönes Wochenende

dbambus, 2023-03-17:
@echo1483 

Wo hast du denn die Metallplatte machen lassen ? Ich hätte auch an so etwas Interesse :-)

Yrrah1960, 2023-03-20:
@dbambus 
Komme ohne deine Hilfe nicht weiter.
Wo ist dem im Programmcode die Anzahl der LEDs versteckt, bzw. wo wird die Anzahl berechnet aus Zeilen und Spalten?

Gruß aus der Pfalz

dbambus, 2023-07-05:
@echo1483 

ich hätte noch immer Interesse an Esoterik einer Stahlplatte :-) Falls sich da was ergibt, wäre es cool. 

Wenn du planst einen neunen Nachbau der Uhr zu machen habe ich etwas cooles in der Software implementiert. Man kann nämlich angeben, dass jeded zweite LED übersprungen werden soll :-) 

@Yrrah1960 

Ich werde noch an einer Version arbeiten die auch zwei LEDs pro Buchstabe unterstützt. 

Mit freundlichen Grüßen
David

echo1483, 2023-07-10:
@dbambus 
Ich denke das war eine einmalige Aktion, trotzdem wäre es super wenn man selbst die LED-Streifen anpassen könnte. Das auslöten und die Brücken einfügen macht  schon viel arbeit. Für das lasern hat er mit Material 1mm Stahlblech etwa 50,00€ und das Pulverbeschichten noch einmal 50,00€ bezahlt. Es war kein Freundschaftsdienst.

LG echo1483

Yrrah1960, 2023-11-17:
@dbambus 
Erstmal vielen Dank, dass du an der Software weiterarbeitest. Ich verfolge interessiert alle Neuerungen.
Arbeitest du noch an einer Version, die 2 LED´s pro Buchstabe unterstützt?

Ein schönes Wochenende

dbambus, 2023-11-20:
Hallo @Yrrah1960,

Ich habe in der kürzlich veröffentlichten Version implementiert, dass jede zweite LED aufleuchten kann. Außerdem wollte ich versuchen, zwei LEDs pro Buchstabe zu realisieren. Das habe ich probiert, aber aus undokumentierten Gründen hat es damals nicht funktioniert und ich habe es mangels Priorität sein lassen. 

Zu ändern wäre diese Passage in der Uhrtype.hpp: 

```
virtual const uint16_t getFrontMatrixIndex(const uint8_t row, uint8_t col) {

        uint8_t newColsWordMatrix = colsWordMatrix();
        uint16_t numPixelsWordMatrix = rowsWordMatrix() * colsWordMatrix();

        if (G.buildTypeDef == BuildTypeDef::DoubleResM1) {
            newColsWordMatrix = 2 * colsWordMatrix() - 1;
            numPixelsWordMatrix = rowsWordMatrix() * newColsWordMatrix;
            col *= 2;
        }
        if (row % 2 != 0) {
            col = newColsWordMatrix - col - 1;
        }
        uint16_t returnValue = col + (row * newColsWordMatrix);

        if (returnValue > numPixelsWordMatrix) {
            Serial.println(
                "[ERROR] getFrontMatrixIndex() returnValue out of Bounds");
        }

        return returnValue;
    };
```

Hier müsste man noch einen Wert BuildTypeDef::DoubleRes mit entsprechender Logik hinzufügen. 

Mal sehen, vielleicht komme ich noch dazu. 

Mit freundlichen Grüßen
David

NormalZeit, 2025-03-28:
Hallo  @Yrrah1960  und  @echo1483 ,

wie sieht denn Euer Fortschritt mit den Stahlplatten Großuhren aus?

Wir haben hier in Fürth vom FabLab ein Projekt mit einer 4 mm dicken Stahlplatte (Schwarzblech) mit gezunderter Oberfläche in der Größe 980 x 980 mm gerade zur Planung/Umsetzung in der Pipeline.

Erfahrungsaustausch wäre Klasse, das Bild (Fotomontage) stammt aus der Designphase im Februar ;-)

Viele Grüße aus Franken

NormalZeit

[image]

SkylightXD, 2025-03-28:
So sieht meine Uhr aus :-)

[image]

NormalZeit, 2025-03-28:
Klasse!

Wieviele LEDs sind hinter jedem Buchstaben und wie bekommst Du die Ausleuchtung so gleichmäßig hin? Filterfolie?

Wie groß? und was hängt da unten rechts heraus?

Sorry für die vielen Fragen.

SkylightXD, 2025-03-29:
Eine LED pro Buchstabe und Butterbrotpapier als Diffusor. Dahinter die Löcher sind in Kegel oder Zylinderform und weiß angemalt.

80x80 cm oder so. Unten hängt der ESP raus, ich weiß noch nicht, ob die Metallplatte das WLAN Signal platt macht ...

Fragen immer gerne :) wenn ich Zeit finde antworte ich ;)

dbambus, 2025-03-29:
Hey @NormalZeit, 

so von FabLab Erlangen zu Fablab Fürth. Ich denke wir können uns da austauschen ^^

Eingeschlafen ist das Projekt noch nicht, dass zwei LEDs leuchten sollen anstatt einer, aber da ist konzeptionell viel zu machen, damit das mit den Übergängen dann noch hinhaut.

Mit freundlichen Grüßen
David

echo1483, 2025-03-30:
Hallo,
ich habe noch zwei weitere Uhren gebaut, in Cortenstahl die Buchstaben wurden gelasert. Der Stahl wurde dann noch mit einem Rostaktivator behandelt. LG Echo

[image]

[image]

[image]

dbambus, 2025-03-30:
Hey @echo1483,

nach diesen Bildern jetzt brauch, würde ich gerne eine 50x50 Platte haben. Hast du da Vorläge wo man das Lasern lassen könnte?

Mit freundlichen Grüßen
David

echo1483, 2025-03-30:
Hallo David,
Das war nicht teuer, die Firma hat 40,00 € pro Platte genommen. Ich hatte die dxf Datei weitergegeben. Die ganze Aktion wurde durch einen Freund der die Firma kannte ausgeführt. Ob es ein Freundschaftspreis war kann ich nicht sagen. Er hat die Stahlplatten besorgt und ich den Rest. Ich wohne in NRW (Kempen) ich denke es sollte kein Problem sein eine Firma zu finden, ist mittlerweile zum Standard geworden. LG Echo

umakemetilt, 2025-05-15:
> Hey @echo1483,
> 
> nach diesen Bildern jetzt brauch, würde ich gerne eine 50x50 Platte haben. Hast du da Vorläge wo man das Lasern lassen könnte?
> 
> Mit freundlichen Grüßen
> David

Hallo David,

ich hätte eine Anlaufstelle. Wenn du Interesse hast, nehme ich Kontakt zu dir auf. Brauche eh noch 2 weitere kleine in 45x45 und eine weitere Große in 90x90. Ich stehe übrigens auch vor dem Problem, mehrere LED je Buchstabe zu verwenden. BastiMu ist da bereits sehr nahe an einer Lösung. Ich selbst bin leider weit entfernt davon das selbst realisieren zu können. Tolles Projekt!

kidbuster, 2025-06-08:
> Ist schon echt groß, wenn die Uhr fertig ist stelle ich mal ein Foto ein. LG echo [image] [image] [image]

Sehr schöner Rahmen. Kannst du mir evtl. Die fräsdatei überlassen? 😬

echo1483, 2025-06-10:
@kidbuster 

Bitte ddie Datei prüfen ich übernehme keine Garantie.
LG Echo

[Wordclock850.zip](https://github.com/user-attachments/files/20665517/Wordclock850.zip)

kidbuster, 2025-06-17:
> Hallo [@Yrrah1960](https://github.com/Yrrah1960),
>
> Ich habe in der kürzlich veröffentlichten Version implementiert, dass jede zweite LED aufleuchten kann. Außerdem wollte ich versuchen, zwei LEDs [...]

@dbambus 
Das wäre wirklich eine tolle erweiterung, plane grad auch eine große Uhr zwischen 90-120cm (muss noch gucken, wie ich das genau aufteile mit den LEDs) da wäre die Variante 2LEDs pro Buchstabe ideal. denke für die Minuten ist dass wiederrum nicht wichtig.

Danke auf jeden Fall für die tolle Arbeit, die du leistest.