ESPWortuhr/Multilayout-ESP-Wordclock #222: MQTT Frage

Nighthawk2008, 2023-03-29:
Gibt es eine Doku für die Befehle Par Mqtt?

vielen dank.

PS. Super Arbeit!!

udo1toni, 2023-04-01:
Meines Wissens funktioniert mqtt bisher leider noch nicht.

pfefferle, 2023-04-06:
MQTT support wäre großartig! Ich hab die Uhr im Flur stehen und würde gerne, abhängig vom Status von Sensoren in Home Assistant (z.B. "fenster offen", "wasser im keller", ...) die Farbe der Uhr anpassen um zu sehen ob alles in ordnung ist, wenn ich das Haus verlasse oder nach hause komme ;)

Übrigens: Vielen Dank für das großartige Projekt, es läuft wunderbar und hat ein paar schönes Gimmicks :)

dbambus, 2023-04-06:
Hallo an Alle ;-),

ich werde mich der Sache mal annehmen und schauen wo der Wurm drin ist ;-) 

Bin aber grad noch im Urlaub ohne Laptop mit Sonnenschein und 20°C 

Mit freundlichen Grüßen
David

pfefferle, 2023-04-06:
Home Assistant integration funktioniert auch teilweise:

```YAML
light:
  - name: Wortuhr
    unique_id: "wortuhr"
    schema: template
    effect_list:
      - Wortuhr
      - Sekundenanzeige
      - Laufschrift
      - Regenbogen
      - Farbwechsel
      - Farbe
    command_topic: "wortuhr" #topic name
    state_topic: "wortuhr" #topic name
    command_on_template: >
      {"state": "ON"
      {%- if red is defined and green is defined and blue is defined -%}
      , "color": [{{ red }}, {{ green }}, {{ blue }}]
      {%- endif -%}
      {%- if effect is defined -%}
      , "effect": "{{ effect }}"
      {%- endif -%}
      }
    command_off_template: '{"state": "off"}'
    state_template: "{{ value_json.state }}"
    red_template: "{{ value_json.color[0] }}"
    green_template: "{{ value_json.color[1] }}"
    blue_template: "{{ value_json.color[2] }}"
    effect_template: "{{ value_json.effect }}"
```

Ein Problem ist, dass Home Assistant mit "on"/"off" arbeitet, im code aber explizit auf "ON" geprüft wird.

dbambus, 2023-04-16:
Hallo @Nighthawk2008, 

vielen Dank fürs testen. ich habe es selbst Mel probiert und einige Fehler entdeckt die ich jetzt sukzessive beheben möchte. 

Hier ein erster Commit: https://github.com/ESPWortuhr/Wortuhr/pull/231

Hast du denn lust dich auch am Projekt zu beteiligen und die Matt Funktionen auszubauen ? 

Mit freundlichen Grüßen
David

pfefferle, 2023-04-17:
Hallo @dbambus 

schön dass du wieder aus dem Urlaub zurück bist.

Ich hab mir nochmal die MQTT implementierung angeschaut und mir sind nich ein paar unschöne Dinge aufgefallen:

1. https://github.com/ESPWortuhr/Wortuhr/blob/main/include/mqtt.hpp#L70

Hier wird auf "ON" geprüft und in Home Assistant ist "on" der Standard. Außerdem Frage ich mich, warum es dieses Flag braucht um weitere Werte zu verarbeiten.

2. MQTT geht immer davon aus, dass alle Werte geschickt werden. In Home Assistant werden die Werte aber auf verschiedenen "Settings Seiten" angezeigt und einzeln geschickt. Das führt meines Erachtens dazu dass die Uhr abschmiert.

Könnte man den Code ab Zeile 70 nicht folgendermaßen umbauen:

```
    if (doc["effect"]) {
        const char *str_buff = doc["effect"];
        std::string str_buffer = str_buff;
        if (str_buffer.compare("Wortuhr")) {
            G.prog = COMMAND_MODE_WORD_CLOCK;
        } else if (str_buffer.compare("Sekundenanzeige")) {
            G.prog = COMMAND_MODE_SECONDS;
        } else if (str_buffer.compare("Laufschrift")) {
            G.prog = COMMAND_MODE_SCROLLINGTEXT;
        } else if (str_buffer.compare("Regenbogen")) {
            G.prog = COMMAND_MODE_RAINBOWCYCLE;
        } else if (str_buffer.compare("Farbwechsel")) {
            G.prog = COMMAND_MODE_RAINBOW;
        } else if (str_buffer.compare("Farbe")) {
            G.prog = COMMAND_MODE_COLOR;
        }
    }
    
    if (doc["color"][0] && doc["color"][1] && doc["color"][2]) {
        G.color[Foreground] =
            RgbColor(doc["color"][0], doc["color"][1], doc["color"][2]);
        G.color[Foreground].alpha = doc["White"];
    }
```

Ich kenne mich leider in C nicht so wirklich aus und habe keine Ahnung, wie man prüft ob ein Array-Key gesetzt ist (`if (doc["effect"])`). Wenn du mir da etwas unter die Arme greifst, würde ich an einem Pull Request arbeiten.

dbambus, 2023-04-17:
Hey @pfefferle,

in meinem test branch habe ich das so umgebaut: 

```
    if (doc.containsKey("state")) {
        if (!strcmp("on", doc["state"])) {
            G.state = true;
            mqttClient.publish("WordClockReceived", "On");
        } else if (!strcmp("off", doc["state"])) {
            mqttClient.publish("WordClockReceived", "Off");
            led.clear();
            led.show();
            G.state = false;
        }
    }

    if (doc.containsKey("effect")) {
        if (!strcmp("Wordclock", doc["effect"])) {
            G.prog = COMMAND_MODE_WORD_CLOCK;
        } else if (!strcmp("Seconds", doc["effect"])) {
            G.prog = COMMAND_MODE_SECONDS;
        } else if (!strcmp("Scrollingtext", doc["effect"])) {
            G.prog = COMMAND_MODE_SCROLLINGTEXT;
        } else if (!strcmp("Rainbowcycle", doc["effect"])) {
            G.prog = COMMAND_MODE_RAINBOWCYCLE;
        } else if (!strcmp("Rainbow", doc["effect"])) {
            G.prog = COMMAND_MODE_RAINBOW;
        } else if (!strcmp("Color", doc["effect"])) {
            G.prog = COMMAND_MODE_COLOR;
        }
    }

    if (doc.containsKey("color")) {
        G.color[Foreground] =
            RgbColor(doc["color"][0], doc["color"][1], doc["color"][2]);
        G.color[Foreground].alpha = doc["White"];
    }
```

Das sollte ad hoc alle genannten Probleme lösen. Die Frage wäre hier nur noch, ob wir die Funktionalität noch etwas erweitern ? 

bspw:
```        
G.color[Foreground] =
            RgbColor(doc["color"][0], doc["color"][1], doc["color"][2]);
        G.color[Foreground].alpha = doc["White"];
```

Wandelt dieses Stück Code die Farben von RGB Farben in HSB Farben um. Kann HA direkt mit HSB bzw. mit HSBA oder gar HSB + CT umgehen ?

pfefferle, 2023-04-17:
Dein Beispiel sieht schonmal super aus!!!

Ich schau mal ob ich was zu HSB Support für Lichter in Home Assistant finde, bisher habe ich nur mit RGB gearbeitet.

dbambus, 2023-04-17:
Naja HSB hat den ungemeinen Vorteil, dass der Wert für die Helligkeit direkt mit angegeben ist. Ohne das man es erst errechnen muss. Außerdem kann ich damit direkt die HomeKit Implementierung einbauen ^^. 

Was ich noch eingebaut habe ist ein Status von der Wordclock über MQTT, das gibt meines Erachtens auch Sinn, so kann man sehen was angekommen ist und ob die Verbindung überhaupt funktioniert.

pfefferle, 2023-04-17:
perfekt! wird der status über das gleiche topic ausgegeben?

dbambus, 2023-04-17:
Aktuell fix an: "WordClockReceived" dafür ist diese Zeile `mqttClient.publish("WordClockReceived", "Off");` oben im Code verantwortlich. 

Ich wollte nur mal den aktuellen Stand zeigen, ich bin noch am überlegen, wie der Funktionsumfang aussehen soll.

dbambus, 2023-04-19:
Ich habe mit https://github.com/ESPWortuhr/Wortuhr/pull/243 meinen Code dem Hauptbranch hinzugefügt

pfefferle, 2023-04-20:
Ich werde das gleich mal ausprobieren! Dank dir!

dbambus, 2023-04-25:
Ich habe noch nichts negatives gehört, daher würde ich das Issue mal schließen, mit dem verweis auf ein anderes Issue https://github.com/ESPWortuhr/Wortuhr/issues/106, dass die MQTT Funktionen noch erweitert werden.

Nighthawk2008, 2023-05-01:
> Hallo @Nighthawk2008,
> 
> vielen Dank fürs testen. ich habe es selbst Mel probiert und einige Fehler entdeckt die ich jetzt sukzessive beheben möchte.
> 
> Hier ein erster Commit: #231
> 
> Hast du denn lust dich auch am Projekt zu beteiligen und die Matt Funktionen auszubauen ?
> 
> Mit freundlichen Grüßen David

Vielen Dank für dein Angebot. Aber ich glaube das überschreitet leider meine Fähigkeiten.
LG

mrschloemp, 2024-04-07:
Moin,
gibt es mittlerweile eine kleine Anleitung, wie man die Wortuhr mit MQTT in Home Assistant einbindet?
Bin MQTT Neuling und nicht so bewandt in der Anwendung.
Für mich interessant das ein/ausschalten, ideal als Light Sensor, um mit Alexa steuern ebenfalls steuern zu können.

Danke

Pauli29, 2024-04-08:
Ich wäre auch über eine kleine Anleitung dankbar. Denn ich bekomme einfach keine Verbindung :-(

dbambus, 2024-04-10:
Hallo @Pauli29 @mrschloemp ,

mich werde mal Versuchen eine zu schrieben. Bin aber aktuell im Urlaub und kann nur vom Handy aus Supporten. 

Ich werde mich kommende Woche nochmal melden ;-)

Mit freundlichen Grüßen
David

Torrid16V, 2024-04-21:
Hallo zusammen, 
gibt es zur MQTT Einbindung mittlerweile eine Instruktion.
Bei mir will es in HomeAssistant leider auch nicht klappen...

Danke euch.

dbambus, 2024-04-21:
Hey,

ich werde heute Abend noch zwei, drei Bugs ausmerzen, die ich bei der Anleitung schreiben, gefunden habe. Ich muss jetzt das System noch ausgiebig testen. 

Mit freundlichen Grüßen
David

Prinzipiell aber: 

Die MQTT Daten eingeben und !wichtig! MQTT aktivieren und dann erst auf speichern klicken. Anschließend auf HA Discovery klicken. Im Besten Fall sollte dann HA Das Device erkannt haben.

Pauli29, 2024-04-21:
Hey

Ich warte Mal auf deine Doku vielleicht lese ich da ja irgendeinen kleinen Trick ;-)

Denn bei mir wird einfach im HA nichts entdeckt.

Kann es auch nicht denn der MQTT-Eqxplorer sieht auch nichts.

Beste Grüße

Paul Reinhart

"Organization", "name": "GitHub", "url": "https://github.com" } } ]

Torrid16V, 2024-04-21:
> Hey Ich warte Mal auf deine Doku vielleicht lese ich da ja irgendeinen kleinen Trick ;-) Denn bei mir wird einfach im HA nichts entdeckt. Kann es auch nicht denn der MQTT-Eqxplorer sieht auch nichts. Beste Grüße Paul Reinhart
> […](#)

Dito das ist bei mir ebenso das Thema... Danke für die Mühe @dbambus

udo1toni, 2024-04-21:
Ich habe gestern testweise die aktuelle Version geflasht (eher unfreiwillig... egal...) und mqtt ist in der 3.3 ziemlich umgekrempelt.
V3.2: Steuerung der Farbe ausschließlich über ein Tupel `{"color":[r,g,b]}` wobei r,g,b für rot, grün, blau, jeweils dezimal 0 - 255 stehen. Das ist so auch in der mqtt.hpp so dokumentiert.

V3.3: die Farbe wird ausschließlich als hsb gesetzt, wobei h und s als separate Werte innerhalb color gehalten werden, b aber als separater Wert als brightness. Dafür kann man halt die Helligkeit einzeln setzen. Ein gültiges json sieht dann so aus:
`{"state":"ON","color":{"h":h,"s":s},"brightness":b}` wobei h 0 - 360, s 0 - 100 und b 0 - 255 sein dürfen.
Dafür gibt es auch ein status Topic, in welchem im Sekundentakt die Farbe im obigen Format ausgegeben wird. state ist der Zustand der Anzeige (eben ON oder OFF), jeder der drei Werte state, color und brightness ist optional. Setzt man color, müssen sowohl h als auch s angegebn werden.

Ansosnten gibt es noch ein Feld effect und ein Feld marquee_text, ein gültiger Befehl wäre also z.B.
`{"marquee_text":"Hallo Welt! ","effect":"Scrollingtext"}` um den Text **Hallo Welt!** als Laufschrift wiederzugeben.
Ich konnte mich auch davon überzeugen, dass das soweit funktioniert.

Wichtig ist, die Zugangsdaten für den Broker korrekt einzutragen und am besten auch noch mindestens ein, zwei Mal zu prüfen (bei mir hat das Speichern der Werte nicht auf Anhieb funktioniert, und einschalten muss man mqtt auch noch).
Auf der "über" Seite, zeigt Wordclock eine Debug-Nachricht, falls die Konfiguration korrekt geladen wurde.
Mit V3.3 hat man auch einen Knopf, um einmalig eine Home Assistant Discovery zu schicken. Allerdings hat zumindest openHAB es damit nicht geschafft, eine funktionierende Konfiguration zu zaubern. Händisch ging es aber. 
Die V3.3 hat aber - Stand gestern - eine Bug, und zwar werden die Minuten nicht aktualisiert, bzw. nur einmalig, wenn man die Minutenanzeige deaktiviert und wieder aktiviert.

dbambus, 2024-04-21:
Hallo zusammen, 

zunächst einmal vielen Dank an @udo1toni für die ausführliche Erklärung :-) genau so funktioniert es derzeit. Ich habe mich dabei im Wesentlichen an dem orientiert, was [MQTT-Light](https://www.home-assistant.io/integrations/light.mqtt/) von Homeassistant hergibt, um einen einheitlichen Standard zu verwenden. Leider war dafür der Bruch zur vorherigen Version V3.2 notwendig. 

Ich habe heute Abend noch 13 Commits zu dem Projekt gemacht, aber es fehlt noch eine Funktion: Ein automatischer Reconnect nach Verbindungsverlust. 

Aber ich konnte paar Verbindungsprobleme der Uhr in Bezug auf MQTT und die Minutenanzeige beheben :-) Siehe https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/396.

Ich werde die nächsten Tage daran arbeiten, die restlichen Bugs zu beheben und ein fertiges Release zu haben. Heute ist es dafür schon zu spät. 

Mit freundlichen Grüßen
David

pfefferle, 2024-10-21:
@dbambus MQTT läuft fast perfekt mit 4.0, nur wenn ich die farbe ändere setzt er gleichzeitig das licht auf "aus". das heißt ich muss nach jedem farbwechsel danach wieder das licht an machen.

dbambus, 2024-10-21:
Hey @pfefferle, 

bei mir war das nur einmal hab mir dann die MQTT Nachrichten angeschaut und dann ging das wieder. Aber ich habe nichts am Code verändert. Kannst du mal nachschauen von wem der Befehl kommt mit OFF? evtl als cmd vom Homeassistant ? 

Bzw. passiert das auch bei einer manuell gesendeten Nachricht an das cmd topic? 

Mit freundlichen Grüßen
David

pfefferle, 2024-10-21:
Ich hab mal im code geschaut: https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/blob/063cf74812ecdbd019bb8e2d4ba61facd993c162/include/mqtt.hpp#L206C14-L206C20

Ist das nicht genau falsch rum? Er überprüft doch auf "nicht ON" (`!strcmp(state, "ON")`) und setzt dann auf `true`???

pfefferle, 2024-10-21:
und ich glaube bei der farbe muss man noch `color_mode: hs` mit schicken https://www.home-assistant.io/integrations/light.mqtt/#hs-color

dbambus, 2024-10-21:
Ahh, evtl haben wir hier ein andere Fehler Ursache. Funktioniert der Code der 4.0.0 nach einem erase_flash() des microcontrollers und anschließenden neuem Falschen des Chips normal ?

dbambus, 2024-10-21:
> und ich glaube bei der farbe muss man noch color_mode: hs mit schicken https://www.home-assistant.io/integrations/light.mqtt/#hs-color

Das ist nicht nötig, es reicht wenn man HA bei der Einrichtung über HA-Discovery alle Informationen mitgibt. Danach funktioniert es auch ohne 'hs'

pfefferle, 2024-10-21:
> Ahh, evtl haben wir hier ein andere Fehler Ursache. Funktioniert der Code der 4.0.0 nach einem erase_flash() des microcontrollers und anschließenden neuem Falschen des Chips normal ?

werde ich mal probieren!

pfefferle, 2024-10-21:
> Das ist nicht nötig, es reicht wenn man HA bei der Einrichtung über HA-Discovery alle Informationen mitgibt. Danach funktioniert es auch ohne 'hs'

Das Problem ist dann wohl wenn man das Gerät schon mal im HA hatte!?! Egal wie oft ich es schicke, Home Assistant scheint bei RGB zu bleiben, außer ich schicke `color_mode` mit.

dbambus, 2024-10-21:
Kannst du nochmal ein HA-Discovery schicken und neu einrichten, evtl über ein neues Topic ?

pfefferle, 2024-10-21:
> Kannst du nochmal ein HA-Discovery schicken und neu einrichten, evtl über ein neues Topic ?

Alles probiert: Raus gelöscht, anderer Topic, andere Client-ID... nichts hat funktioniert... nur wenn ich einen call mit dem MQTT Explorer mache und `color_mode` mit dran hänge.

pfefferle, 2024-10-21:
Aber in Home Assistant scheint es richtig zu stehen:

```JSON
{
    "brightness": true,
    "color_mode": true,
    "supported_color_modes": [
        "hs"
    ],
    "schema": "json",
    "name": "ESPWordclock",
    "device": {
    },
    "state_topic": ".../status",
    "command_topic": ".../cmd",
    "unique_id": "",
    "plattform": "mqtt",
    "effect": true,
    "effect_list": [
        "Wordclock",
        "Seconds",
        "Digitalclock",
        "Scrollingtext",
        "Rainbowcycle",
        "Rainbow",
        "Color",
        "Symbol"
    ]
}
```

🤔

Pauli29, 2024-10-22:
Ich kann das leider auch bestätigen.

Habe die Uhr nur noch über die Web-Oberfläche wieder zum Leuchten gebracht.

Beste Grüße

Paul Reinhart

dbambus, 2024-10-22:
@pfefferle  @Pauli29 

Wenn man ein neues Topic für die Uhr anlegt und/oder neu bei HA hinzufügt sollte es doch wieder gehen, oder ? 

Mit freundlichen Grüßen
David

pfefferle, 2024-10-22:
Ne, alles probiert... neues topic, neue Client-Id... gleiches Problem... sowohl hs als auch on/off... ich Versuch die Tage nochmal das komplette flushen...

Pauli29, 2024-10-23:
Kann ich leider so bestätigen.

Beste Grüße

Paul Reinhart

dbambus, 2024-10-23:
Der Fehler ist genial. 

Nun, ich habe ihn gefunden. Der Fehler ist in der Funktion `Led::setState()` Zeile 206ff in der led.hpp.

Hier wird bei einem `ON`-Befehl (von MQTT) die Helligkeit von einem zwischengespeicherten Wert an die Uhr zurückgegeben. Das funktioniert einwandfrei. Nur nicht, wenn der zwischengespeicherte Wert vorher noch nie gesetzt wurde, denn dann ist er 0 und damit auch immer die Helligkeit 0, de facto die Uhr `OFF`. Denn die Uhr wertet die Helligkeit als `ON`(color.B != 0) und `OFF` (color.B == 0).

Warum das bei mir die ganze Zeit funktioniert hat, ist relativ einfach erklärt. Wenn ich die Uhr neu flashe, dann teste ich die meisten Dinge mit einem `OFF` Befehl, bevor ich ein `ON` sende. Dadurch wird die aktuelle Helligkeit zwischengespeichert und color.B = 0 gesetzt. Dadurch tritt das Problem beim nächsten `ON` nicht mehr auf. 

Ich werde die entsprechende Zeile Code mit einem initOnFirst Start erweitern. 

Ein Workaround wäre, die Helligkeit über die Website setzen und dann über MQTT ein `OFF` senden. Danach geht alles wie es soll.

dbambus, 2024-10-23:
Vielen Dank an euch beide @Pauli29 und @pfefferle für das melden des Bugs :-)

pfefferle, 2024-10-23:
@dbambus ich kann bestätigen dass dein Workournd funktioniert! ☺️

Jetzt muss ich nur noch schauen ob das mit dem HS farbschema noch hin haut!

dbambus, 2024-10-23:
Fixed with https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/468

Pauli29, 2024-10-23:
Kannst du mir bitte eine Version bauen damit ich über die Website updaten kann das wäre super &#128591;

Beste Grüße

Paul Reinhart

dbambus, 2024-10-23:
@Pauli29 ist schon als Release draußen :-) 4.0.1 die entsprechende Bin kann man dann hier https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/releases/tag/V4.0.1 runterladen

dbambus, 2024-10-23:
Ich schließe mal das Issue, falls es doch noch was neues gibt, dann einfach ein neues aufmachen. :-)

closed 2024-10-23 by dbambus