ESPWortuhr/Multilayout-ESP-Wordclock #232: 16x18 Wortuhr

natti07, 2023-04-17:
Hello i am setting up a 16x18 WordClock. And for this i thought that i use your software. However at present only the first half of the LED is used the 2nd half of the LED with the words for the hours is not used.
it there an version or branch avaliable where this works or there are some updates necessary?
Where i could support

BR Stefan

dbambus, 2023-04-17:
Hello Stefan,

i only implemented it rudimentary, since I don’t have a clock for coding here. But I could help to set it up correctly with your help ;-) 

Mit freundlichen Grüßen
David

natti07, 2023-04-17:
Hallo David
Dann werde ich mir Mal den Code am WE anschauen 

Gruß Stefan

dbambus, 2023-07-05:
hallo @natti07,

in den neusten Commits abbe ich das 16x18 Layout etwas verbessert. Könntest du mit dem Aktellen Stand die Funktionen testen ? 

Mit freundlichen Grüßen
David