ESPWortuhr/Multilayout-ESP-Wordclock #240: Take Timesettings form device

dbambus, 2023-04-19:
Hallo, 

aktuell besteht die Möglichkeit neben dem Timeserver oder einer Batterie RTC die Zeit auch manuell einzustellen. Das Menü dahinter ist allerdings nicht perfekt und es wäre eine coole Idee, wenn es auch die Möglichkeit gäbe die aktuelle Zeit des verwendeten Browsers zu übernehmen? 

Bzw. die aktuelle Zeitzone einstellbar zu machen, die ist ja aktuell hardcoded auf MEZ bzw. MESZ. 

Könntest du dir das ggf. mal anschauen ? @Wandmalfarbe 

Mit freundlichen Grüßen

dbambus, 2023-04-19:
Hallo @Wandmalfarbe,

dann machen wir das so, ich baue im Backend ein COMMAND Call für die Änderung der Zeitzone ein und sag dir welche werte ich erwarte und du könntest es dann im Frontend einbauen ? 

```Die automatisch eingetragene Zeit kann man ganz einfach aus dem Browser ermitteln, und man müsste nur noch speichern.```

Ich denke so wäre es am besten :-)

dbambus, 2023-04-19:
First Issue solved with Pull https://github.com/ESPWortuhr/Wortuhr/pull/242, Thank you :-)

melvinisken, 2023-04-24:
Ich hätte hier noch eine Ergänzung: Wäre es möglich, das Time-Setting auch direkt in das Setup-Menü einzubauen? Sprich, wenn die Uhr kein WLAN findet bzw. den eigenen AP offen hat, kann man sie quasi nicht benutzen. Es wäre super, wenn man dann zusätzlich zu den WLAN-Einstellungen einfach die Uhrzeit einstellen könnte und die Uhr direkt benutzen, auch ohne WLAN. Wenn dann irgendwann WLAN eingestellt wird, darf das dann wieder überschrieben werden, würde ich sagen. Ist natürlich auch weg, wenn der Strom weg ist, aber das wäre egal in so einem Fall (beim Herd muss ich nach nem Stromausfall die Uhr auch neu stellen :-) )

dbambus, 2023-04-25:
Hallo @melvinisken, 

Das habe ich auch schon länger vor, anscheinend gibt es im WifiManager die Möglichkeit custom HTML einzubinden. Aber ich hatte bisher noch nicht genug Zeit mir das in Ruhe anzuschauen. Könntest du dazu ein neues Issue aufsetzen ? Sodass ich anhand des Titels ein Überblick habe ? 

Mit freundlichen Grüßen
David

closed 2023-04-29 by dbambus