ESPWortuhr/Multilayout-ESP-Wordclock #255: Bessere RGBW Darstellung

jojo61, 2023-05-10:
Derzeit wird beim sk6812 die weisse LED nicht genutzt.  Dazu habe ich einen Patch erstellt damit auch die weisse LED genutzt wird.  Damit wird das Farbbild deutlich harmonischer und bei einigen Buchstaben (wo die LEDs nur teilweise zu sehen sind) ist das es nun auch einheitlich weiss.  Anbei ein patch für das aktuelle GIT.

[rgbw_patch.txt](https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/files/11441240/rgbw_patch.txt)

dbambus, 2023-05-10:
Hallo @jojo61,

vielen Dank für den Vorschlag :-) ich werde es mal testen. 

Mit freundlichen Grüßen
David

dbambus, 2023-05-10:
Hallo @jojo61, 

Wenn du die Seite neu lädst, bspw. um die aktualisierten Einstellungen aus dem GRBW-Layout vollständig anzuwenden. Erscheint da der Alpha-Schieberegler unter dem HSL-Wahlfeld? 

Dieser ist bisher für die weiße LED verwendet worden, als extra "Farbe". Ich habe den Anwendungsfall, dass ich gerne nur Weiß mit etwas Orange mixe. Eine reine Alpha Channel Farbe ist mit deinem Patch etwas umständlich zu erreichen, oder ?

jojo61, 2023-05-10:
Mit dem Patch ist ein unabhängiges "alpha" nicht mehr einstellbar. Die weisse LED ist ja nicht dazu da noch ein Farbelement dazuzumischen, sondern um das Weiss darzustellen (so sehe ich das). Dazu wird der Weissansteil aus den RGB Teilen rausgerechnet (maxmaler gleicher Wert der RGB Teile) und der weissen LED zugeordnet.  Je nach Farbtemperatur der weissen LED sieht das natürlich anders aus als das weiss wenn man es per RGB erzeugt.
Bei mir sieht das bei RGB erzeugte weiss nach kaltweiss 6500K aus.  Das mag ich aber nicht so, und das weiss der LED ist warmweiss (4000K) das gefällt mir bessser. Für mich sieht es mit dem Patch einfach harmonischer aus.  Aber letztlich müsst ihr den ja nicht übernehmen.  Ich habe den ais FASTLED übernommen und der ist auch in WLED drin.

dbambus, 2023-05-10:
Hallo @jojo61,

das war nicht als Kritik gemeint :-). Ich finde es super, wenn solche konstruktiven Vorschläge kommen und neue Anregungen präsentiert werden. 

Ich hatte zum Zeitpunkt der Umstellung auf HSB-Farben genau so etwas im Sinn, dass man die HSB-Farben direkt in RGBW umwandelt. Ich hatte es nur aus Zeitmangel verworfen. Im Prinzip möchte ich auch vom Alpha Slider wegkommen. Stattdessen könnte man einen Farbtemperatur-Slider für eine Mischung aus RGBW und CCT einfügen.

https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/compare/main...dbambus:Wortuhr:RGBW_TrueWhite

Hier wäre ein Vorschlag einer Implementierung (bisher nur RGB to RGBW) von mir, den ich mal im Auge hatte. Die geänderten Codezeilen sind angelegt an das DiyHue-Projekt, welches Ähnliches programmiert hat. Nur von RGB auf RGBW. Wie findest du diese Idee, vielleicht findet man ja eine Conversion von HSB zu RGBW unter Berücksichtigung der Farbtemperatur der weißen LED. Ich denke, man sollte dazu auf der Website eine Einstellungsmöglichkeit hinzufügen, in der man angeben kann, welche Art von weißer LED (WW, NW, CW) auf dem SK6812 installiert ist.

Guten Abend 
David

dbambus, 2023-07-01:
Added with https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/288

closed 2023-07-01 by dbambus