ESPWortuhr/Multilayout-ESP-Wordclock #297: Kleine Verbesserungen an der Website

dbambus, 2023-07-15:
Hallo @Wandmalfarbe 

ich habe mal wieder an der Wordclock-Software gearbeitet und ein paar Funktionen hinzugefügt. Es ist Tagsüber einfach zu warm und ich hatte mal wieder lust weiter zu machen. Beim Bugtesten ist mir etwas auf der Website aufgefallen. 

- Bei dem Helligkeits- und Geschwindigkeitsregler in Funktionen->Rainbow (und anderen) "rutscht" der Regler bei Zahlenwerten <10. Könnte man diesen Slider möglicherweise fixieren? 

- Ich habe den Alphakanal-Slider durch einen Farbtemperatur-Slider ersetzt. Für eine bessere Benutzerfreundlichkeit könnte man vielleicht einen zentrierten Text über dem Slider auf der Hauptseite einfügen wie: Helligkeit oder Farbtemperatur. 

- Ich habe bereits eine relativ große Anzahl von Funktionen/Betriebsarten für die Uhr. Was hältst du von einem 2x4 Raster mit Piktogrammen für die Modi anstatt der Radio Buttons ? Liese sich sowas leicht implementieren ? 

Mit freundlichen Grüßen
David

Wandmalfarbe, 2023-07-25:
Hi @dbambus,

habe gerade recht wenig Zeit, wieder an meiner Uhr zu basteln. Hier nur kurz meine Gedanken:

* Ein merkwürdiges Rutschen ist mir bei einigen Reglern Regler auch schon aufgefallen. Auch hatte ich das Gefühl, dass manche Zahlenwerte bei der Geschwindigkeit des Farbwechsels auch gar nicht angewendet werden? Ich weiß jetzt nicht genau, was du mit fixieren meinst, aber da könnte man mal untersuchen, warum die Slider sich so merkwürdig verhalten.
* Weiß auch gerade nicht genau, welchen Alphakanal Slider du meinst ... aber ich bin noch auf Version 3.0.0. Grundsätzlich sollten alle Formularfelder und generell auch Controls ein Label haben. Muss mal schauen, wie man da was hinzaubern kann.
* Ja, die Radio-Optionen werden einfach zu viel und zu unübersichtlich. Ich denke auch, Icons im Grid sind da die bessere Wahl. Allerdings sollten wir die Radios und den Text dazu nicht ausblenden. Ich komme selber immer schon durcheinander zwischen Regenbogen und Farbwechsel. Icons sind auf jeden Fall relativ einfach möglich (siehe https://stackoverflow.com/questions/17541614/use-images-instead-of-radio-buttons).

dbambus, 2023-11-12:
Hallo @Wandmalfarbe, 

Es ist mal wieder die dunkle Jahreszeit :-) nach einer langen Pause setze ich mich mal wieder an den Code. Gerne lade ich dazu ein mitzumachen.  

Mit freundlichen Grüßen
David

dbambus, 2024-08-26:
Hallo @Wandmalfarbe 

Ich möchte mich herzlich für die Änderungen in deinem Pull-Request bedanken. Sie sind wirklich sehr willkommen und tragen zu einer wesentlichen Vereinfachung der Usability bei. Ich werde sie zeitnah mergen.
Vielen dank, das du an dem Projekt mitwirkst. 

https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/431

Mit freundlichen Grüßen
David

closed 2024-10-27 by dbambus