ESPWortuhr/Multilayout-ESP-Wordclock #301: Ein paar Probleme...

udo1toni, 2023-07-17:
1. Nachdem ich die neue Version 3.1.1 über 3.0.0 installiert habe (eingebauter Update-Manager) ist die Web UI quasi nicht mehr bedienbar (incl. Updater)
2. Bisher habe ich Updates immer über den Updater eingespielt. Nun versuche ich also, das über PlatformIO zu machen, aber das Projekt wird nicht fertig gebaut, es gibt stattdessen Warnmeldungen:
'''
[{
	"resource": "/c:/Users/hartm/source/wortuhr/Multilayout-ESP-Wordclock/.pio/libdeps/nodemcuv2/PubSubClient/src/PubSubClient.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "comparison of integer expressions of different signedness: 'unsigned int' and 'int' [-Wsign-compare]",
	"startLineNumber": 523,
	"startColumn": 16,
	"endLineNumber": 523,
	"endColumn": 16
},{
	"resource": "/c:/Users/hartm/source/wortuhr/Multilayout-ESP-Wordclock/.pio/libdeps/nodemcuv2/WebSockets/src/WebSocketsServer.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "'WiFiClient WiFiServer::available(uint8_t*)' is deprecated: Renamed to accept(). [-Wdeprecated-declarations]",
	"startLineNumber": 649,
	"startColumn": 96,
	"endLineNumber": 649,
	"endColumn": 96
},{
	"resource": "/c:/Users/hartm/source/wortuhr/Multilayout-ESP-Wordclock/include/config.h",
	"owner": "cpp",
	"severity": 4,
	"message": "unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '311' to '55' [-Woverflow]",
	"startLineNumber": 135,
	"startColumn": 15,
	"endLineNumber": 135,
	"endColumn": 15
}]
'''
Das ist irgendwie ungünstig... Wie bekomme ich denn das Binary einfach per USB auf den D1 Mini drauf?

dbambus, 2023-07-17:
Hallo @udo1toni,

Ich hatte bisher nur Probleme, wenn die Seriennummer von der neunen Build und der alten Build gleich ist, deshalb muss man sie immer ändern. 
Die SRNR `"message": "unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} changes value from '311' to '55' [-` ist mit 311 ziemlich unglücklich gewählt. Hier könnte man einen quasi beliebigen uint8 Wert ind er config.h eintragen (nur nicht 30 wie in Version 3.0.0), um die Fehlermeldungen zu umgehen.

Wieso das Projekt nicht über PlatformIO baut, verstehe ich nicht. Es wurde ja extra ein Text hier auf Github eingebaut, damit immer sichergestellt wird, dass die Software auch gebaut werden kann. Eventuell mal Platformio Clean ausführen. 

Mit freundlichen Grüßen
David

udo1toni, 2023-07-17:
ok, Clean hat geholfen :) Danke dafür!
Und wenn man dann noch ein USB-Kabel nimmt, welches auch die Datenleitungen hat, funktioniert sogar der Upload per USB...

Aber:

Leider gibt es reproduzierbares Fehlverhalten.
1. Setzen eines Hostnamen führt beim Speichern zum Reset. 
2. Der Hostname ist anschließend nicht geändert, lautet also fix ESPWordclock - unschön. 
3. Leider werden auch die Einstellungen für MQTT bei der Gelegenheit gekillt. Erhalten bleiben aber WLAN und andere Eigenschaften, aber:
4. Nach einem Neustart werden auf der Einstellungsseite die Standardwerte angezeigt, statt die gesetzten Werte zu zeigen. Das betrifft sowohl den Zeitserver (egal ob per fqdn oder per ip) als auch den LED-Typ.

dbambus, 2023-07-17:
Hehe :-) Klassiker, aber cool das es jetzt doch funktioniert. 

Ich werde mir mal anschauen, was das für Fehler sind und wo der Ursprung ist :-) Danke für die Bug Auflistung. Gerne auch  Vorschläge für die Uhrsoftware :-).

udo1toni, 2023-07-17:
Es scheint so, dass die Einstellungen für mqtt nicht gespeichert werden (hab gerade den ESP wieder in die Uhr verpflanzt)

melvinisken, 2023-08-06:
Kann ich unterstützen, habe gerade eine alte Uhr auf 3.1.1 geupdated und leider wird das Layout auch nicht gespeichert. Nach jedem Neustart ist das Layout wieder falsch.

dbambus, 2023-08-07:
Hallo @melvinisken, 

weiß du noch Tele Version davor drauf war ? Hast du mein Build zu flachen verwendet ? 

Gruß David

melvinisken, 2023-08-08:
Ich hatte vorher die 2.7 drauf, wenn ich mich recht entsinne. Ich hatte die binary von der releases Seite genommen. Bin jetzt zurück auf 3.1.0 und die funktioniert.

pfefferle, 2023-08-25:
Bei mir sind alle Einstellungen nach einem Neustart auch wieder weg. Ich benutze Version 3.1.1.

dbambus, 2023-09-11:
Hallo an alle, 

nach langer Zeit habe ich mir das mal angeschaut. Schuld ist wohl die Seriennummer in der config.h. Hier hatte ich angenommen, dass man 16 Bit Zahlen verwenden kann, aber anscheinend funktionieren nur 8 Bit Zahlen. Hier sollte man lieber eine Zahl <255 ein. 

Ich muss das noch testen, aber das scheint am Wahrscheinlichsten zu sein. Ein entsprechender Fix kommt, wenn ich das getestet habe. 

Mit freundlichen Grüßen
David

irqnet, 2023-11-09:
> Hallo an alle,
>
> nach langer Zeit habe ich mir das mal angeschaut. Schuld ist wohl die Seriennummer in der config.h. Hier hatte ich angenommen, dass man 16 Bit Zahlen verwenden kann, aber anscheinend [...]

Gibt es hierzu schon ein Update? Hänge genau am gleichen Problem, würde ggfs. auf ein älteres Release ausweichen wenn möglich.

Habe parallel aber auch mit diversen Seriennummern probiert, ohne großen Erfolg.

dbambus, 2023-11-09:
Hey, 

du hast es schon mit anderen Seriennummern probiert, die jeweils kleiner waren als 255 ? Hast du davor mal den ESP über -erase-flash geleert ? 

Das hört sich sehr mysteriös an. 

ich habe nach langer Sommerpause erst wieder vor paar Tagen mit der aktiven Weiterentwicklung begonnen. Vielen Dank für den Hinweis, dass dieses Problem nach wie vor besteht. 

Mit freundlichen Grüßen
David

irqnet, 2023-11-09:
> Hey,
>
> du hast es schon mit anderen Seriennummern probiert, die jeweils kleiner waren als 255 ? Hast du davor mal den ESP über -erase-flash geleert ?
>
> Das hört sich sehr mysteriös an.
>
> ich habe [...]

Ich würde mal behaupten, ja.

Habe jetzt zum Test mal 3.1.0 geflasht, da existiert das Problem nicht.

dbambus, 2023-11-12:
Es sollte jetzt mit dem neunen Release 3.2 behoben sein 
https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/releases/tag/V3.2

closed 2023-11-12 by dbambus

irqnet, 2023-11-12:
> Es sollte jetzt mit dem neunen Release 3.2 behoben sein 
> https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/releases/tag/V3.2

Kann ich bestätigen. Funktioniert.

pfefferle, 2023-11-13:
Funktioniert! Danke @dbambus !