ESPWortuhr/Multilayout-ESP-Wordclock #331: Anzeige von Symbolen

irqnet, 2023-11-23:
Im Webinterface kann man ja die Anzeige auf Symbol umschalten, hier gibt es allerdings keine Auswahl, welche Symbole existieren. Ist das irgendwann mal rausgeflogen? Im Code sieht man ja zumindest den Verweis auf das Smiley Icon.

[image]

dbambus, 2023-11-23:
Hallo @irqnet,

eigentlich ist das Gegenteil der Fall :-). Es gibt schon seit Ewigkeiten zwei "Symbole" im Code, die aber nie aktiv genutzt wurden. Kurz vor dem Release 3.1 habe ich das gesehen und spontan das Herz mit einem Herzschlag implementiert. 

Grundsätzlich würde ich es begrüßen, wenn es mehrere Symbole gäbe, wie: Sprechblase, Musik, eventuell auch das aktuelle Wetter als Symbol. Wenn jemand die Symbole für mich generiert, könnte ich sie ohne weiteres zusammenfügen. Ich müsste mir aber überlegen wie man sinnvoll mehrfarbige Symbole organisiert. 
Im Zuge dessen würde ich auch das Menü ein wenig anpassen, wie hier beschrieben https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/issues/297

Ich arbeite in meiner Freizeit aktuell an V3.3 mit einer Home-Assistant-Integration, die fast fertig ist, siehe https://github.com/dbambus/Wortuhr. In diesem Zusammenhang könnte man z.B. ein Nachrichtensymbol einblenden, wenn man eine Nachricht auf seinem Handy erhält. 

Mit freundlichen Grüßen
david

irqnet, 2023-11-23:
Klingt spannend, wenn ich ein wenig Zeit finde kann ich da gerne mal was zusammen basteln.

Sehe ich das richtig das wir nur eine Ergänzung in der icons.h brauchen?

https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/blob/main/include/icons.h

Homeassistant Integration klingt sehr spannend :) bin auch dort sehr aktiver Nutzer.

dbambus, 2023-11-23:
Genau, 

in der Icons.h kann man einfach weitere "Symbole" an das Ende des Arrays hinzufügen. Animierte Varianten, könnten als Abfolge von Einzelbildern realisiert werden, analog zum Heartbeart.  

Aktuell würden ja nur monochromatische Versionen funktionieren, ich denke deswegen ist Smiley nie fertig implementiert worden, weil ich das gerne mit Farben realisiert hätte. Hierzu habe ich aber noch keine sinnige bzw. speicherschonende Möglichkeit gesehen. 

Zwecks HA: Es kann gerne getestet werden, für HA verwende ich die MQTT Schnittstelle und die Discovery Funktion ist auch schon implementiert. Einzig muss man aktuell noch drauf achten den Toipc Namen nicht zu lang zu wählen.

irqnet, 2023-11-23:
Ok danke für die Info.

Bzgl. MQTT bekomme ich gar keine Verbindung zum Broker hin. Sind Sonderzeichen im Passwort evtl. problematisch?

dbambus, 2023-11-23:
Das muss ich erst testen, ein Lehrzeichen könnte unter Umständen ein Problem darstellen.

irqnet, 2023-11-23:
> Das muss ich erst testen, ein Lehrzeichen könnte unter Umständen ein Problem darstellen.

In meinem Fall ein # am Ende des Passwortes.

dbambus, 2023-12-03:
Ich hab mir das mal angeschaut. 
Das Passwort wird korrekt von der Webseite geparsed. Eventuell kann es ein Problem der PubSubClient.h sein. Das kann nicht aber nicht mehr Debuggen.

Mit freundlichen Grüßen
David

dbambus, 2024-04-18:
Hey, 

ich würde dieses Issue gerne schleißen, da ich aktuell das mit den Symbolen in einem anderem Issue weiter bearbeite und den Fehler mit dem # im Passwort eventuell jetzt korrigiert habe mit den Änderungen die ich am MQTT backend gemacht habe. Ein Upload kommt heute Abend. 

Mit freundlichen Grüßen
David

closed 2024-04-18 by dbambus