ESPWortuhr/Multilayout-ESP-Wordclock #333: SP511E LED Controller (WLED Kompatibel)

irqnet, 2023-11-29:
Hallo,

ich habe gerade aus Interesse ausprobiert ob die Wordclock binary auch auf einem alternativen Controller als dem NodeMCU oder WEMOS D1 mini läuft.

Ich hatte noch ein paar SP511E LED Controller hier liegen, mit denen ich ein paar WLED Projekte umgesetzt habe und habe darauf die Wordclock binary geflasht.

Erstaunlicherweise funktioniert nach Anschluss des Netzteiles und des LED Streifens alles wie gewünscht :)

Man hat hier also für einen schmalen Taler (8-9 EUR bei AlieExpress) einen Controller der direkt in einem Gehäuse kommt und mit einem Standard 5V Netzteil via Pfostenstecker betrieben werden kann.

[image]

Die Anschlüsse zum flashen sind auf der Platine frei zugänglich, eine Anleitung findet man unter folgendem Link:

https://github.com/scottrbailey/WLED-Utils/blob/gh-pages/sp511e_wled.md

[image]
[image]

Interessant ist, das der Controller drei Buttons sowie einen IR Emfpänger für die Fernbedieunung hat, ggfs. könnte man dies auch noch implementieren.

Belegung ist laut Beschreibung:

- LED PIN: GPIO3
- IR PIN: GPIO5
- POWER BUTTON: GPIO2
- M Button: GPIO 13
- S Button: GPIO 14

dbambus, 2024-08-26:
Hey @irqnet, 

Ich habe das Projekt jetzt eine ganze Weile aufgeschoben, aber jetzt habe ich es mir doch noch einmal angeschaut. Ich glaube, dass dieses Board einen anderen Pin für die LEDs verwendet, als es die Library ansteuern könnte, die ich verwende. Daher sind Änderungen in Form von Löten notwendig, damit das mit diesem Board funktioniert, oder ? 

Nichtsdestotrotz werde ich die zugrundeliegende Idee aufgreifen und eine Steuerung über Buttons vorschlagen, damit man einen analogen Steuermechanismus hat. Das ist wirklich eine super Idee. :-) 

Mit freundlichen Grüßen
David

irqnet, 2024-08-26:
Hi David,

ich kann betätigen das es läuft :)

Vg

dbambus, 2024-08-26:
Hey @irqnet, 

okay, dann muss ich mir wohl einen kaufen xD. 

Mit freundlichen Grüßen
David

irqnet, 2024-08-27:
Der Vollständigkeit halber der Link vom Shop wo ich die Controller bestellt habe:

https://de.aliexpress.com/item/1005004989435773.html

Wichtig ist; es muss die Variante SP511E One CH sein.

dbambus, 2025-04-23:
Ich habe mal hier was dazu geschrieben https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/574#issuecomment-2823340901