ESPWortuhr/Multilayout-ESP-Wordclock #337: V3.3 Farbe wird nicht gespeichert

melvinisken, 2023-12-06:
Nachdem der allgemeine Bug mit der Speicherung raus ist, scheinen zwar alle Einstellungen gespeichert zu werden, aber die Farbe der Anzeige wird nicht gespeichert. Nach jedem Reboot ist die Uhr wieder auf Grün eingestellt.

dbambus, 2023-12-06:
Hallo @melvinisken, 

Okay, also das mit der nicht gespeicherten Farbe hatte ich bisher nicht auf dem Schirm, ich dachte das wäre mit V3.2 behoben.

Ich konnte aber das Problem auffinden, denn ein `eeprom::write();` fehlt nach Änderung der Farbwerte. Quickfix: Trage in WebPageAdapter.h in Zeile 200 vor dem break; ein 'eeprom::write();' ein. Das sollte das Problem beheben. 

In der finalen Version muss ich das Sichern der aktuellen Werte aber an einer anderen Stelle hinzufügen, in WebpageAdapter ist das dauerhaft nicht gut angelegt.

Zum anderen Problem: Der aktuelle Stand ist noch nicht final getestet. Das der WiFi-manager nicht geht ist ärgerlich, aber hier verwende ich nur eine vorhandene Lib. Diese habe ich jetzt fest auf einen spezifischen Stand festgelegt, der womöglich älter ist als der von V3.2. Da kann es zu Fehlern kommen. 

Ich habe die Hoffnung, dass es jetzt besser wird. 

Mit freundlichen Grüßen
David

melvinisken, 2023-12-06:
Super, danke für die schnelle Antwort. Das teste ich gleich mal. 
Bei dem WiFi-Manager bin ich mir auch nicht sicher, ob nicht vielleicht bei OTA der Flash nicht ordentlich gelöscht wurde oder so. Die Buttons auf der ersten Seite kamen auch noch, OTA hätte vermutlich auch noch mal funktioniert, nur wenn man auf Wi-Fi-Settings geklickt hat, kam nur eine leere Seite.

melvinisken, 2023-12-06:
Ja, das Problem mit dem Speichern der Farbe ist damit gelöst :-) Hat geklappt.

irqnet, 2023-12-15:
> Super, danke für die schnelle Antwort. Das teste ich gleich mal. Bei dem WiFi-Manager bin ich mir auch nicht sicher, ob nicht vielleicht bei OTA der Flash nicht ordentlich gelöscht wurde oder so. [...]

Das hab ich tatsächlich auch öfter, und ich kann das nur lösen (unter Android) wenn ich die IP der Uhr (192.168.4.1) direkt im Browser eingebe und nicht über den intergrierten Aufruf bei der WLAN Verbindung. Manchmal muss man die Seite aber auch mehrfach neu laden.

Keine Ahnung ob die Version des Wifi Managers da ein Problem hat?

dbambus, 2023-12-31:
Fixed with https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/346/commits/87c36a837207dffa712b85ceaf01fa7b4537a4d4

closed 2023-12-31 by dbambus