ESPWortuhr/Multilayout-ESP-Wordclock #342: Wifi Manager settings reset

irqnet, 2023-12-18:
Ist es möglich die Einstellungen komplett auf Werksreset zurück zu setzen ohne Zugriff auf die Webseite?

Ich habe das Problem das ich zwar eine IP aus meinem Heimnetz bekommen habe aber die Uhr nicht mehr erreichbar ist.

Bei tasmota oder ähnlichen esp Firmwares kann man glaube ich mit 5x power cycle den AP Mode starten?

dbambus, 2023-12-18:
@irqnet 

Sowas in der Art habe ich bisher noch nicht implementiert. Aber das wäre etwas für die Zukunft. 

Mit freundlichen Grüßen
David

dbambus, 2023-12-18:
Ja, leider schon.

irqnet, 2023-12-18:
Da wären die Buttons von meinem SP511 Controller ja prädestiniert :)

dbambus, 2023-12-18:
In der tat, das wäre auch super Leicht zu implementieren ^^ gerne mal ein Vorschlag machen ^^

irqnet, 2023-12-18:
Ok meine Programmierskills halten sich für die Integration in Grenzen :) Aber ein erster Versuch, zumindest die Buttons einzubinden, noch ohne Code. Evtl. kann man includen abhäning von einer Variable? z.b. 

Controller=SP511E, WEMOS, NODEMCU etc.

Die Buttons gibts ja default mäßig bei einem Wemos/NodeMCU nicht, man könnte aber freie GPIOs definieren.

Für den SP511E sähe das aktuell so aus:

`// Define the GPIO pins for the buttons
const int buttonPin1 = 2;
const int buttonPin2 = 13;
const int buttonPin3 = 14;

// Add more button pins as needed

void setup() {
  // Initialize serial communication for debugging
  Serial.begin(9600);

  // Set button pins as inputs
  pinMode(buttonPin1, INPUT_PULLUP);
  pinMode(buttonPin2, INPUT_PULLUP);
  pinMode(buttonPin3, INPUT_PULLUP);
  // Set additional button pins as inputs

}

void loop() {
  // Read the state of the buttons
  int buttonState1 = digitalRead(buttonPin1);
  int buttonState2 = digitalRead(buttonPin2);
  // Read additional button states as needed

  // Check if button POW is pressed
  if (buttonState1 == LOW) {
    Serial.println("Button POW pressed");
    // missing code for button action
    delay(500);  // debounce delay
  }

  // Check if button M is pressed
  if (buttonState2 == LOW) {
    Serial.println("Button M pressed");
    // missing code for button action
    delay(500);  // debounce delay
  }
  // Check additional buttons as needed

  // Check if button S is pressed
  if (buttonState3 == LOW) {
    Serial.println("Button S pressed");
    // missing code for button action
    delay(500);  // debounce delay
  }

}
`

dbambus, 2024-04-18:
Hinzugefügt mit https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/394

closed 2024-04-18 by dbambus