ESPWortuhr/Multilayout-ESP-Wordclock #356: 16x18 Layout Fehler

Micha7719, 2024-01-15:
Hi, ab dem Release V3.1.0 zeigt die Digital Clock und die normale Wortuhr bei dem 16x18 Layout 
nur Kauderwelsch an.

Es fehlen dann Worte wie " Vor, Minuten...usw."

Bei 3.0 funktioniert es noch.

dbambus, 2024-01-15:
Hallo @Micha7719,

das Problem liegt wohl daran, dass ich aktuell keine 24H Uhr hier habe, auf der Ich Debuggen. Ich werde mir mal was einfallen lassen wie ich das hinbekommen.  Das kann aber bis zum Wochenende Dauern. 

Mit freundlichen Grüßen
David

Micha7719, 2024-01-15:
danke, die 3.0 zeigt nach ner Zeit genauso Kauderwelsch an :(

merlin2k, 2024-02-02:
Modi 4 fände ich besser;)
Modi 1 hat ja nur 12h

trotzdem schon mal vielen Dank

dbambus, 2024-02-02:
Okay, dann mache ich Modi 4 fürs erste :-)

dbambus, 2024-02-04:
Ich habe mal einen Pull Request mit den Änderungen für die 16x18 Wordclock erstellt und gemerged, könntet Ihr den stellen Stand mal testen ? 

Es war doch einiges zu tun um das gangbar zu bekommen :-) 

https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/366

Micha7719, 2024-02-04:
danke, die aktuelle Uhrzeit sieht schon mal gut aus. Muss man mal beobachten wie es sich den ganzen Tag verhält. 
Was schon mal nicht funktioniert ist:  " Es ist " kann man nicht mehr ausschalten. Die Funktion hat keinen Effekt

dbambus, 2024-02-04:
Mit https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/367 kommt der Fix für das "Es Ist". Hatte da anscheinend ein `Break;` vergessen.

Micha7719, 2024-02-05:
perfekt, funktioniert:) wenn jetzt noch irgendwann die andern Modis dazu kommen ist sie perfekt.
Die Uhrzeit sieht bisher auch fehlerfrei aus.
Ein Glück hatte ich nicht den Bau der 16x18 aus dem microcontroller Forum angefangen. Das wäre einiges mehr an Arbeit und Teilen geworden.

Vielen Danke für die tolle Arbeit!

closed 2024-02-05 by dbambus