ESPWortuhr/Multilayout-ESP-Wordclock #359: Zusatz LED´s für Logo

Heiliger-Gral, 2024-01-19:
Hallo, 
erst mal Vielen Dank für das Geile Projekt .. hatte vor Jahren schon mal eine Uhr gebaut mit einer 0815 Software :D 

Jetzt hab ich die gerade endeckt und wieder Lust bekommen was Großes zu bauen in 50x50cm 

Jetzt ist meine Frage ich Möchte zu den Ecken Minuten LED´s unten noch ein Logo/Schriftzug haben den ich auch Beleuchten will (3-5 Zusatz LED´s) 
Ist sowas vorgesehen ? oder leicht umsetzbar ? 

Danke Schon mal

dbambus, 2024-01-20:
Hey @Heiliger-Gral, 

kommt drauf an was du mit dem extra Licht anstellen willst. Soll das Statisch sein? soll das dynamisch die Farbe annehmen wie die Worduhr ? 

Mit freundlichen Grüßen
David

Heiliger-Gral, 2024-01-20:
@dbambus 
Das kann ruhig Statisch sein oder auch die LED farbe nehmen das wäre mir fast egal und brauch auch auf keine Effekt Reagieren 

Gruß

dbambus, 2024-01-22:
Ich Werde mir mal eine Lösung einfallen lassen :-)

dbambus, 2024-02-04:
Ich habe die Idee, die "Frame"-Variante der Uhr mit einer Art Ambilight auszustatten, anstatt nur einen virtuellen Sekundenzeiger zu emulieren. 

So kann man im Code eine zusätzliche Beleuchtung haben, die dann auch separat behandelt wird. Die Implementierung wird wahrscheinlich noch eine Woche oder zwei dauern, da ich momentan nicht viel Zeit habe. 

Ähnlich wie: 
[image]

@pienmatla it is happening

Heiliger-Gral, 2024-02-05:
Das klingt gut Danke

MVarious, 2024-09-29:
Oh daran wäre ich auch interessiert! Klasse !