ESPWortuhr/Multilayout-ESP-Wordclock #360: 11x11 V3 Fehlt im Gui

Heiliger-Gral, 2024-01-20:
Hi hab mir die Uhr die ich in Groß bauen möchte mal in klein gebaut und Festgestellt das in der WebGUI die Version 11x11v3 fehlt und natürlich hab ich genau die version ;) 

Auch hier wird die Minuten Anzeige in der Letzten Zeile und nicht ganz unten bzw in den Ecken dargestellt 

die "DE11x11.v3.hpp" scheint ja vorhanden zu sein

dbambus, 2024-01-20:
Hallo @Heiliger-Gral,

diese version sollte als "🇩🇪 11 × 11 (panbachi)" in der GUI auftreten, oder ?

Mit freundlichen Grüßen
David

Heiliger-Gral, 2024-01-20:
@dbambus 

Die Version ist da und sieht gerade recht gut aus , war aber vor ner stunde noch etwas komisch ich beobachte das mal ...und die  LED´s in den Ecken da nimmt er die ersten 4 der letzten reihe 

MFG

AngeloMontesano, 2024-01-22:
Du musst in /include/Uhrtypes/DE11x11.hpp  , in   folgenden abschnitt    Zeile 51.    " case 2: // LEDs für "Corners" type minute display, 
folgende änderung machen:
Alt:                returnArr[i] = 110 + i;
Neu:                returnArr[i] = 121 + i;

dbambus, 2024-01-22:
Hey @AngeloMontesano, 

danke für den Fix in DE11x11.hpp da muss in der tat ein 121 rein. Gerne ein Pull request für erstellen :-)

Für den zweiten Kommentar, bei beiden "Vor" handelt es sich um das selbe Wort. Nur wird das "v_vor" einzig in der 11x11v2 bzw. 16x18 seperat verwendet. Hierbei handelt es sich um das "Vor" nach dem "Viertel". Bei allen anderen Uhrtypen bedarf es keiner separaten Deklaration und durch das weglassen des "breaks" kann man sich die 3 Zeilen Code sparen, die sonst bei allen anderen Uhrtypen anfallen. 

Analog dazu gibt es auch eine "nach/v_nach" Kombination paar Zeilen zuvor.  

Mit freundlichen Grüßen
David

closed 2024-01-31 by dbambus