ESPWortuhr/Multilayout-ESP-Wordclock #363: Farbwechsel bei Zeitanzeige

Feger2, 2024-01-26:
Hallo!

Ich finde es ein sehr gelungenes Projekt und Hochachtung vor eurer Arbeit.

Ist es möglich während der normalen Zeitanzeige einen Farbwechsel einzustellen?
Ich könnte auch versuchen das selbst einzubauen.
Kann mir vielleicht jemand einen Tip geben in welcher Datei ich anfangen muss zu suchen?
Der Anfang bei einem fertigen Code ist für mich immer das schlimmste.

mfg Feger

dbambus, 2024-01-26:
Hey @Feger2, 

Vielen Dank. Für einen Farbwechsel gibt es mehrere Interpretationsmöglichkeiten? Was genau willst du ausprobieren? 

Relativ einfach wäre in dem der minütliche Zyklus in der `clockwork.hpp` eine Modifikation von `G.color.H`, für Hue, einbauen, damit sich die Farbe minütlich anpasst. 
Vergiss nicht das `parametersChanged` Flag auf `true` zu setzen. 

Vielleicht ist das ja genau das, was du suchst :-) minütlich am Farbrad drehen ^^

Mit freundlichen Grüßen
David

Feger2, 2024-01-26:
Leider ist das nicht das was ich mir vorstelle.
Unter "Funktionen" gibt es ja schon den "Farbwechsel" (schaltet leider alle LEDs ein)
Diesen Farbwechsel würde ich gerne in der normalen "Wortuhr" Anzeige anwenden.
Ich hoffe ich habe es jetzt verständlicher geschrieben.

mfg Feger

dbambus, 2024-01-26:
Ahh okay, 

das ist in der Tat ein interessantes Feature, was ich mir auch schon mal gedacht hatte zu implementieren, aber da ist die Pipeline aktuell zu lang. Aber ich kann gerne unterstützen bei dem programmieren, wenn du willst. 

Letztlich ist es eine Kombination aus der Funktion: 

`setbyFrontMatrix(uint8_t colorPosition, bool applyMirrorAndReverse)`in der `Led.hpp `

und der

`rainbowCycle()` in der `Clockwork.hpp `

Versuche dich da mal einzulesen :-)

Mit freundlichen Grüßen
David

Feger2, 2024-01-26:
Danke erstmals für deine Tips.
Im Moment bin ich mal erschlagen und muss mich erstmal zurecht finden.
Mal sehen ob ich es hin bekomme. :-)

Feger2, 2024-01-26:
Hallo!
Bis jetzt habe ich es geschafft, das bei den Funktionen "Sekunden", "Laufschrift" und "Symbol" ein Farbwechsel stattfindet.
Nur leider leuchten bei der Funktion "Wortuhr" alle 110 LEDs in grün und die Minuten LEDs in der eingestellten Farbe.
Die Minuten LEDs per Farbauswahl einstellen, das gefällt mir gut.
Kannst du dir erklären warum das so ist?

Folgendes habe ich in der led.hpp (geborgt von `void ClockWork::rainbow()` :-)) eingetragen:
```
void Led::setbyFrontMatrix(uint8_t colorPosition, bool applyMirrorAndReverse) {
    static uint16_t hue = 0;

    if (applyMirrorAndReverse) {
        applyMirroringAndReverseIfDefined();
    }
    
    for (uint8_t row = 0; row < usedUhrType->rowsWordMatrix(); row++) {
        for (uint8_t col = 0; col < usedUhrType->colsWordMatrix(); col++) {

            bool boolSetPixel = usedUhrType->getFrontMatrixPixel(row, col);
            if (colorPosition == Background) {
                boolSetPixel = !boolSetPixel;
            }

            if (boolSetPixel) {
                setPixel(row, col,HsbColor(hue / 360.f, 1.f, G.effectBri / 100.f));
            } else if (colorPosition != Background) {
                clearPixel(usedUhrType->getFrontMatrixIndex(row, col));
            }
        }
    }
    show();
    hue++;
    checkIfHueIsOutOfBound(hue);
}
```

Bist du sicher das `void ClockWork::rainbowCycle()` die richtige ist? (ist das nicht die Funktion "Regenbogen"?)
Sollte das nicht die `void ClockWork::rainbow()` sein? (ist das nicht die Funktion "Farbwechsel"?)

Hast du noch einen Tip für mich?

mfg Feger

Feger2, 2024-02-01:
Hallo,
Ich bin jetzt einen Schritt weiter.
Er macht den Farbwechsel, jedoch sehr langsam.
Kann es sein das die Routine `Led::setbyFrontMatrix` nur alle 5min aufgerufen wird?
Wenn ja, dann braucht ein Durchlauf der Farben 60 Stunden. (5min x 360 Farben = 1800min)
Kann man das irgendwie ein wenig beschleunigen? (jede Minute oder alle 30s)

Weiters komme ich mit dem Eintrag `boolSetPixel = !boolSetPixel;` nicht klar. Wenn ich den drinnen lasse, dann leuchten alle 110 Leds. Wenn ich ihn entferne, dann passt es. Hat sicher was mit der Hintergrundfarbe zu tun, aber das Verhalten kapiere ich nicht.

Weiters habe ich eine Checkbox unterhalb der Checkbox "mit Hintergrundfarbe" eingebaut.
Leider habe ich noch Schwierigkeiten, auf den Wert der Checkbox, in der `Led.hpp` zuzugreifen. Kannst du mir bitte dabei helfen?

mfg Feger

dbambus, 2024-02-01:
Hallo @Feger2 

Nice, ich bin schon gespannt wie das aussieht :-). In der Tat, die `setByFrontMatrix` wird nur alle 5 min aufgerufen, da sich auch nur alle 5 min die Front ändert, sofern nicht die Anzeige `MinutenInWords` aktiviert ist. Aber Anscheinend ist in der Uhrsoftware gerade ein Bug vorhanden und die Uhr zeigt keine Minuten an, oder ? 

Bei `if (colorPosition == Background) { boolSetPixel = !boolSetPixel; }` handelt es sich tatsächlich, um den Code speziell für die Hintergrundfarbe. Hier sollen nämlich beim Funktionsaufruf `setByFrontMatrix(Background)` nur alle LEDs leuchten, welche nicht in der Frontmatrix definiert sind.

#Weiters habe ich eine Checkbox unterhalb der Checkbox "mit Hintergrundfarbe" eingebaut.
Leider habe ich noch Schwierigkeiten, auf den Wert der Checkbox, in der Led.hpp zuzugreifen. Kannst du mir bitte dabei helfen? 

Da kann ich dir gerne helfen :-) ich suche die Tage mal alles zusammen :-)

Feger2, 2024-02-03:
Ich glaube die 'Led::setbyFrontMatrix' ist der falsche Ansatz.

- Man müsste in der Hauptroutine eine Globale Variable definieren z.B. autocolorchangeinterval = 30.
- dann mit millis den Intervall feststellen
- Nur dann wenn die Checkbox Farbwechsel und die Funktion Wortuhr angeklickt ist.
- Die hue+1 Farbe mit der aktuellen Saturation und Brightness nach Ablauf des Intervals setzen.

Klingt relativ einfach, aber ich habe noch Schwierigkeiten zu den erforderlichen Werten und Variablen zu kommen. Der Code ist für mich nicht leicht zu interpretieren.
Was hälst du davon?
mfg Feger

dbambus, 2024-03-03:
Hallo @Feger2,

leider gibt es keine einfache Möglichkeit einen GDB mit dem ESP8266 zu realisieren. Besser wäre ein Port für den ESP32 oder RP2040 ^^, aber das dürfte auch ein großer Aufwand sein. 

Vorerst bleibt nur die Möglichkeit Serial.println() gezielt einzufügen und im Serial Monitor zu beobachten.

Nun.
Es gibt etwas ähnliches wie das von dir vorgeschlagene `autocolorchangeinterval = 30`, nämlich die `demo` bei der Transition. man müsste mal sehen, ob man das verwenden kann. 

Wie hast du dir den Code vorgestellt, soll der Farbwechsel der Uhr kontinuierlich erfolgen? also immer? Oder nur wenn die Anzeige wechselt? 

Ich werde deinen Code mal in einen Entwicklungszweig bei mir aufnehmen. Irgendwie habe ich gerade Zeit gefunden mich dem Projekt zu widmen. 

Liebe Grüße
David

Feger2, 2024-03-15:
Hallo!

> Wie hast du dir den Code vorgestellt, soll der Farbwechsel der Uhr kontinuierlich erfolgen? also immer? Oder nur wenn die Anzeige wechselt?

Ich hätte gerne den Farbwechsel kontinuierlich.
Der HSV Farbraum geht ja von 0 bis 360°, also z.b. alle 10 Sekunden +1°
So habe ich es momentan, aber es treten dabei leichte Probleme auf:
- manchmal flackert es, sieht so aus als ob zeitweise die normale Zeitanzeige (ohne Farbwechsel) ganz kurz angezeigt wird
- das Häkchen auf der Farbseite wird nicht gespeichert (bei einem Refresh oder Reboot ist es weg obwohl der Farbverlauf funktioniert)

mfg Feger

dbambus, 2024-03-20:
Hallo @Feger2, 

In der Tat ist es notwendig, alle 60/x s eine Nebenläufigkeit einzuführen. Dazu könnte man einen Zähler in der clockwork.hpp jede Sekunde inkrementieren und ab einem Schwellwert eine Aktion ausführen und den Zähler zurücksetzen. 

Ich habe schon eine Idee, woher das Flackern kommt. Möglicherweise wird die Aktion setbyFrontMatrix() zweimal ausgeführt, was zu dem beschriebenen Flackern führt. Das muss ich aber noch genauer untersuchen. 

Das mit dem Speichern des Häkchens muss ich mir nochmal genauer anschauen, das weiß ich nicht aus dem Kopf. 

Ich hoffe nur, dass ich bald wieder einen freien Tag zum Coden habe :-) 

Mit freundlichen Grüßen
David