ESPWortuhr/Multilayout-ESP-Wordclock #372: Geburtstage auf Webseite eingeben

STT65, 2024-02-25:
Hallo @Wandmalfarbe 

Ich benutze die Uhrenvariante „DE 10x11 Alternativer Rahmen“ (class De10x11AlternativeFrame_t).
Diese Uhrenvariante kann den Text "Happy Birthday" anzeigen.
Diese Feature wird meines Erachtens noch nicht von der Software unterstützt. 

Ist das richtig?

Ich würde mal eine Implementierung versuchen.
Hierzu würde ich in der Webseite der Uhr eine Seite ergänzen, auf der man mehrere Geburtstage eintragen kann.

Hierzu eine Frage: Wurde die Webseite mit einem Tool erstellt oder ist das alles handgeschrieben?

Viele Grüße
  Stefan

dbambus, 2024-02-25:
Hey an @STT65, 

von meiner Seite wurde die Implementierung noch nicht genutzt. ließe sich aber relativ schnell über eine Funktion showHappyBirthday() implementieren. 

Soweit ich weiß wieder die Website und das nötige JavaScript von Hand geschrieben. 

Mit freundlichen Grüßen
David

STT65, 2024-02-28:
Hallo @Wandmalfarbe,
vielen Dank für die schnelle Antwort. 
Ich habe mich schon mal an die Arbeit gemacht. 
Im webPageAdapter.h habe ich ein Problem beim Evaluieren der payload.
Der Code:
           Serial.println(split(payload, 3, 5));
           Serial.println(split(payload, 8));
           Serial.println(split(payload, 11));
           uint16_t year1 = split(payload, 3, 5);
           uint16_t month1 = split(payload, 8, 3);
           uint16_t day1 = split(payload, 11, 3);
           Serial.println(year1);
           Serial.println(month1);
           Serial.println(day1);
Ausgabe der Werte im Monitor, angereichert mit meinen Kommentaren:
2023 --> ok
12 --> ok
23 --> ok
2023 --> ok
1223 --> warum wird hier nicht die 12 ausgegeben
2323 --> warum wird hier nicht die 23 ausgegeben

Hast du eine Idee?

Viele Grüße
  Stefan

STT65, 2024-03-01:
Hallo @dbambus,
ich bin ganz gut vorangekommen. Die Eingabe von 2 Geburtstagen im WebInterface inklusive der Verwaltung der Daten (EEPROM. Globale Variable G, etc) ist fertig. Als nächstes implementiere ich die showHappyBirthday(). 
Den heutigen Stand habe ich in einen neuen Branch namens "birthday" comitted.
Wenn du/ihr Interesse habt, könnte ich den Branch bei euch veröffentlichen!?

dbambus, 2024-03-03:
Hey @STT65,

ja super cool und vielen dank für dein interesse am projekt. ich arbeite parallel noch an ein paar fiesen bugs die sich nicht im code verstecken ^^. Werde heute Abend mal einen Blick auf das werfen, was du an Code geschrieben hast. 

Wenn du noch Fragen hast, dann melde dich einfach :-) 

Ich werde die Clockwork.hpp in den nächsten Tagen auch nochmal ein wenig kommentieren und etwas vereinfachen. Damit es gerade für Einsteiger leichter wird.

Liebe Grüße
David

STT65, 2024-03-04:
Hallo @dbambus, supi, vielen Dank!
Mein GitHub Desktop behauptet, das ich keinen Schreibzugriff auf das Projekt habe. Kann das sein?
Viele Grüße, Stefan

STT65, 2024-03-17:
Hallo @dbambus,
ich würde dir gerne meinen Branch hochladen. Vermutlich benötige ich dafür aber entsprechende Schreibrechte. Kannst du helfen?
Viele Grüße, Stefan

dbambus, 2024-03-20:
Hallo @STT65, Stefan, 

wenn du deine Änderungen in deinen Fork hochlädst, dann kannst du hier einen Pull Request für diesen Branch erstellen. Ich kann ihn dann freigeben :-) und mergen. 

Viele Grüße
David

STT65, 2024-03-24:
Hallo @dbambus,
ok, ich hoffe es hat mit dem Pull Request funktioniert.
Viele Grüße
 Stefan

dbambus, 2024-04-22:
Ich habe den Pull request https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/398 so akzeptiert und werde es nun testen. 
Vielen Dank Stefan.

closed 2024-04-22 by dbambus