ESPWortuhr/Multilayout-ESP-Wordclock #375: Übergang ZUFÄLLIG klappt nicht.

HHasemann, 2024-03-02:
Hallo zusammen,

ich nutze deinen Code und stehe vor einer Herausforderung. Seit dem Update auf Version 3.3 funktioniert der Übergang ZUFÄLLIG nicht mehr so reibungslos wie zuvor. Hast du eine Idee, ob in den neueren Versionen Änderungen vorgenommen wurden, die diesen Effekt verursachen könnten?

Des Weiteren würde mich interessieren, ob es eine Möglichkeit gibt, das Timing für diesen Übergang einzustellen. Es wäre großartig, wenn ich das beispielsweise alle 5 Minute konfigurieren könnte. Kennst du eine elegante Methode, dies umzusetzen?

Ich wäre dir sehr dankbar für deine Hilfe und Anregungen, um dieses Problem zu lösen.

Vielen Dank im Voraus für deine Unterstützung und Expertise!

LG Joachim

dbambus, 2024-03-02:
Hallo @HHasemann 

ich habe heute nach längerer Pause wieder etwas Zeit gefunden, mich dem Code zu widmen. Tatsächlich habe ich einen Bug gefunden, der die Routine für die zufälligen Übergänge nie auslöst. Behoben mit https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/pull/376

Im aktuellen Repo - Main - habe ich das Feature auch etwas verfeinert. Jetzt werden die Transitions nur noch ausgeführt, wenn sich die FrontMatrix ändert, was in der Regel alle 5 Minuten ist :-). Vorher war es schon etwas nervig, wenn bei einer Minutenänderung so viel auf der Uhr passiert. 

Aber vielen Dank für den Hinweis. Ich versuche die Uhr immer fehlerfreier zu machen, aber manchmal gibt es ganz fiese Bugs, die ich vielleicht noch gar nicht bemerkt habe, deshalb bin ich für jeden Bericht dankbar ^^.

Mit freundlichen Grüßen
David

closed 2024-03-02 by dbambus

HHasemann, 2024-03-02:
Hallo David @dbambus,
das ging ja schnell. Habe den Code in meiner Uhr Aktualisiert, ohne Probleme. Vielen vielen Dank.
Gruß Joachim.