ESPWortuhr/Multilayout-ESP-Wordclock #382: Symbol ändern

mrschloemp, 2024-03-21:
Moin, 
anfängerfrage.
Hab die Uhr nachgebaut und jetzt mal ein Update gemacht.
In den Funktionen der Uhr ist jetzt auch Icon enthalten und standartmäßig ein Herz.
Ich finde das Symbol nicht im Source-Zip und wüsste auch nicht wie man das aktualisieren kann. 
Gibt´s dazu ein HowTo?

Danke

dbambus, 2024-03-21:
Hallo @mrschloemp, 

vielleicht kann nicht da an diesen Post verweisen. https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/issues/331

Ich habe aktuell nur ein paar Symbole implementiert, auch durch die Hilfe anderer :-). Für diese gibt es allerdings kein Auswahlmenü, das soll man in einer späteren Version kommen. 

Mit freundlichen Grüßen
David

mrschloemp, 2024-03-21:
Die Antwort kam echt flott. Danke
Richtig nice wäre wenn man auch eigene Symbole hochladen könnte.

Dann warte ich jetzt gespannt auf die kommenden Releases.
Danke

mrschloemp, 2024-03-22:
Mal so auf die schnelle erstellte Symbole E-Mail, Glocke/Klingel, Stop bzw. X und Standby Symbol.
Am Anfang jeden Symbols die 0b000... habe ich verändert, sollte aber überprüft werden.
[own_symbol.txt](https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/files/14721972/own_symbol.txt)

dbambus, 2024-03-22:
hallo @mrschloemp 

Das sieht doch soweit ganz gut aus :-) Ich werde das mal für den nächsten Push übernehmen.

Mit freundlichen Grüßen
David

mrschloemp, 2024-03-22:
Werde noch ein paar mehr erstellen und bei Gelegenheit hier hochladen.

dbambus, 2024-04-18:
Hallo @mrschloemp,

ich werde die aktuellen Symbole von dir in den Kommenden Stand übernehmen und versuchen eine Art Auswahlmenü zu bauen für die Symbole :-) 

Kommt noch was ? oder kann ich fürs erste das Issue schließen ? 

Mit freundlichen Grüßen
David

mrschloemp, 2024-04-18:
Für den jetzigen Moment kommt nichts. 
Wenn ich noch welche mache, kann ich das hier ja wieder reinsetzen.

mrschloemp, 2024-04-18:
Das dauert noch ein bisschen

dbambus, 2024-04-18:
Alles klar :-) dann Close ich das Issue erstmal dann habe ich die Übersicht. Man kann es ja dann erneut öffnen.

closed 2024-04-18 by dbambus