ESPWortuhr/Multilayout-ESP-Wordclock #389: Fehler beim Kompilieren in PlatformIO

Pauli29, 2024-04-06:
Mehrere "unexpected ')' before '[' "
Was mache ich da falsch?
Habe am Code nichts geändert

dbambus, 2024-04-10:
Hallo @Pauli29,

es kann sein, dass sich paar libs aktualisiert haben und ich die entsprechend anpassen muss. Ich bin aber aktuell im Urlaub und kann nur am Handy antworten. Bitte hab noch etwas Geduld.

Mit freundlichen Grüßen
David

dbambus, 2024-04-18:
Hallo @Pauli29,

ich habe es mittlerweile kontrolliert. Bei mir geht es mit dem aktuellen Stand der Website zu kompilieren :-) selbst mit der neuen Version 2.7.9 der Neopixelbus library.

kannst du mal einen git pull machen ? oder Platformio clean ?

Mit freundlichen Grüßen
David

Pauli29, 2024-04-19:
Hallo!

Ich habe vorerst die fertige Bin geflashed aber ich werde es zeitnah testen und eine Rückmeldung geben.

Jedoch bringe ich mqtt nicht zum Laufen :-(

Die eingegebenen sind definitiv richtig aber der Broker und somit auch Homeassistent empfangen nichts.

Was mache ich da falsch?

Beste Grüße

Paul Reinhart

"Organization", "name": "GitHub", "url": "https://github.com" } } ]

udo1toni, 2024-04-21:
In der Version 3.2 bekommt der Broker keinerlei Lebenszeichen über Topics (ist aber in der V3.3 schon drin)
Um zu prüfen, ob der Kontakt da ist, kannst Du das Logging für Mosquitto auf all setzen und den Dienst neu starten, dann muss es einmal pro Sekunde ein Request geben.
Alternativ kannst Du versuchen, einen Effekt einzuschalten, Z.B. mit Topic wordclock:
`mosquitto_pub -h 192.168.178.55 -t wordclock/cmd -m '{"effect":"Rainbow"}'`
Ohne Gewähr, ich nutze immer MQTT Explorer, zumindest Topic und JSON sollten aber korrekt sein. Die IP ist natürlich die Deines MQTT Brokers...

dbambus, 2024-08-26:
Da sich hier in der Issue schon länger nichts mehr getan hat, gehe ich jetzt mal davon aus, dass das Problem gelöst worden ist.

closed 2024-08-26 by dbambus