ESPWortuhr/Multilayout-ESP-Wordclock #413: Menütaster

DirkSchlecht, 2024-06-30:
Hallo,
erstmal großen Respekt vor eurer Arbeit. Ich habe die Uhr nachgebaut und sie funktioniert super.
Ist für die Zukunft geplant, 2-3 Taster zu integrieren um ein paar Grundfunktionen ohne Weboberfläche umzuschalten? Ich stelle mir vor, über die Taster, die im Rahmen verbaut werden sollen, zum Beispiel die Farben und Modi einzustellen.
LG 
Dirk

dbambus, 2024-06-30:
Hallo Dirk, 

ja, das ist geplant und wird von mir auch schon in experimentellen Branches getestet. Leider habe ich im Moment wegen eines Umzugs viel um die Ohren, so dass ich kaum zum Entwickeln komme. 

Viele Grüße
David

DirkSchlecht, 2024-06-30:
Danke für deine schnelle Antwort. Dann warte ich gespannt ab. Leider bin ich bezüglich Programmierung ein absoluter Anfänger. 
Eine digitale Datumsanzeige wäre auch noch klasse. Schön wäre, wenn im Drop-down Menü für den Übergang auch die Sekundenanzeige, Symbole oder das Datum, wenn es das gibt, angewählt werden könnten. 
Viele Grüße Dirk

dbambus, 2024-08-26:
Deine tolle Idee wurde bereits von jemandem auf GitHub aufgegriffen. Siehe dazu https://github.com/ESPWortuhr/Multilayout-ESP-Wordclock/issues/333. Ich werde mich in einer kommenden Version darum kümmern.

closed 2024-08-26 by dbambus